adafruit/Adafruit_CircuitPython_DHT #22: Detector fails on Raspi 3b+ (at least)

andrematzke, 2019-06-01:
Many thanks for this Tipp, to use the newer library Adafruit_CircuitPython_DHT. But this one also fails on my raspi 3b+ to detect the correct raspberry variant:

`[matzke@rasp3 Adafruit_Blinka-1.3.3-py3.7.egg]$ python
Python 3.7.3 (default, Mar 29 2019, 06:12:39) 
[GCC 8.2.1 20180831] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from adafruit_blinka.agnostic import board_id, detector
>>> import sys
>>> import adafruit_platformdetect.board as ap_board
>>> detector.board.any_raspberry_pi_40_pin
False
>>> `

`>>> import platform
>>> print(platform.platform())
Linux-5.1.5-1-MANJARO-ARM-aarch64-with-arch
>>> `

ladyada, 2019-06-01:
what distro is this?

andrematzke, 2019-06-02:
the distro is manjaro-arm.
It seems I constantly have to jump over hurdles, if I try a raspi with that distro to use it for developing and play with sensors and more via GPIO.
Not only the "detector"-mechanism of circuitpython module fails to detect, others do, too (e.g. RPi.GPIO modul). I give up and will use raspbian.

ladyada, 2019-06-02:
yes that is highly recommended  - its the official distro for support!

closed 2019-06-02 by ladyada