adafruit/Adafruit_CircuitPython_DHT #27: Unable to set line 4 to input

a1p4ca, 2019-11-27:
I'm trying to make rest api provides temperature and humidity with dht22. The whole code:

```python3
from flask import Flask, jsonify, request
from sds011 import SDS011
from adafruit_dht import DHT22
import board
import os
import time

app = Flask(__name__)
dht = DHT22(board.D4)

def get_dht_data():
    while True:
        try:
            temperature, humidity = dht.temperature, dht.humidity
            print(temperature, humidity)
            if temperature is not None and humidity is not None:
                return temperature, humidity
            else:
                raise
        except:
            time.sleep(0.5)

@app.route('/', methods=['GET'])
def status():
    temperature, humidity = get_dht_data()

    return jsonify({
        'temperature': temperature,
        'humidity': humidity
    })

if __name__ == '__main__':
    app.run(debug=True)
```

However, when I start server, it shows message 'Unable to set line 4 to input' and temperature and humidity is always None. But If I don't run flask app but just dht code, it works really cool. I think this is belong to Flask, but I have no idea to solve this. Please help me.

ladyada, 2019-11-28:
maybe you have to run as root? not sure - the code works as you said, just not in Flask

monkeywire, 2020-01-24:
..it seems to work fine when I first boot up the raspberry pi and I can run the test code once and it works.... After that if I try to run to the test code a second time I get the same error....

ladyada, 2020-01-24:
try killing any 'libgpiod_pulsein' processes

GratefulCoder, 2020-01-26:
> try killing any 'libgpiod_pulsein' processes

Same issue occurring here. Worked on the first try when on fresh boot, but then subsequent executions would fail. Changing the pin and running the program had the same effect, worked the first time and subsequent attempts would fail with example: 
```
Unable to set line 4 to input
Timed out waiting for PulseIn message
Timed out waiting for PulseIn message
Timed out waiting for PulseIn message
...
```

Locating the process `libgpiod_pulsein` ( `pgrep libgpiod_pulsein` for example ) and `kill <PID found>` did the trick.

Thanks!!!

monkeywire, 2020-01-28:
Works, thank you!

sczerwinski, 2020-04-04:
In my case, `pgrep libgpiod_pulsein` returned nothing. Reboot didn't change anything either.

Fortunately, changing the PIN to `D18` worked.

Still no idea what is blocking GPIO4 though.

jsubirat, 2020-04-16:
### Short answer
Remove `debug=True` and enjoy. :-)

### What was going on?

This problem drove me nuts but I think I found the root cause! In my case, I was encapsulating the DHT22 object in another object, like so:

```
...
class DHT22Custom:
    def __init__(self):
        print("**** BUILDING by {0}!".format(threading.currentThread().getName()))
        self.dht_device = adafruit_dht.DHT22(board.D17)
...
```

My `main.py` looked like:
```
import RPi.GPIO as GPIO
from sensor.config.app_config import create_app

if __name__ == '__main__':
    app = create_app()         # builds DHT22Custom inside
    app.run(debug=True)
    print("Cleaning up GPIO before exiting...")
    GPIO.cleanup()
```

And look what an interesting output I got:

```
root@05600e5:/app# python -m sensor.main
**** BUILDING by MainThread!
 * Serving Flask app "FlaskWS" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: on
 * Running on http://127.0.0.1:5000/ (Press CTRL+C to quit)
 * Restarting with stat
**** BUILDING by MainThread!
Unable to set line 17 to input
```

The MainThread was initializing my object twice! How was that? Well, if you look at the documentation of Flask's `run()`, you'll see the following:

```
If the :attr:`debug` flag is set the server will automatically reload
 for code changes and show a debugger in case an exception happened.
```

So, it seems that Flask just relaunches the application or something like this. Not clear to me, to be honest. But well, if you just remove `debug`, you'll see something like:

```
root@05600e5:/app# python -m sensor.main
**** BUILDING by MainThread!
 * Serving Flask app "FlaskWS" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://127.0.0.1:5000/ (Press CTRL+C to quit)
```

I hope this helps. Happy coding!

RashmiMendis, 2020-04-18:
> try killing any 'libgpiod_pulsein' processes

It's a temporary solution right? It's happening every time I stop the code and running it again. Any solution?

rashmendis, 2020-04-18:
> try killing any 'libgpiod_pulsein' processes

This is a temporary solution right? It happens every time I stop the code and re-run it. Any solution?

poisa, 2020-04-21:
>Fortunately, changing the PIN to D18 worked.

In my case, Pi 3 A+, only pin 11 (aka GPIO17/board.D17) worked. Neither of the other pins suggested here worked. ¯\_(ツ)_/¯

Saeven, 2020-04-26:
> > try killing any 'libgpiod_pulsein' processes
> 
> This is a temporary solution right? It happens every time I stop the code and re-run it. Any solution?

I'm seeing the exact same problem.

```
import time
import board
import adafruit_dht

# Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT22(board.D4)
print("{:.1f},{}".format(dhtDevice.temperature,dhtDevice.humidity))
```

I get **one** reading, any subsequent readings fail.  ps aux | grep pulse reveals this:

    ..../bcm238x/pulseio/libgpiod_pulsein --pulses 81 --queue 7906 -i gpiochip0 4  

Surely, we're not expected to kill PIDs before every reading?

rashmendis, 2020-05-01:
Their old library worked for me. https://github.com/adafruit/Adafruit_Python_DHT

SmackTO, 2020-05-02:
I'm also seeing this problem on a RPi4.  Also, the libgpio_pulsein task uses 25% of my CPU throughput!

The old library worked fine for me too.  I'm going to revert and see what happens.

jsubirat, 2020-05-02:
Did you guys try to remove `debug=True` when initializing Flask? I think it's as simple as that...

Saeven, 2020-05-03:
Not even using Flask - just this script, is enough to make the RPI4 hang:

```
import time
import board
import adafruit_dht

dhtDevice = adafruit_dht.DHT22(board.D4)
```

RashmiMendis, 2020-05-03:
Yeah it is happening even to their sample code. Just try to run the code. Stop it. They run it again.

Saeven, 2020-05-03:
Reverted to the old lib just now, that one's working a treat.

kerel-fs, 2020-06-09:
First of all, if you don't have the  `pulseio` module available (for whatever reason), `adafruit_dht` will use BitBanging and this bug will not appear.

I'm also affected by this bug as my temperature logger script (5min cronjob) suddenly stopped working after 23 hours, so I did some analysis:

### Following the string
The output string `Unable to set line X to input` is generated by the executable `libgpiod_pulsein` at
```c
  // set to an input
  if (gpiod_line_request_input(line, consumername) != 0) {
    printf("Unable to set line %d to input\n", offset);
    exit(1);
  }
```
[adafruit/libgpiod_pulsein:/src/libgpiod_pulsein.c#L237-L241](https://github.com/adafruit/libgpiod_pulsein/blob/19cff93d2bd803e49a874ec0ec7d364170908182/src/libgpiod_pulsein.c#L237-L241).

This executable is called in `PulseIn.__init__()` by `self._process = subprocess.Popen(cmd)`
[adafruit/Adafruit_Blinka:/src/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py#L65](https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py#L65)

### PulseIn ?

A quick look into [PulseIn.py](https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py#L65) reveals that `PulseIn` provides the `__enter__()` and `__exit__()` functions in order to be used as a context manager using the `with` statement.
```python
    def deinit(self):
        """Deinitialises the PulseIn and releases any hardware and software
        resources for reuse."""
        # Clean up after ourselves
        self._process.terminate()
        procs.remove(self._process)
        self._mq.remove()
        queues.remove(self._mq)

    def __enter__(self):
        """No-op used by Context Managers."""
        return self

    def __exit__(self, exc_type, exc_value, tb):
        """Automatically deinitializes the hardware when exiting a context."""
        self.deinit()
```

### PulseIn usage in adafruit_dht

As `adafruit_dht` is not using `PulseIn` as a context manager, I guess it should implement the functionality of `__exit__()` itself by calling `PulseIn.deinit()` at the approriate moment (e.g. in `DHTBase.__exit__()`? This would ensure that there is no long-running `libgpiod_pulsein` process left behind after the python script exited.

Best wishes,
kerel

PS:
```python
# We don't use a context because linux-based systems are sluggish
# and we're better off having a running process
if _USE_PULSEIO:
    self.pulse_in = PulseIn(self._pin, 81, True)
```
[adafruit/Adafruit_CircuitPython_DHT:adafruit_dht.py#L66-L67](https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/master/adafruit_dht.py#L66-L67)
Now I'm not sure if the running process was maybe intentional (but I can't see why this would be a good idea...)?

rinty35, 2020-07-30:
@kerel-fs thanks for your explain.
Could you push a sample to how to deinit the pulsein int the  adafruit_dht context ?

I've a case with two call in the same py script. The First is OK and the second failed with "Unable to set line xxx to input"

rinty35, 2020-07-30:
I find an unclean solution with this code to kill process
_# recherche du process libgpio pour le tuer et pouvoir refaire une mesure dans la meme exec du script python
for proc in psutil.process_iter():
                if proc.name() == 'libgpiod_pulsein' or proc.name() == 'libgpiod_pulsei':
                        proc.kill()_

guytas, 2020-08-04:
I agree with you.  Why not solving the problem instead of patching it?  
The way i see this, is that we are using a 10 wheeler dump truck to move a tiny pail of sand.

I got fed up dealing with stack of problems just to read this device. I wrote a simple python program that sends the start signal via the GPIO output, and then reads the 40 bits from the dth22, and get the result in humidity and temperature. It is a very simple process that uses only the gpio and time libraries. What else should we really need?

brocktucker, 2020-08-08:
> I got fed up dealing with stack of problems just to read this device. I wrote a simple python program that sends the start signal via the GPIO output, and then reads the 40 bits from the dth22, and [...]

Would you mind sharing this script? I've been fighting multiple issues with libgpoid and would love to just leave it behind.

guytas, 2020-08-08:
Yes no problem. Let me figure out how to make it public on github.  I hope you have a pi4.  I do not check or validate the os.

I’ll probably do that tonight.

Kleminator, 2020-08-11:
So this is a super hacky work around to this problem, but all I've done is force an ImportError when the adafruit_dht.py script attempts to import PulseIn. Once it fails to import PulseIn it will attempt to use bitbang instead, this works for me but I guess it depends on your specific use case whether or not it will work for you

Anyway, I modified line 37 in the adafruit_dht.py file located in dist-packages (remember to sudo here as this file is protected). I changed "from pulseio import PulseIn" to "from pulseio import PulseIn1" this then causes an ImportError and then PulseIo is not used which doesn't cause the pin to be blocked

As I said, it's hacky but for now it's a work around to this problem

I tried using the old repo but for some strange reason it wouldn't work on my RPi 4, it kept trying to run the BeagleBone code which obviously does not work.

Hopefully the Adafruit team can solve this properly in a future release

guytas, 2020-08-14:
@brocktucker ,  I have it ready for you to try amd let me know how it works for you.

How do we send personal message in here?

goldyfruit, 2020-08-14:
Same issue here with `fastapi`, I got processes running in background.
```ubuntu     23715  0.1  0.0  10420   704 pts/0    Sl+  01:45   0:00  |               |           |           \_ /home/ubuntu/.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein --pulses 81 --
ubuntu     23718  0.0  0.0      0     0 pts/0    Z+   01:45   0:00  |               |           |           \_ [libgpiod_pulsei] <defunct>
```

goldyfruit, 2020-08-14:
I revert back to the old library, everything works fine.

kerel-fs, 2020-08-14:
This issue will be fixed by Pull Request #46 with implementing what is described in https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/27#issuecomment-641427883.

goldyfruit, 2020-08-14:
> This issue will be fixed by Pull Request #46 with implementing what is described in [#27 (comment)](https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/27#issuecomment-641427883).

I didn't see the PR, thanks. The `dhtDevice.exit()` method solved the issue.

engez7, 2020-09-13:
> The `dhtDevice.exit()` method solved the issue.

Yes! Thank you! 
I've two questions...
1) Why `del dhtDevice` doesn't work ?
2) Why the code below in a Flask app file doesn't release the pin, but in a not Flask script (called by cron) the pin is available after execution?
```
import adafruit_dht
dhtDevice = adafruit_dht.DHT22(4)
temperature, humidity = dhtDevice.temperature, dhtDevice.humidity
print(temperature, humidity)
```

WinBase, 2020-09-27:
I aint using flask, i dont even know what it is, just a basic board layout using the DHT11 as per many tutorials, i switched to circuit python and find it's quite rubbish giving me the input and pulse errors frequently documented elsewhere as well which i cant resolve via code. rebooting/killing process's every time to make buggy code run is just no solution at all, so i changed back to use the old library as per the link here:

https://stackoverflow.com/questions/63232072/cannot-import-name-beaglebone-black-driver-from-adafruit-dht/64090175#64090175

(i added the following to the file described)

elif match.group(1) == 'BCM2711':
    return 3

c'mon adafruit developers, fix it or at least tell us how to reset everything when your script is launched

lifetyper, 2020-10-27:
> 
> 
> In my case, `pgrep libgpiod_pulsein` returned nothing. Reboot didn't change anything either.
> 
> Fortunately, changing the PIN to `D18` worked.
> 
> Still no idea what is blocking GPIO4 though.

Did you enabled 1-wire interface when using GPIO4?
Try disable 1-wire interface from raspi-config when using gpio4 with adafruit python lib.

WinBase, 2020-12-03:
I thought i'd retry this today, i'm using D18 and 1-wire interface has never been enabled. i found that by killing the libgpiod process it all worked, but came back the next time the program was run (same as others found). i also tried using dht.exit() when my program closed, but no luck here either. I'm shocked at how bad the circuit python code is, with no support from the developers. i just cant see how anyone can run the published circuit python code using the standard instructions and sample code, as id consider my pi pretty standard, with the old deprecated code seemingly working fine (which i'm still forced to stick with). - anyone any new ideas?

thanks

kheiakiyama, 2021-01-03:
In my case, a program executes a python script parallels.
I got same error from this simple example.

```
time seq 1 5 | xargs -I ZZ -P 5 python3 /home/pi/dht_simpletest.py
```

`dht_simpletest.py` means below.
https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/master/examples/dht_simpletest.py

`pulseio` does not work my situation.
So I changed code like below.

```
# dhtDevice = adafruit_dht.DHT22(board.D18)
dhtDevice = adafruit_dht.DHT22(board.D18, False)
```

It works for me.

Amit2016-17, 2021-01-18:
I am getting this error , i am using raspi4b  pin GPIO21 ( pin -40 )
pi@raspberrypi:~/IOTstack/Adafruit_CircuitPython_DHT $ sudo python  adafruit_dht.py
Traceback (most recent call last):
  File "adafruit_dht.py", line 17, in <module>
    from digitalio import DigitalInOut, Pull, Direction
  File "build/bdist.linux-armv7l/egg/digitalio.py", line 10, in <module>
  File "build/bdist.linux-armv7l/egg/adafruit_blinka/agnostic/__init__.py", line 8, in <module>
ImportError: No module named adafruit_platformdetect

z0rus26, 2021-04-03:
> So this is a super hacky work around to this problem, all I've done is force an ImportError when the adafruit_dht.py script attempts to import PulseIn. Once it fails to import PulseIn it will attempt to use bitbang instead, this works for me but I guess it depends on your specific use case whether or not it will work for you
> 
> Anyway, I modified line 37 in the adafruit_dht.py file located in dist-packages (remember to sudo here as this file is protected). I changed "from pulseio import PulseIn" to "from pulseio import PulseIn1" this then causes an ImportError and then PulseIo is not used which means the pin won't be blocked after exiting the code
> 
> As I said, it's hacky but for now it's a work around to this problem
> 
> I tried using the old repo but for some strange reason it wouldn't work on my RPi 4, it kept trying to run the BeagleBone code which obviously does not work.
> 
> Hopefully the Adafruit team can solve this properly in a future release

Hello for everyone, im already using raspi w zero, my sensor works really good with old version, but with the new one, i am stock after this step, bc i get this response, when :

import time
import board
import adafruit_dht
dhtDevice = adafruit_dht.DHT11(board.D27)
temperature_c = dhtDevice.temperature

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/ibee/py37-venv/lib/python3.7/site-packages/adafruit_dht.py", line 244, in temperature
    self.measure()
  File "/home/pi/ibee/py37-venv/lib/python3.7/site-packages/adafruit_dht.py", line 200, in measure
    raise RuntimeError("A full buffer was not returned. Try again.")
RuntimeError: A full buffer was not returned. Try again.

apparently it fails for some buffer, but i dont know what should i do next. thanks for any help

z0rus26, 2021-04-03:
I also tried to install on my venv "python 3.7" the oldest version, but it doesnt works, it says that this requirement is already satisfaid, so i check using pip list on venv , and i realize that i cant install correctly. i really apreciate any help.

runnerbob, 2021-07-16:
### CONSIDERATION 1) 
I think the line that would prevent the 504 Error page is this:
slightly modified from @kheiakiyama solution
`dhtDevice = adafruit_dht.DHT22(board.D18, use_pulseio=False)`
Reference line 14 of [Adafruit_CircuitPython_DHT/examples/dht_simpletest.py Commit 8d393f3 on Jan 14](https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/6bb55076631a1f13a8c05595529f0b3a442347fd/examples/dht_simpletest.py)

### CONSIDERATION 2)
For the 502 Bad Gateway Error, I discovered when running the above _dht_simpletest.py_ and watching, that every few lines, the sensor would not be read and it would throw an _Runtime Error_...it would read like this:
```
...
Temp: 78.1 F / 25.6 C    Humidity: 53.5%
Temp: 78.1 F / 25.6 C    Humidity: 53.5%
Temp: 78.1 F / 25.6 C    Humidity: 53.6%
Checksum did not validate. Try again.
Temp: 78.1 F / 25.6 C    Humidity: 53.5%
...
```
The `Checksum did not validate. Try again.` was caught in the above linked _dht_simpletest.py_ by running a **while...try...except** loop.

So, I implemented this same principal in my Flask python setup, and the HTML page has been running for 15 minutes without crashing :)

The **try** portion is the original code, but see in the **except**, I catch the _RuntimeError_, and then I just hard-coded 99 for both temperature and humidity. This could probably be setup better for the database just to ignore this reading...or to get really fancy, use linear interpolation and get a value of the prior and next :)

```
@app.route("/lab_temp2")
def lab_temp2():
    while True:
        try:
            dhtDevice = adafruit_dht.DHT22(board.D17, use_pulseio=False)
            temperature, humidity = dhtDevice.temperature, dhtDevice.humidity
            if humidity is not None and temperature is not None:
                return render_template("lab_temp2.html",temp=temperature,hum=humidity)
            else:
                return render_template("no_sensor.html")

        except RuntimeError as error:
            #This catches *RuntimeError* thrown if sensor value not read
            temperature, humidity = 99, 99
            return render_template("lab_temp2.html",temp=temperature,hum=humidity)
```
Hope this helps some of the Raspberry Pi / Flask users on this issue...and maybe give similar ideas to the rest.

Kim-Hyeong-Seob, 2021-07-16:
> In my case, a program executes a python script parallels.
> I got same error from this simple example.
> 
> ```
> time seq 1 5 | xargs -I ZZ -P 5 python3 /home/pi/dht_simpletest.py
> ```
> 
> `dht_simpletest.py` means below.
> https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/master/examples/dht_simpletest.py
> 
> `pulseio` does not work for my situation.
> So I changed code like below.
> 
> ```
> # dhtDevice = adafruit_dht.DHT22(board.YourPin)
> dhtDevice = adafruit_dht.DHT22(board.YourPin, False)
> ```
> 
> It works for me.

It works for me to. But I don't know why.

ldiazcharris, 2021-11-18:
The solution that I found was:
`dhtDevice = adafruit_dht.DHT22(board.D4, use_pulseio=False)`

Acording to the group comments, this declaration `use_pulseio=False` disable the PulseIn functionality.

axolotl190, 2021-11-22:
adding    dhtDevice.exit()   at the end of the code apparently kills the obstinate system process  from within the program. 

I just see that this has been pointed out before. I had to replace the While statement by a finite "range" loop, however, as the last line must be reached naturally, I guess.
This the only way it works here, but also: no idea why.

axolotl190, 2021-11-27:
Everytime the programs stops without reaching ' dhtDevice.exit()', the offending process must be killed manually.
'os.system("pgrep libgpiod_pulse")'  tells me the relevant process id.  I don't know, however, how to import this number into "os.system("kill ....."). Anybody with an idea?
Thanks

axolotl190, 2021-12-10:
resolved & closed. 
Thank you

ropering, 2022-02-06:
``` python
for proc in psutil.process_iter():
        if proc.name() == 'libgpiod_pulsein' or proc.name() == 'libgpiod_pulsei':
            proc.kill()

```
it works for me

axolotl190, 2022-02-14:
Thank you, ropering!
I cannot try it now, but will soon. In the meantime, I have found a workaround, which I will resort asap.

Zip-Bleep-Boop, 2022-03-02:
my problems mysteriously went away, could have just been loose jumper cables or something?

onykage, 2022-06-24:
so I too was having this issue.  I found that if you set use_Pulseio=False when you declare the pin# then the issue clears up.

didnt see this anywhere commented above.  Just wanting to help out.