adafruit/Adafruit_CircuitPython_DHT #28: Unable to read DHT11 sensor

tukno, 2019-11-30:
Hello, Im getting an error when I trying to create an DHT11 sensor instance in my Raspberry Pi Zero W with Raspbian Stretch:
[image]
I checked my circuit wiring and it is ok. I have tried to install (and reinstall) libgpiod as said in this [link] (https://learn.adafruit.com/circuitpython-on-raspberrypi-linux?view=all#faq-3048170), rebooted the board, tried to run the code again but it did not work.

tukno, 2019-11-30:
Checking the [tutorial](https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/wiring), I've noticed a small text box near the tutorial title that lists the following boards: Raspberry PI / PI A+, B+, 2, 3 .
Does it means the library is not compatible with Raspberry Pi Zero W?

ladyada, 2019-11-30:
try
sudo apt-get install libgpiod2

tukno, 2019-12-01:
I tried but it was unable to find the package in repository. 
I have also tried to download the libgpiod2 armhf package from [Debian package repository](https://packages.debian.org/sid/libgpiod2). Installed it, but still got the same errors.
For the moment I will use the [deprecated python library](https://github.com/adafruit/Adafruit_Python_DHT) that works :man_shrugging:
Thanks for helping.

closed 2019-12-01 by tukno

qgeissmann, 2019-12-11:
Hello there.
Having the exact same issue on pi zero running arch. 
I compiled libgpiod using the AUR PGKBUILD:
```sh
sudo -u alarm git clone  https://aur.archlinux.org/libgpiod.git /tmp/libgpiod.git
cd /tmp/libgpiod.git
sudo -u alarm makepkg -s
pacman -U libgpiod-*.pkg.tar.xz
```
In the end, I get the same error message:

```
Traceback (most recent call last):
  File "fixme.py", line 4, in <module>
    dht_dev = adafruit_dht.DHT11(26)
  File "/usr/lib/python3.8/site-packages/adafruit_dht.py", line 246, in __init__
    super().__init__(True, pin, 18000)
  File "/usr/lib/python3.8/site-packages/adafruit_dht.py", line 66, in __init__
    self.pulse_in = PulseIn(self._pin, 81, True)
  File "/usr/lib/python3.8/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 64, in __init__
    message = self._wait_receive_msg()
  File "/usr/lib/python3.8/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 80, in _wait_receive_msg
    raise RuntimeError("Timed out waiting for PulseIn message")
RuntimeError: Timed out waiting for PulseIn message
```
Run the exact same code, compiled the same library and used the same sensor and wiring on a pi3. It worked out of the box. Will try falling back on the deprecated like @tukno.

Juice805, 2020-05-28:
Hitting this same issue with Pi Model B+  V1.2. Ran `sudo apt-get install libgpiod2` and using the example in [this tutorial](https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/python-setup). The deprecated library works.

```
Traceback (most recent call last):
  File "serial.sh", line 8, in <module>
    dhtDevice = adafruit_dht.DHT11(board.D18)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_dht.py", line 263, in __init__
    super().__init__(True, pin, 18000)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_dht.py", line 69, in __init__
    self.pulse_in = PulseIn(self._pin, 81, True)
  File "/home/justinoroz/.local/lib/python3.7/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 71, in __init__
    message = self._wait_receive_msg()
  File "/home/justinoroz/.local/lib/python3.7/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 89, in _wait_receive_msg
    "Timed out waiting for PulseIn message. Make sure libgpiod is installed."
RuntimeError: Timed out waiting for PulseIn message. Make sure libgpiod is installed.
```

CamDavidsonPilon, 2020-06-23:
I had this issue with the DHT22. 

It initially was_ working, but then I set up a one-wire DS18X20, which made me edit the `config.txt` to include some settings there. After including these settings and switching back to the DHT22, it was not working. So I removed the recently-added lines from the config.txt, rebooted, and everything worked again. Hope this helps someone!

SebastianCl, 2020-07-12:
I have the same error message.
In my case it was caused by a bad connection, connect GND to 3v and 3v to GND.
I guess I have to buy another sensor.

alaub81, 2020-12-28:
I use also a Raspberry Pi zero w. Installed everything which is needed. No chance to get this working on the Pi zero w. On my other Pis (3 and 4) everything worked well with the adafruit-circuitpython-dht library! I hope someone could find a solution on that. So I use the old Library at the moment on my Raspberry Pi zeros.

JavanXD, 2021-08-14:
It works when I run the python script without sudo. Very strange behaviour.
When I run the script with sudo I get RuntimeError "Timed out waiting for PulseIn message. Make sure libgpiod is installed."
- Raspberry Pi Zero WH