adafruit/Adafruit_CircuitPython_DHT #29: Error: libgpiod.so.2: cannot open shared object file: No such file or directory

phazebroek, 2019-12-11:
Hi,

I'm running the example as demonstrated here: https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/python-setup#

I was using the deprecated library (https://github.com/adafruit/Adafruit_Python_DHT) initially with the exact same setup and it is working. So I'm certain the sensor is working.

I have updated my script to import the new library, made some changes as per example but when I'm running the script I get the following error.

I'm fairly new to python, so any help would be welcome to resolve this.

PS. `GPIO_PIN_NR` is declared as `GPIO_PIN_NR = board.D4` because I have connected the data wire of the DHT11 sensor to GPIO pin 4.

Console output:
```
(.env) pi@raspberrypi:~/dht-sensor-reader $ python3 temp.py
/home/pi/dht-sensor-service/.env/lib/python3.7/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein: error while loading shared libraries: libgpiod.so.2: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "temp.py", line 19, in <module>
    dhtSensor = adafruit_dht.DHT11(GPIO_PIN_NR)
  File "/home/pi/dht-sensor-service/.env/lib/python3.7/site-packages/adafruit_dht.py", line 246, in __init__
    super().__init__(True, pin, 18000)
  File "/home/pi/dht-sensor-service/.env/lib/python3.7/site-packages/adafruit_dht.py", line 66, in __init__
    self.pulse_in = PulseIn(self._pin, 81, True)
  File "/home/pi/dht-sensor-service/.env/lib/python3.7/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 64, in __init__
    message = self._wait_receive_msg()
  File "/home/pi/dht-sensor-service/.env/lib/python3.7/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 80, in _wait_receive_msg
    raise RuntimeError("Timed out waiting for PulseIn message")
RuntimeError: Timed out waiting for PulseIn message
```
The full script can be found here:
https://gitlab.com/dht-sensor/dht-sensor-reader/blob/feature/adafruit_lib/temp.py
The DHT11 is instantiated at line 19. The values are read at 62,63

ladyada, 2019-12-11:
run `sudo apt-get install libgpiod2`

phazebroek, 2019-12-11:
Thanks, this did the trick

closed 2019-12-11 by ladyada

zom-pro, 2019-12-31:
I can't find that library in my raspbian installation. I have run update, upgrade, dist-upgrade but can't find it.

```
$ sudo apt install libgpiod2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libgpiod2
```
my OS version

```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.11 (stretch)
Release:	9.11
Codename:	stretch
```

When I try to search for it, I get nothing. When I search for 

```
$  apt-cache search libgpiod
gpiod - Tools for interacting with Linux GPIO character device - binary
libgpiod-dev - C library for interacting with Linux GPIO device - static libraries and headers
libgpiod-doc - C library for interacting with Linux GPIO device - library documentation
libgpiod1 - C library for interacting with Linux GPIO device - shared libraries
libgpiod1-dbgsym - Debug symbols for libgpiod1
```
What am I missing? can I install it from another source?

ladyada, 2020-01-01:
you need to upgrade to buster

phazebroek, 2022-01-11:
I got a fresh new image with Raspberry Pi OS with bullseye, but got the same error again. Sadly the above doesn't work anymore. I noticed https://pimylifeup.com/upgrade-raspberry-pi-os-bullseye/ but since I have more pi's running still on Buster I decided to reinstall the card with the legacy buster image for now, so I can't tell if it helped if I ran `sudo apt install libgcc-8-dev gcc-8-base -y`. I'd love to know if that would help or if something else has to be done?

CesMak, 2022-03-12:
I have a similar issue. If I run my python programm all is fine however if I let a systemd service run my python file I get this error:
```bash
Mar 12 20:33:36 pitwo systemd[1]: Started dht logging service.
Mar 12 20:33:38 pitwo python[1010]: Traceback (most recent call last):
Mar 12 20:33:38 pitwo python[1010]:   File "/home/pi/dht_logger.py", line 18, in <module>
Mar 12 20:33:38 pitwo python[1010]:     dht_device = adafruit_dht.DHT22(board.D4)
Mar 12 20:33:38 pitwo python[1010]:   File "/usr/local/lib/python3.9/dist-packages/adafruit_dht.py", line 305, in __init__
Mar 12 20:33:38 pitwo python[1010]:     super().__init__(False, pin, 1000, use_pulseio)
Mar 12 20:33:38 pitwo python[1010]:   File "/usr/local/lib/python3.9/dist-packages/adafruit_dht.py", line 86, in __init__
Mar 12 20:33:38 pitwo python[1010]:     self.pulse_in = PulseIn(self._pin, maxlen=self._max_pulses, idle_state=True)
Mar 12 20:33:38 pitwo python[1010]:   File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line >
Mar 12 20:33:38 pitwo python[1010]:     message = self._wait_receive_msg(timeout=0.25)
Mar 12 20:33:38 pitwo python[1010]:   File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line >
Mar 12 20:33:38 pitwo python[1010]:     raise RuntimeError(
Mar 12 20:33:38 pitwo python[1010]: RuntimeError: Timed out waiting for PulseIn message. Make sure libgpiod is installed.
Mar 12 20:33:38 pitwo systemd[1]: dht_logger.service: Main process exited, code=exited, status=1/FAILURE
Mar 12 20:33:38 pitwo systemd[1]: dht_logger.service: Failed with result 'exit-code'.
Mar 12 20:33:38 pitwo systemd[1]: dht_logger.service: Consumed 1.922s CPU time.
pi@pitwo:~ $ cd /lib/systemd
```

my service file is:
```
[Unit]
Description=dht logging service
After=multi-user.target

[Service]
Type=idle
ExecStart=/usr/bin/python /home/pi/dht_logger.py
Environment=PYTHONUNBUFFERED=1

[Install]
WantedBy=default.target
```

swapnil91991, 2023-05-03:
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
my raspberry pi

getting error as 

/home/swapnil/dht/.venv/lib/python3.9/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein: error while loading shared libraries: libgpiod.so.2: cannot open shared object file: No such file or directory

can any one help me