adafruit/Adafruit_CircuitPython_DHT #32: DHT-Blinka fail when using libgpiod

yeyeto2788, 2020-02-05:
I just wanted to test the DHT22 sensor as in https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/python-setup but first test the functionality of Blinka.

Also I just encountered this using `digitalio.DigitalInOut`:
```console
In [9]: dhtDevice = adafruit_dht.DHT22(pin)

In [10]: dhtDevice.temperature
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-10-bb6e5b54f869> in <module>()
----> 1 dhtDevice.temperature

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in temperature(self)
    225             data returned from the device (try again)
    226         """
--> 227         self.measure()
    228         return self._temperature
    229 

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in measure(self)
    179                 pulses = self._get_pulses_pulseio()
    180             else:
--> 181                 pulses = self._get_pulses_bitbang()
    182             #print(len(pulses), "pulses:", [x for x in pulses])
    183 

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in _get_pulses_bitbang(self)
    135         """
    136         pulses = array.array('H')
--> 137         with DigitalInOut(self._pin) as dhtpin:
    138             # we will bitbang if no pulsein capability
    139             transitions = []

/usr/local/lib/python3.7/dist-packages/digitalio.py in __init__(self, pin)
     82 
     83     def __init__(self, pin):
---> 84         self._pin = Pin(pin.id)
     85         self.direction = Direction.INPUT
     86 

AttributeError: 'DigitalInOut' object has no attribute 'id'
```

and using the `board.PXX` I get this:
```console
In [1]: import board

In [2]: import adafruit_dht

In [3]: dhtDevice = adafruit_dht.DHT22(board.PC4)

In [4]: dhtDevice.temperature
---------------------------------------------------------------------------
NotImplementedError                       Traceback (most recent call last)
<ipython-input-4-bb6e5b54f869> in <module>()
----> 1 dhtDevice.temperature

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in temperature(self)
    225             data returned from the device (try again)
    226         """
--> 227         self.measure()
    228         return self._temperature
    229 

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in measure(self)
    179                 pulses = self._get_pulses_pulseio()
    180             else:
--> 181                 pulses = self._get_pulses_bitbang()
    182             #print(len(pulses), "pulses:", [x for x in pulses])
    183 

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in _get_pulses_bitbang(self)
    147             dhtval = True   # start with dht pin true because its pulled up
    148             dhtpin.direction = Direction.INPUT
--> 149             dhtpin.pull = Pull.UP
    150             while time.monotonic() - timestamp < 0.25:
    151                 if dhtval != dhtpin.value:

/usr/local/lib/python3.7/dist-packages/digitalio.py in pull(self, pul)
    137             self.__pull = pul
    138             if pul is Pull.UP:
--> 139                 self._pin.init(mode=Pin.IN, pull=Pin.PULL_UP)
    140             elif pul is Pull.DOWN:
    141                 if hasattr(Pin, "PULL_DOWN"):

/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py in init(self, mode, pull)
     45                 if pull != None:
     46                     if pull == self.PULL_UP:
---> 47                         raise NotImplementedError("Internal pullups not supported in libgpiod, use physical resistor instead!")
     48                     elif pull == self.PULL_DOWN:
     49                         raise NotImplementedError("Internal pulldowns not supported in libgpiod, use physical resistor instead!")

NotImplementedError: Internal pullups not supported in libgpiod, use physical resistor instead!
```

Is it possible to set a different pull on pin? Let's say I have already connected the physical resistor on the sensor.

More details:
```console
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> board.detector.chip.id
'SUN8I'
>>> board.detector.board.id
'ORANGE_PI_PLUS_2E'
>>> 
```

This relates to: https://github.com/adafruit/Adafruit_Blinka/issues/245

dherrada, 2020-07-16:
@yeyeto2788 I just wanted to confirm that you do in fact have a physical resistor connected, your wording was slightly vague.

yeyeto2788, 2020-07-16:
Hey @dherrada I connected it to the pin directly but I don't know if the error comes from the library by requiring to have a pull up or pull down resistor connected, as it checks for it. You can see it on the log already attached to this issue.

If there is anything else I can help woth just let me know.

dherrada, 2020-07-16:
@yeyeto2788 Yeah, the error is coming from the library, as in the library (libgpiod) is raising the error, but it's failing exactly like it's supposed to. It's not a parameter that you pass, it's a physical resistor you connect. I'm not sure if you purchased the DHT22 from Adafruit, but if you did, you should have received also received a 4.7k or 10k resistor, which is required to use the DHT11 or DHT22 because libgpiod does not support using internal pullups.

Make sure it's hooked up like it is in this picture here: https://cdn-learn.adafruit.com/assets/assets/000/001/861/original/raspberry_pi_dht22wiring.gif?1447864317

By the way, you're using a dht22 that has headers that you put into a breadboard, not an AM2302 which has 3 wires coming out of it, right?

yeyeto2788, 2020-07-16:
I did connected to a sensor board like this one https://www.amazon.es/DSD-TECH-humedad-temperatura-Raspberry/dp/B07CM2VLBK/ref=mp_s_a_1_1_sspa?adgrpid=63969568209&dchild=1&gclid=EAIaIQobChMIgbnQl8rS6gIVWbvVCh30eQlwEAAYASAAEgJUsPD_BwE&hvadid=308125539048&hvdev=m&hvlocphy=9047050&hvnetw=g&hvqmt=e&hvrand=12206774846927860259&hvtargid=kwd-300180952450&hydadcr=26398_1795523&keywords=dht22+sensor&qid=1594930252&sr=8-1-spons&tag=hydesmo-21&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExWDZHQjNPVTVSSUZMJmVuY3J5cHRlZElkPUExMDQ0NzU5V01ZWThUM0VUVFRWJmVuY3J5cHRlZEFkSWQ9QTEwMzk0NjkyOVc3NzBESjZSTDZLJndpZGdldE5hbWU9c3BfcGhvbmVfc2VhcmNoX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU= ans as far as I know it has a resistor conected to the sensor. 

So let's say that sensor board has the resistor on the board, how do tell libgpiod that you have connected a resistor to the sensor?

dherrada, 2020-07-16:
Ok, that's definitely not supposed to happen. I can't tell you off the top of my head, but I can take a look into this sometime within the next few days

dherrada, 2020-07-16:
Also, do you mind posting the code you're using? I'd assume it's probably the same as what's in the tutorial you linked, but it's a lot easier to replicate the issue if I can be sure I'm running exactly what you are

yeyeto2788, 2020-07-17:
```python
import board​
import adafruit_dht​
dhtDevice = adafruit_dht.DHT22(board.PC4)​
dhtDevice.temperature
``` 
That would be code.

SecT0uch, 2020-11-27:
Hi all,

I'm having the exact same issue with an [AM2302](https://www.adafruit.com/product/393) (Supposed to have a built-in pull-up  resistor) on my Pine A64 LTS.

@yeyeto2788, did you figure it out ?

yeyeto2788, 2020-11-30:
Hey @SecT0uch! Nope, I didn't figure out the issue but I can test it again to see if it now works (assuming there have been changes on dependencies and or this repository).

SecT0uch, 2020-11-30:
Hey @yeyeto2788 ! Yes, if you have time testing that, that'd be great. 
It doesn't seem to work on a non-pi device.
Thanks,

yeyeto2788, 2020-12-01:
Tried and same error:
```shell
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    print(dhtDevice.temperature)
  File "/home/yeyeto2788/workspace/dht11_test/venv/lib/python3.8/site-packages/adafruit_dht.py", line 253, in temperature
    self.measure()
  File "/home/yeyeto2788/workspace/dht11_test/venv/lib/python3.8/site-packages/adafruit_dht.py", line 200, in measure
    pulses = self._get_pulses_bitbang()
  File "/home/yeyeto2788/workspace/dht11_test/venv/lib/python3.8/site-packages/adafruit_dht.py", line 163, in _get_pulses_bitbang
    dhtpin.pull = Pull.UP
  File "/home/yeyeto2788/workspace/dht11_test/venv/lib/python3.8/site-packages/digitalio.py", line 170, in pull
    self._pin.init(mode=Pin.IN, pull=Pin.PULL_UP)
  File "/home/yeyeto2788/workspace/dht11_test/venv/lib/python3.8/site-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 54, in init
    raise NotImplementedError(
NotImplementedError: Internal pullups not supported in libgpiod, use physical resistor instead!
```

Python: **Python 3.8.5**

Libraries installed:
```
Adafruit-Blinka==5.8.0
adafruit-circuitpython-dht==3.5.1
Adafruit-PlatformDetect==2.22.2
Adafruit-PureIO==1.1.7
```

Test script:
```python
import board
import adafruit_dht
dhtDevice = adafruit_dht.DHT11(board.PG7)
print(dhtDevice.temperature)
```

yeyeto2788, 2020-12-01:
**Update:**
Maybe I was too optimistic and secure on my side to do some local changes to the [libgpiod.py](https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py) file and got this error:
```shell
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    print(dhtDevice.temperature)
  File "/home/yeyeto2788/workspace/dht11_test/venv/lib/python3.8/site-packages/adafruit_dht.py", line 253, in temperature
    self.measure()
  File "/home/yeyeto2788/workspace/dht11_test/venv/lib/python3.8/site-packages/adafruit_dht.py", line 209, in measure
    raise RuntimeError("A full buffer was not returned. Try again.")
RuntimeError: A full buffer was not returned. Try again.
```
**NOTE:** Forget the 2 first lines since I just change the `raise` for a `print` to see what this will lead me to.

The interesting part though is that I got some pulses by uncommenting [this](https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/master/adafruit_dht.py#L201):
```shell
Internal pullups not supported in libgpiod, use physical resistor instead!n3.8/site-packages/adafruit_dht.py
Invalid pull for pin: 199
24 pulses: [10897, 6215, 13771, 6207, 13796, 6184, 13811, 6201, 13776, 6202, 13817, 6183, 13826, 6183, 13807, 6200, 13788, 6198, 13820, 6175, 13814, 6206, 13763, 6209]
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    print(dhtDevice.temperature)
  File "/home/yeyeto2788/workspace/dht11_test/venv/lib/python3.8/site-packages/adafruit_dht.py", line 253, in temperature
    self.measure()
  File "/home/yeyeto2788/workspace/dht11_test/venv/lib/python3.8/site-packages/adafruit_dht.py", line 209, in measure
    raise RuntimeError("A full buffer was not returned. Try again.")
RuntimeError: A full buffer was not returned. Try again.
```

So I decided to try at lest 10 time by modifying the `test.py` file to:
```python
import time
import board
import adafruit_dht
dhtDevice = adafruit_dht.DHT11(board.PG7)

for try_number in range(1,11):
  print(f"Try number {try_number}")
  try:
    print(dhtDevice.temperature)
  except RuntimeError:
    time.sleep(2)
```

And got different pulse every time:
```shell
Try number 1
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199
24 pulses: [13987, 6131, 13866, 6106, 13882, 6120, 13892, 6103, 13872, 6110, 13875, 6117, 13865, 6112, 13902, 6093, 13914, 6077, 13891, 6103, 13864, 6136, 13896, 6084]
Try number 2
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199
25 pulses: [3409, 6760, 13294, 6569, 13457, 6504, 13489, 6496, 13487, 6493, 13498, 6515, 13495, 6494, 13502, 6490, 13472, 6530, 13501, 6474, 13489, 6519, 13468, 6520, 13512]
Try number 3
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199
24 pulses: [10641, 6552, 13439, 6544, 13470, 6510, 13481, 6490, 13501, 6497, 13462, 6538, 13500, 6485, 13514, 6472, 13506, 6475, 13484, 6528, 13449, 6544, 13460, 6522]
Try number 4
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199
24 pulses: [17770, 6501, 13478, 6518, 13486, 6504, 13437, 6552, 13469, 6531, 13480, 6499, 13505, 6485, 13494, 6497, 13461, 6533, 13461, 6520, 13512, 6497, 13521, 6461]
Try number 5
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199
25 pulses: [4826, 6570, 13385, 6508, 13436, 6557, 13449, 6527, 13483, 6515, 13484, 6489, 13534, 6484, 13471, 6504, 13474, 6520, 13481, 6503, 13482, 6531, 13462, 6522, 13497]
Try number 6
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199
24 pulses: [12096, 6522, 13435, 6546, 13431, 6552, 13479, 6502, 13491, 6499, 13482, 6501, 13459, 6533, 13466, 6529, 13501, 6489, 13516, 6465, 13521, 6476, 13498, 6490]
Try number 7
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199
24 pulses: [19275, 6572, 13415, 6554, 13269, 6852, 13337, 6537, 13430, 6548, 13457, 6539, 13428, 6536, 13492, 6494, 13509, 6518, 13457, 6517, 13447, 6541, 13437, 6568]
Try number 8
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199
25 pulses: [7159, 6622, 13386, 6569, 13447, 6526, 13454, 6535, 13478, 6520, 13438, 6532, 13471, 6519, 13473, 6527, 13454, 6540, 13443, 6546, 13474, 6528, 13478, 6495, 13506]
Try number 9
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199
24 pulses: [14201, 6427, 13555, 6435, 13570, 6403, 13582, 6405, 13600, 6400, 13614, 6374, 13615, 6384, 13634, 6351, 13656, 6340, 13631, 6355, 13643, 6353, 13651, 6322]
Try number 10
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199
26 pulses: [1721, 5858, 13692, 6497, 13466, 6506, 13510, 6490, 13490, 6493, 13497, 6507, 13527, 6444, 13522, 6485, 13521, 6470, 13513, 6481, 13521, 6448, 13545, 6473, 13532, 6459]
```

SecT0uch, 2020-12-01:
Thanks for your extensive tests. :)
I have no idea how DHT reading works internally.
Do you get the initial error with a physical resistor ?

Maybe @dherrada you know why we don't get a full buffer ?

yeyeto2788, 2020-12-01:
Hey! Another update.

I finally got to read at least once 😃 with another sensor (Seems like the other one was defective) **SEE TRY NUMBER 9**:
```shell
Try number 1
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199 with pull 1
81 pulses: [87, 85, 51, 35, 53, 18, 51, 35, 52, 68, 52, 18, 68, 68, 52, 68, 53, 18, 68, 18, 51, 18, 68, 18, 52, 20, 52, 35, 51, 18, 51, 35, 51, 18, 51, 36, 51, 18, 51, 35, 51, 67, 52, 18, 68, 67, 53, 18, 51, 84, 52, 18, 51, 35, 126, 18, 51, 35, 54, 19, 52, 35, 52, 18, 51, 35, 51, 18, 51, 68, 68, 18, 55, 68, 52, 35, 52, 68, 51, 68, 51]
Checksum did not validate. Try again.

Try number 2
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199 with pull 1
0 pulses: []
DHT sensor not found, check wiring

Try number 3
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199 with pull 1
81 pulses: [51, 18, 52, 34, 51, 18, 52, 83, 52, 18, 51, 83, 51, 18, 52, 67, 67, 18, 51, 35, 51, 18, 51, 36, 51, 18, 51, 35, 51, 18, 51, 35, 51, 19, 50, 35, 51, 18, 67, 67, 51, 18, 67, 67, 52, 18, 51, 84, 51, 18, 51, 55, 22, 35, 51, 18, 53, 34, 51, 18, 51, 34, 51, 18, 67, 18, 51, 18, 67, 67, 53, 18, 67, 68, 51, 18, 51, 84, 51, 18, 51]
Checksum did not validate. Try again.

Try number 4
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199 with pull 1
0 pulses: []
DHT sensor not found, check wiring

Try number 5
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199 with pull 1
81 pulses: [52, 18, 52, 35, 52, 18, 52, 84, 51, 18, 51, 68, 68, 18, 53, 68, 52, 35, 51, 18, 51, 35, 51, 19, 51, 35, 51, 18, 51, 35, 51, 18, 51, 36, 51, 18, 51, 35, 51, 68, 51, 34, 51, 68, 53, 18, 68, 68, 52, 18, 51, 35, 51, 18, 51, 35, 53, 18, 51, 35, 51, 18, 59, 19, 52, 35, 51, 18, 51, 67, 70, 18, 51, 68, 51, 35, 51, 68, 51, 18, 67]
Got temp of: 21
And hum of: 21

Try number 6
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199 with pull 1
81 pulses: [52, 35, 52, 18, 51, 35, 52, 68, 51, 18, 68, 68, 51, 18, 53, 84, 51, 18, 51, 35, 51, 18, 51, 36, 51, 18, 51, 34, 51, 18, 51, 35, 51, 19, 58, 19, 51, 35, 51, 68, 52, 18, 67, 68, 53, 18, 68, 68, 51, 18, 51, 35, 51, 18, 51, 35, 53, 19, 52, 35, 51, 18, 51, 35, 51, 18, 51, 35, 51, 68, 53, 18, 67, 68, 51, 18, 51, 83, 51, 18, 51]
Got temp of: 21
And hum of: 21

Try number 7
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199 with pull 1
0 pulses: []
DHT sensor not found, check wiring

Try number 8
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199 with pull 1
81 pulses: [87, 84, 51, 35, 52, 18, 51, 35, 52, 68, 51, 18, 68, 68, 51, 67, 53, 34, 51, 18, 51, 35, 51, 18, 51, 35, 51, 18, 51, 34, 51, 18, 51, 35, 51, 19, 51, 35, 51, 18, 51, 83, 51, 18, 51, 67, 69, 18, 51, 67, 68, 18, 123, 35, 52, 18, 51, 34, 53, 18, 51, 35, 51, 18, 51, 35, 51, 18, 51, 83, 51, 18, 52, 84, 51, 18, 51, 67, 68, 67, 51]
Checksum did not validate. Try again.

Try number 9
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199 with pull 1
81 pulses: [51, 35, 52, 18, 51, 18, 69, 68, 51, 18, 51, 84, 51, 68, 53, 18, 68, 18, 51, 18, 67, 18, 51, 36, 51, 18, 51, 35, 51, 18, 51, 35, 52, 20, 51, 35, 59, 21, 52, 67, 51, 18, 67, 68, 53, 18, 51, 84, 51, 18, 51, 35, 51, 18, 51, 35, 53, 18, 51, 35, 51, 18, 51, 35, 51, 18, 51, 34, 51, 67, 53, 35, 51, 67, 51, 18, 67, 67, 51, 67, 51]
Got temp of: 21
And hum of: 22

Try number 10
Internal pullups not supported in libgpiod, use physical resistor instead!
Invalid pull for pin: 199 with pull 1
0 pulses: []
DHT sensor not found, check wiring
```

So it is actually returning 81 pulses and several tries but it only converts one of the tries with the same 81 pulses.

Script I used below, I wanted to make sure there was a delay regardless of it read process was right or not.
```python
import time
import board
import adafruit_dht
dhtDevice = adafruit_dht.DHT11(board.PG7)

for try_number in range(1,11):
  print(f"Try number {try_number}")
  try:
    print(f"Got temp of: {dhtDevice.temperature}")
    print(f"And hum of: {dhtDevice.humidity}")
  except RuntimeError as e:
    print(e)
    time.sleep(2)
  else:
    time.sleep(2)
  print("\n")
```

yeyeto2788, 2020-12-01:
@dherrada, @makermelissa maybe a possible solution would be:

1. Set an environment `bool` variable to suppress warnings and just pass over the exception if it is set up on [this lines](https://github.com/adafruit/Adafruit_Blinka/blob/master/src/digitalio.py#L166-L183) within Blinka.
2. In this module (DHT one) maybe a bit of refactoring to take the last 80 pulses if the length of the pulses is within a threshold (81-83) ?.

yeyeto2788, 2020-12-01:
Sorry for the spam.

I have added the option mentioned in my last comment within my fork and execute the following commands which are to get similar results from "scratch":
```shell
mkdir dth_test
cd dht_test
python3 -m virtualenv venv --system-site-packages 
sudo su
```
Type your password 😛 

within that folder create a file called `test.py` (could be whatever name you want) with the following code:
```python
import time
import board
import adafruit_dht
from adafruit_blinka.constants import SUPPRESS_WARNINGS
dhtDevice = adafruit_dht.DHT11(board.PG7)

print(f"value of SUPPRESS_WARNING: {SUPPRESS_WARNINGS}")

for try_number in range(1,11):
  print(f"Try number {try_number}")
  try:
    print(f"Got temp of: {dhtDevice.temperature}")
    print(f"And hum of: {dhtDevice.humidity}")
  except RuntimeError as e:
    print(e)
    time.sleep(2)
  else:
    time.sleep(2)
  print("\n")
```

And finally, execute the rest of the commands.

```shell
source venv/bin/activate
PYTHON=$(which python)
$PYTHON -m pip install --upgrade pip
pip install git+https://github.com/yeyeto2788/Adafruit_Blinka.git
pip install adafruit-circuitpython-dht
export SUPPRESS_WARNINGS=True
$PYTHON test.py
```

The output should be similar to this:
```shell
value of SUPPRESS_WARNING: True
Try number 1
DHT sensor not found, check wiring

Try number 2
DHT sensor not found, check wiring

Try number 3
DHT sensor not found, check wiring

Try number 4
Got temp of: 20
And hum of: 23

Try number 5
Got temp of: 20
And hum of: 22

Try number 6
DHT sensor not found, check wiring

Try number 7
DHT sensor not found, check wiring

Try number 8
Got temp of: 20
And hum of: 23

Try number 9
Got temp of: 20
And hum of: 23

Try number 10
DHT sensor not found, check wiring
```

Let me guys know if I can file a PR to Blinka directly.

yeyeto2788, 2020-12-11:
Hey guys! Any updates here?

@ladyada, @makermelissa, @kattni do you guys think it is worth it to make a PR with the changes proposed to blinka on this issue so anyone can benefit from it? 

My main concern is to know if we could damage the boards by having the pull-up resistors connected properly to the board and suppressing the warnings? Could it be a side effect to it? So far I haven't experienced any but just to make sure or double-check for any possibilities of damage.

SecT0uch, 2020-12-11:
@dherrada Any news on that? 
For instance this module is not working for Pine64, Orange Pi and potentially more board brands..

@yeyeto2788 is offering to submit a PR to fix that

dherrada, 2020-12-11:
> Maybe @dherrada you know why we don't get a full buffer ?
@SecT0uch Yeah, I have no idea

dherrada, 2020-12-11:
As far as the PR, I really can't speak to the specifics as I'm not super familiar with Blinka, although it seems like you've already done the bulk of the work in fixing it so I really can't see any reason why someone would have an issue with you opening a PR.

Also, @yeyeto2788 excellent job fixing this. It had me completely stumped and I had to put in a solution that didn't actually fix anything, just caught the errors.

ladyada, 2020-12-12:
@yeyeto2788 please submit a PR so we can see what you're doing and we can comment on it there :)

PR #56 (2020-12-14): Fix issue #32
This will finally fix the issue on other boards like the Orange Pi boards 😄 

Fixes #32
closed 2020-12-14 by ladyada