adafruit/Adafruit_CircuitPython_DHT #33: A full buffer was not returned.  Try again.

fniko, 2020-02-14:
Hi,
using Raspberry Pi Zero with DHT11 sensor and I am observing this issue "A full buffer was not returned.  Try again." after running 'dht_simpletest.py'. 
I edited the values to match DHT11 and P4 (i use)
```
dhtDevice = adafruit_dht.DHT11(board.D4)
```

I tested the sensor using the old library and it works (_90+ % succ read rate_), but using the new library I can barely make one successful request.

How should I troubleshoot this?

scholi, 2020-03-07:
same problem here:  Python 3.7.3, raspeberry pi 0, Linux raspberrypi 4.19.97+ #1294 
adafruit_dht.__version__ '0.0.0-auto.0'

thatsjet, 2020-04-21:
Same here. I've implemented a try/raise with sleep of 2 seconds but it's a LOT less reliable than the old Python 2.7 library.

Iril, 2020-04-26:
This was also my problem and so back to the ﻿﻿Adafruit_DHT library. Working 95% of the time on Python3 and Balena.

lilcoltsmith, 2020-05-02:
I'm getting the same issue. I have the following:

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

- Raspberry Pi Zero W v1.1
- DHT-22 
- Python 3.8
- adafruit-circuitpython-dht==3.4.1

Here is a sample of the output from my script:
```zsh
+-----------------------------------------------------------------+
[DHT-22] Connection to API refused. Starting in local mode.
[DHT-22] A full buffer was not returned. Try again.
[DHT-22] Sensor failure. Check wiring.
+-----------------------------------------------------------------+

+-----------------------------------------------------------------+
[DHT-22] Connection to API refused. Starting in local mode.
[DHT-22] A full buffer was not returned. Try again.
[DHT-22] Sensor failure. Check wiring.
+-----------------------------------------------------------------+

+-----------------------------------------------------------------+
[DHT-22] Connection to API refused. Starting in local mode.
[DHT-22] A full buffer was not returned. Try again.
[DHT-22] Sensor failure. Check wiring.
+-----------------------------------------------------------------+

+-----------------------------------------------------------------+
[DHT-22] Connection to API refused. Starting in local mode.
[DHT-22] Temperature: 20.0C Humidity: 53.7% Date/Time: 2020-05-01 23:41:14
[DHT-22] Queue size: 1
[DHT-22] Local Mode: False, Zip Code Exists: False
[DHT-22] Writing queue to file.
+-----------------------------------------------------------------+

+-----------------------------------------------------------------+
[DHT-22] Connection to API refused. Starting in local mode.
[DHT-22] Temperature: 19.9C Humidity: 53.0% Date/Time: 2020-05-01 23:41:25
[DHT-22] Queue size: 1
[DHT-22] Local Mode: False, Zip Code Exists: False
[DHT-22] Writing queue to file.
+-----------------------------------------------------------------+

+-----------------------------------------------------------------+
[DHT-22] Connection to API refused. Starting in local mode.
[DHT-22] A full buffer was not returned. Try again.
[DHT-22] Sensor failure. Check wiring.
+-----------------------------------------------------------------+
```

This is with a 10 second interval between checks. The only message that isn't from my script is:
*__A full buffer was not returned. Try again.__*

skddc, 2020-07-03:
Just ran into the same error. Any news on this one?

jerryneedell, 2020-07-03:
FYI -- I just tried this on a Pi ZeroW -- it is intermittent for me as well-- This has been a common problem with the DTH22.   
```
pi@gjnpideck:~/projects/blinka $ python3 dht_simpletest.py 
Temp: 78.4 F / 25.8 C    Humidity: 48.2% 
A full buffer was not returned. Try again.
A full buffer was not returned. Try again.
Temp: 76.6 F / 24.8 C    Humidity: 52.1% 
Temp: 76.6 F / 24.8 C    Humidity: 52.1% 
Temp: 76.6 F / 24.8 C    Humidity: 52.1% 
Temp: 76.6 F / 24.8 C    Humidity: 52.2% 
Temp: 76.6 F / 24.8 C    Humidity: 52.2% 
Temp: 76.6 F / 24.8 C    Humidity: 51.3% 
A full buffer was not returned. Try again.
Temp: 76.6 F / 24.8 C    Humidity: 52.2% 
Temp: 76.6 F / 24.8 C    Humidity: 52.1% 
Temp: 76.6 F / 24.8 C    Humidity: 52.0% 
Temp: 76.6 F / 24.8 C    Humidity: 52.0% 
```

lane-maxwell, 2020-07-09:
Same here, it intermittently fails

     humidity = dhtDevice.humidity
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_dht.py", line 252, in humidity
    self.measure()
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_dht.py", line 198, in measure
    raise RuntimeError("A full buffer was not returned. Try again.")
RuntimeError: A full buffer was not returned. Try again.

dbforma, 2020-07-26:
I have run into the same problems. I have made some slight changes to the code and now I do not get these errors anymore.

```
DHTSensor = adafruit_dht.DHT22(board.D27)

class AirMoistureAndTemperatureSensor(SensorRepository):

    AirMoistureValue = DHTSensor.humidity
    AirTemperatureValue = DHTSensor.temperature

    def getSensorValues(self):
        try:
            return self.AirMoistureValue,  self.AirTemperatureValue

        except RuntimeError as error:
            print(error.args[0])

        time.sleep(2)
```

snrkl, 2020-09-05:
I am having the same issues.. If I pop out to a python 2.7.2 environment, the old library works a charm...

> `Python 3.7.3 (default, Jul 25 2020, 13:03:44)`
> `[GCC 8.3.0] on linux`
> `Type "help", "copyright", "credits" or "license" for more information.`
> `>>> import time`
> `>>> import board`
> `>>> import adafruit_dht`
> `>>> dhtDevice = adafruit_dht.DHT11(board.D17)`
> `>>> temperature_c = dhtDevice.temperature`
> `Traceback (most recent call last):`
> `  File "<stdin>", line 1, in <module>`
> `  File "/home/snrkl/python-venvs/envlogd/lib/python3.7/site-packages/adafruit_dht.py", line 253, in temperature`
> `    self.measure()`
> `  File "/home/snrkl/python-venvs/envlogd/lib/python3.7/site-packages/adafruit_dht.py", line 209, in measure`
> `    raise RuntimeError("A full buffer was not returned. Try again.")`
> `RuntimeError: A full buffer was not returned. Try again.> `

> `>>> temperature_c = dhtDevice.temperature`
> `Traceback (most recent call last):`
> `  File "<stdin>", line 1, in <module>`
> `  File "/home/snrkl/python-venvs/envlogd/lib/python3.7/site-packages/adafruit_dht.py", line 253, in temperature`
> `    self.measure()`
> `  File "/home/snrkl/python-venvs/envlogd/lib/python3.7/site-packages/adafruit_dht.py", line 209, in measure`
> `    raise RuntimeError("A full buffer was not returned. Try again.")`
> `RuntimeError: A full buffer was not returned. Try again.> `

> `>>> temperature_c = dhtDevice.temperature`
> `>>> temperature_c`
> `>>> str(temperature_c)`
> `'None'> `

> `>>> dhtDevice.humidity`
> `Traceback (most recent call last):`
> `  File "<stdin>", line 1, in <module>`
> `  File "/home/snrkl/python-venvs/envlogd/lib/python3.7/site-packages/adafruit_dht.py", line 263, in humidity`
> `    self.measure()`
> `  File "/home/snrkl/python-venvs/envlogd/lib/python3.7/site-packages/adafruit_dht.py", line 209, in measure`
> `    raise RuntimeError("A full buffer was not returned. Try again.")`
> `RuntimeError: A full buffer was not returned. Try again.`
> `>>> dhtDevice.humidity> `

> `>>> dhtDevice.temperature`
> `Traceback (most recent call last):`
> `  File "<stdin>", line 1, in <module>`
> `  File "/home/snrkl/python-venvs/envlogd/lib/python3.7/site-packages/adafruit_dht.py", line 253, in temperature`
> `    self.measure()`
> `  File "/home/snrkl/python-venvs/envlogd/lib/python3.7/site-packages/adafruit_dht.py", line 209, in measure`
> `    raise RuntimeError("A full buffer was not returned. Try again.")`
> `RuntimeError: A full buffer was not returned. Try again.> `

> `>>> dhtDevice.humidity`
> `Traceback (most recent call last):`
> `  File "<stdin>", line 1, in <module>`
> `  File "/home/snrkl/python-venvs/envlogd/lib/python3.7/site-packages/adafruit_dht.py", line 263, in humidity`
> `    self.measure()`
> `  File "/home/snrkl/python-venvs/envlogd/lib/python3.7/site-packages/adafruit_dht.py", line 209, in measure`
> `    raise RuntimeError("A full buffer was not returned. Try again.")`
> `RuntimeError: A full buffer was not returned. Try again.> `

_**Python 2 with the old library:**_
> `0 /home/snrkl/snrkl/bin $ python`
> `Python 2.7.16 (default, Oct 10 2019, 22:02:15)`
> `[GCC 8.3.0] on linux2`
> `Type "help", "copyright", "credits" or "license" for more information.`
> `>>> import Adafruit_DHT`
> `>>> humidity, temperature = Adafruit_DHT.read_retry(11, 17)`
> `>>> humidity`
> `59.0`
> `>>> temperature`
> `24.0`

gnorbsl, 2020-09-06:
same here, after reboot nothing works anymore... I'm going to switch back to the old since since one seems to be not maintained either.

austin7777, 2020-09-30:
Same problem

WinBase, 2020-10-10:
same problems, i've tried for hours with no luck

the old deprecated library (on latest python 3.7.3) works, even though i get plenty of misreads and have a 'retry' loop every 3 seconds to eventually get a good read, but circuitpython is total b*ll*cks, it gives the following messages in a random order:

sometimes at the start it says: 'unable to set line 23' to input (the old library works just fine on this pin)

then:

Timed out waiting for PulseIn message. Make sure libgpiod is installed. -  FYI: latest libgpiod2 IS installed)
Reading from DHT failure:  ('DHT sensor not found, check wiring',)
Reading from DHT failure:  ('Checksum did not validate. Try again.',)
Reading from DHT failure:  ('A full buffer was not returned. Try again.',)

i rebooted the pi zero, and whoopee started to get the occasional reading inbetween 10-20 failures, used ctrl-c to cancel, re-tried a minute later, and then no readings at all, just error messages.

to deprecate a working library replacing it with a buggy non-working one leaves me (almost) speechless, and i'm unlikely to consider adafruit software or products for my projects, i think i'll look elsewhere if it aint sorted soon.

btw my pi is fully upgraded and updated

GL

Bob

davidoesch, 2020-10-11:
same here ---Adafruit_CircuitPython_DHT  does not work in 80% of the cases

mafrosis, 2020-10-25:
Some more anecdata. From my logs - successful reads are followed by 10 second sleep, failed reads are followed by 0.2 secs sleep.
```
.. snip
19.8 45.8
19.9 45.8
A full buffer was not returned. Try again.
19.9 45.8
DHT sensor not found, check wiring
19.9 45.8
19.8 44.5
A full buffer was not returned. Try again.
19.8 44.5
19.8 45.6
19.8 45.6
19.8 45.7
19.8 45.7
A full buffer was not returned. Try again.
19.8 45.7
19.8 45.8
19.8 45.7
19.8 45.6
DHT sensor not found, check wiring
19.8 45.6
19.8 45.8
A full buffer was not returned. Try again.
19.8 45.8
19.8 45.8
.. snip
```

Likewise, the old non-circuitpython library works fine, but the new one is unreliable. Possibly the old library hid the unreliability of the DHT22 itself?

@jerryneedell @dherrada @kattni @ladyada - As major contributors and/or representatives of Adafruit, could you comment on this thread? I'm interested to know if this is by design / fixable / on the roadmap etc. Thanks

tannewt, 2020-10-26:
>  Possibly the old library hid the unreliability of the DHT22 itself?

This appears to be the case: https://github.com/adafruit/Adafruit_Python_DHT/blob/master/Adafruit_DHT/common.py#L83-L98

If you used `read_retry` in the old library, it would retry 15 times by default and wait 2 seconds between each retry. Folks above didn't specify the old code they were comparing against so I can't be sure.

The underlying pulse code of the old library is written in C here: https://github.com/adafruit/Adafruit_Python_DHT/blob/master/source/Raspberry_Pi/pi_dht_read.c I'm not sure how pulses are counted via Blinka. @makermelissa would know.

> I'm interested to know if this is by design / fixable / on the roadmap etc. Thanks

My API design preference is to *not* hide unreliability because it allows the user to decide what to do. It's simple enough to rewrite `read_retry` in your own code.

It's important to note this library is shared with CircuitPython and problems with *this* library would show with it as well. Linux isn't great for real-time so this is more likely an issue with the underlying pulse tracking on Linux.

WinBase, 2020-10-27:
> Possibly the old library hid the unreliability of the DHT22 itself?

quite possibly indeed, but at least you can get readings to some degree, which i'm finding almost impossible in the new circuitpython version. I think i stumbled across a library written by someone else that said it worked fine with the DHT22, so maybe i'll try find it again when i get the opportunity  and give it a spin.

regards.

shawnerz, 2020-11-02:
I thought it was just me! 
I spent over 8 hours today (Sunday) trying to figure out what went wrong. I switched between a Pi Zero 1.2 and a Pi Zero W 1.1. I tried a 1K resistor (as mentioned in the manufacture's datasheet) and a 10K resistor (in the Adafruit and Pimylifeup pages. I didn't try 4.7k). I tried sample rates as slow as 30 seconds. I'm getting about a 95% failure rate.
In the process, I heat damaged a SD card (from soldering too close to it) and had to reload the OS and what I had done to a new SD card. Nothing seemed to work.
Interestingly enough, 2 months ago (early September), I used an old Pi Zero ver. 1.2 (no wi-fi), a DHT22, and 10K resistor to monitor room temperature. I am getting about a 75% success rate.
Because the DHT22's are so cheap, I bought 6 on Amazon. I'm using the ones from that purchase. I duplicated the set up, and getting about a 95% failure rate.  *Something* had to have changed in the code base from September.
It just so happens I am going to that site tomorrow (Monday). I'm going to image that SD card and run it here at home to see if I get different results.
-Shawn

andrewhao, 2020-11-02:
I have the same issues as people on this thread for Pi Zero and Pi 3 Model
B. For those cases I used the old library. This does work on my Pi 4
though. Hope that data point helps.

shawnerz, 2020-11-02:
Thank you for your response! How do I get and install the old library?

fniko, 2020-11-02:
@shawnerz https://github.com/adafruit/Adafruit_Python_DHT

shawnerz, 2020-11-02:
Thank you very much. I will install it tonight.-Shawn

shawnerz, 2020-12-05:
Oh, I forgot to leave an update. 
My problem was on one Pi Zero, I got a 75% response rate. It is currently in use at a site and monitoring the temperature. 
On another Pi Zero, I got a 25% response rate. 
The Pi Zero'd are the same types (no Wi-Fi). Same Class 10 SD cards. The same IO's are enabled/disabled in raspi-config. I have 4 DHT22's. I cycled through DHT22's. No change. I even captured the communication waveforms with my digital oscilloscope. I compared waveforms between the 75% and 25%. Very little difference. 
I followed Adafruit's instructions here: https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/python-setup. Pip installs the library as adafruit_dht.
fniko's link did give me the older library. But it installs the library as Adafruit_DHT. I made the changes in my Python code, but then started getting an error about a variable in the library being used as a float. I felt myself going down a rabbit hole of code changes deep in libraries. I didn't feel like going down that hole.
On both the 75% Pi and the 25% Pi, I did sudo apt list --installed and saved the results to a file. I thought I installed the same version of Raspbian on both Pi's. I did not. I do not know which version I installed. I cat-ed /etc/os-release and they are the same.
The 25% Pi did not have all of the same GPIO packages installed. I installed them, but there was no change. I felt myself going down another rabbit hole.
But, I solved my problem. I switched to the MCP9808. The Relative Humidity (RH) measurement on the DHT22 is nice. But, I don't need it. For my application, temperature is important.
Version 2 of my temperature monitor will be installed this month. The MCP9808 responds every time when it is queried.
Thank for the suggestions, though.
-Shawn

TerryHowe, 2020-12-18:
I changed the print to show the buffer size that it gets and on my Zero W and I'm getting between 30-65 instead of the required 80+. Maybe my sensor is even slower than most. I switched to the deprecated library and that works fine. I wonder if 80 is just too high an expectation.

shawnerz, 2020-12-19:
I don't know. All I know is the MCP9808 responds every time.
I don't know if anyone is interested, but it's chance for me to show off some of my hardware. :) During the troubleshooting process, I wondered if there was an issue with the line levels, or timing, or something I could see with an oscilloscope. I have a capture of a waveform where I did get a result, and a waveform where I did not get a result. They look almost identical-which tells me that the problem is not a physical interface problem.
[image]
[image]
The capture was with a Digilent Analog Discovery 2 module. 
Enjoy,
-Shawn

roslynn, 2021-01-01:
@shawnerz - in a process of validating whether it is the sensor being broken or maybe PEBCAK - I check with mine too haha I used PicoScope though (PS5444D) with x10 probe and pulse-width trigger 😄 
Do you maybe happen to know what standard do they use (if not custom)? It would be cool to add a decoder for it :)

[image]

This data should represent something ~ 21.5*C and 55.6% humid.

Anyway, my story was - first figuring out what is the mapping for `board` python module and the pins (I thought it was wrong) then I realised I missed the pin name (PEBCAK) - and it finally worked 🥳 but with this message _A full buffer was not returned. Try again._ 🤔
I updated my code to run in a loop (similar to the _simpletest_ one) and I can confirm that it transmits the data fine (no distortion) but the python library must be doing something wrong... (I will check the sources next!)

About 20% it is correct, other times it flickers between the mentioned message, _"Checksum did not validate. Try again."_ and occasionally even "_DHT sensor not found, check wiring_" - which makes me think that maybe the connection on the board might be wacky - thresholds are not met appropriately and the information is misinterpreted. 🤷 

Good luck :)

Markrwp, 2021-01-20:
I am yet another frustrated user of this Adafruit_CircuitPython_DHT11 software on a RaspberryPi Zero W, experiencing exactly the same problems mentioned above. My readings are successful about 66% of the time, with a 10 second sleep time between print of data. The rest of the time I get the error messages given by fniko (usually 'full buffer was not returned'). As others have said, the old library works much better. When I looked at the offending software on https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/master/adafruit_dht.py, I notice that there is in line 116 a 0.25 second timeout in the loop used to collect data pulses. Has anyone at Adafruit tried adjusting this to make the loop run longer to gather sufficient data? (Forgive me everyone who has more experience than me in things Python if I am making a fundamental error in this suggestion - I'm a Newbie here - but we can't all have faulty Pis or faulty sensors. Just trying to make a positive contribution). Come on Lady Ada, sort it out!

m-keller, 2021-02-06:
Experienced the same error and issue was that I have been using bitbanging instead of PulseIO (CircuitPython) because CircuitPython was not installed yet on my RPi Zero W.

I followed https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi with an additional “pip3 install sysv-ipc” (no idea why missing ...) and now my DHT22 works perfectly fine. Bitbanging is likely more of a last resort nevertheless clearly not working well - I got about 60 of the 81 required pulses when bitbanging was used.

grigal, 2021-02-09:
I'm getting the same errors `A full buffer was not returned. Try again.` and less frequently `Checksum did not validate.` while measuring every 60 seconds (with `use_pulseio=False`).
- Raspberry Pi 3b `Linux piberry 4.19.68-v7+ #1255` 
- 3-pin DHT22 sensor
- `adafruit-circuitpython-dht==3.5.4`
- Running on a docker image, python 3.8.7

I used to use the [old library](https://github.com/adafruit/Adafruit_Python_DHT), but as it was deprecated I decided to switch to this one. 
Hope this will be fixed, as I wouldn't like to use the deprecated one.

blosloos, 2021-02-13:
Same error here, continuously `A full buffer was not returned. Try again`. Will this be fixed in the future?

bsolino, 2021-02-17:
Similar issues here, with a Pi Zero W.
I get the `DHT sensor not found, check wiring.` error more often, so I've left my full comment there: https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/49#issuecomment-780196763

QS3445, 2021-03-06:
I'm having similar issue "A full buffer was not returned. Try again.." with Pi Zero WH and the DHT22, reading through these comments, there doesn't seem to be a definitive answer yet on how to fix.

My software setup seems to be good using python3, a Blinka test is good, the wiring i have checked and is good. So what can i try now.... and troubleshoot further steps I could try. Thanks for any suggestions.

shawnerz, 2021-03-07:
As I understand it, there is something wrong in Adafruit's library. I don't know what it is. The transmit and receive waveforms look good on a 'scope.
@roslynn Sorry I missed your comment a few months ago.
I switched to the MCP9808, commented out my DHT22 code, and haven't looked back. 
The MCP9808 doesn't measure humidity. The funny thing was, here in TX, we had a big thaw after the snow. The humidity went way up in our equipment trailer. Our ham radio club's Acurite sensor reported the high humidity alert but mine didn't. :( Oh well, maybe one day in the future.

olixr, 2021-03-28:
Confirmed problem. Works okish on the Pi4 and I can get readings. About a 95%+ failure rate on the Pi zero W. I let some code run on the pi zero all night and it never once was able to get a reading. Exact same code on Pi 4 is able to get a reading 50% of the time. 

Other DHT libraries work perfectly fine so it must be a software issue. Hope there is a discovery on the issue soon as we are now reverting to old / other libraries in the meantime for better stability.

injectedfusion, 2021-03-29:
Can confirm as well the older libraries work better.

bitsvital, 2021-04-05:
> Confirmed problem. Works okish on the Pi4 and I can get readings. About a 95%+ failure rate on the Pi zero W. I let some code run on the pi zero all night and it never once was able to get a [...]

I did the same test late last week with same results. I concur, it has to be a software issue.

ikorbln, 2021-05-31:
Same problem here. Unfortnetly the old library did not work anymore on a fresh pi light install.

alejandrobalderas, 2021-12-28:
I had the same issues when trying to read from a sensor on a pi zero however when using the pi zero 2w it works . I used the same SD card for both tests

shawnerz, 2021-12-29:
That's very interesting. If you query the sensor every 10 seconds for 10 minutes, what is the success rate?
I wanted to try stepselyuk's suggestion. But, I haven't had the need/motivation to do it. I might in the future.
Thanks.

jrc, 2022-04-21:
From https://learn.adafruit.com/circuitpython-on-google-coral-linux-blinka/faq-troubleshooting

> Some libraries, like [Adafruit_CircuitPython_DHT](https://github.com/adafruit/Adafruit_CircuitPython_DHT) will try to bit-bang if pulsein [sic] isn't available. Slow linux boards (<700MHz) may not be able to read the pins fast enough), you'll just have to try!

Installing `libgpiod` (`apt-get install libgpiod2`) and the missing dependency `sysv-ipc` (as mentioned in https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/33#issuecomment-774405577) enables the use of `pulseio`.

ryanw, 2022-04-24:
I installed `sysv-ipc` but it didn't seem to make any difference for me.

ricdijk, 2022-06-08:
I understood the old library used a retry function, so wrote a simple retry function one myself. Seems to do the job ok;

```
succes=False
tries=2
while (succes == False and tries > 0):
	try:
		temperature = DHT_SENSOR.temperature
		humidity    = DHT_SENSOR.humidity
		succes=True
	except:
		tries-=1
		print ('Warning: Adafruit read error, remaining tries:', tries)
		if (tries<1):
			raise
	
```

If after al tries still no result, you can raise am error (if (tries<1)

olastor, 2022-09-25:
It seems like adding the `use_pulseio=True` option (`adafruit_dht.DHT22(board.D3, use_pulseio=True)`) fixed it for me on a pi zero w.

bananenfisch, 2024-02-21:
I have adopted the script from @ricdijk, because the loop stops after the 2nd try (it doesn't raise an exception, and the values are type none):

`
#!/usr/bin/env python3
import board
import adafruit_dht

succes=False
tries=30
while (succes == False and tries > 0):
  print('run', tries)
  try:
    DHT_SENSOR = adafruit_dht.DHT22(board.D4)
    temperature = DHT_SENSOR.temperature
    humidity    = DHT_SENSOR.humidity
    if temperature is None:
      raise
    succes=True
  except:
    tries-=1
    print ('Warning: Adafruit read error, remaining tries:', tries)
    DHT_SENSOR.exit()

DHT_SENSOR.exit()
print('Temparature: ', temperature)
print('Humidity; ', humidity)

`

christianstehning, 2024-03-29:
Two years late, still ran into the same problem. Following olastor's suggestion and using "use_pulseio = True" (i.s.o False, as stated in most manuals) worked for me.

shawnerz, 2024-03-29:
I've switched temp sensors, so it's no longer an issue for me. But, in my next project, I'll add it back in. I like being able to track humidity.Thanks,-Shawn
Yahoo Mail: Search, Organize, Conquer