adafruit/Adafruit_CircuitPython_DHT #38: DHT22 returns value acquired at previous reading, not the current value

JordyVlassembrouck, 2020-03-20:
**Board**
Raspberry Pi 3 Model B v1.2

**Sensor**
DHT22

**Context**
I run a Python script which reads the temperature and humidity from my DHT22 sensor and stores it in a .csv file.
The script is scheduled as a cronjob to run every 5 minutes.

**Problem**
I noticed that the value returned by the DHT22 is the value it actually acquired when taking the previous reading.
I know the DHT22 has a sampling rate of 0.5 Hz, which means it can only take a new reading every 2 seconds.
I trigger the script myself and wait more than 2 seconds between runs, so this is not the issue.

My proof for this issue is the following scenario:
1. I take a reading under normal conditions => humidity level is normal.
2. I breath directly onto the sensor to increase the humidity. (should spike towards 99.9%)
3. I take a reading immediately after increasing the humidity => humidity level is normal.
4. I take another reading, 3 seconds after the previous reading => humidity level spiked to 99.9%

I suspect "reading" a value actually:
- returns the "cached value" from the previous "reading"
- takes a new reading and stores this in the "cache"

I also found a topic of 2 people with the same issue:
http://forum.freetronics.com.au/viewtopic.php?f=15&t=6078&sid=a09aaf19f3fd51f2e25896949365c5d8

jfdona23, 2020-06-16:
This is likely, for some reason, the sensor is being read more than once per every two seconds, therefore it returns that "cached" value. It could be some issue in the library internals maybe.

I've validated the issue is present in both libraries, the old one and the new one.
Old: https://github.com/adafruit/Adafruit_Python_DHT
New: https://github.com/adafruit/Adafruit_CircuitPython_DHT

A good workaround is to perform a "warm up" call and then sleep for 2 or 3 seconds and call the sensor again to store the real data.

dherrada, 2020-07-16:
@JordyVlassembrouck The datasheet specifies that this sensor takes 2 seconds to collect a new reading or something along those lines, their wording wasn't very specific as to what exactly takes 2 seconds, but they were clear that you need to wait 2 seconds to get a new reading from the sensor.

closed 2020-07-16 by dherrada