adafruit/Adafruit_CircuitPython_DHT #40: DHT22 report null after a while

beneton2003, 2020-03-23:
**BUG DESCRIPTION**
Hello *. I have a problem with my DHT22. After some time (hours, days) I get null values from sensor. So I attached 2 sensors to my raspberry pi via GPIO2, 27.

The results are confusing:
If one sensors show null, one other may keep reporting good values!
Reboot of rpi does not help at all.
Hard reset (Power off / on) of rpi always works.
Power reset (3.3-5V off / on) of a sensor only SOMETIMES works.
Data reset (pull cable and replug) of sensor works almost always.
If data reset (on ONE sensor) works, other null sensor will start reporting good values as well!

**TO REPRODUCE**
Nothing it happens all time, works after reboot for a minute then it starts to get null more and more null. After some hours or so only null values.

**EXPECTED BEHAVIOR**
Get both values.

**THE SAME ISSUES**
I had found the same issue on github
https://github.com/arendst/Tasmota/issues/5619
Those guiys was make some change in reading algorythm, i was try to make the same changes, but i found that i can't make any changes to 
/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein
i'm not sure if i am going right way.
I think those guys find a way out
https://github.com/arendst/Tasmota/pull/7440

beneton2003, 2020-03-23:
If I add a print(pulses) around line 178 (file /usr/local/lib/python3.7/dist-packages/adafruit_dht.py), I see:
[image]

beneton2003, 2020-03-23:
I was also try to uncomment debug flag in /usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py

[image]

as we can see libgpiod_pulsein return empty value, i am not sure why is this happens, as i see the commands make right
 ['/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein', '--pulses', '81', '--queue', '10318', '-i', 'gpiochip0', '2']

['/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein', '--pulses', '81', '--queue', '20924', '-i', 'gpiochip0', '27']

I was trying to execute them in bash (without brakets and others), but nothing happens

jnt2007, 2020-04-16:
What brand of sensors are you using? Asair or Aosong?

beneton2003, 2020-04-16:
I use both asair and aosong, but asair have a troubles, should i change it with aosong?

jnt2007, 2020-04-16:
I have problems with Asair too. So I changed it to Aosong and problem not reproduced more than one month

kattni, 2020-05-04:
@beneton2003 @jnt2007 It appears that Adafruit sells the Aosong branded sensor. Where did you purchase the Asair sensor?

jnt2007, 2020-05-05:
On Ebay or Aliexpress, sent from China. Frequently Asair sensor has a little bit lowest price.

mryshakow, 2020-05-23:
I have the same issue with both Asair and Aosong... only reconnecting the sensor will bring it back alive. This is super annoying. Can I provide more information to resolve this issue?

RPi3
DHT22 connected via 1,5m cable
DHT22 connected to 5V
Pull Up Resistor 10K

dherrada, 2020-07-14:
> I can determine that the false behaviour only occurs (Not every time, not all sensor at that time) when another GPIO is switched... In my case its a relay on a relay board that is controlled via Pi4J in an indipendand micro service.

@mryshakow Good job noticing that. That should make it much easier to solve this.

dherrada, 2020-07-14:
@mryshakow Are you using the configuration that needs pulseio?

dherrada, 2020-07-14:
@mryshakow I wasn't able to fully reproduce the error, but I think I may have found the culprit.

Can you comment out lines 37-42 of adafruit_dht.py?

It should look something like this when you're done
[image]

Then, make sure it's setup like this. That way will do bit-banging instead of using pulseio. Basically, just hook up the resistor like in the picture, take the power to 3.3v, the ground to ground, and the data wire to whichever gpio pin you've been using.
[image]

Let me know if that works.

foarsitter, 2020-11-23:
> I can determine that the false behaviour only occurs (Not every time, not all sensor at that time) when another GPIO is switched... In my case its a relay on a relay board that is controlled via Pi4J in an indipendand micro service.

Experiencing the same with a relay and an AM2302. Can a "back electromotive force" be the problem: https://raspberrypi.stackexchange.com/questions/44060/dht22-temp-sensor-randomly-dropping/44109#44109?