adafruit/Adafruit_CircuitPython_DHT #42: 100% CPU usage because of PulseIO

pepijndevos, 2020-04-09:
On my Raspberry Pi, I'm seeing 100% CPU usage from PulseIO.

The code mentions that it's using a continuously running process because "Linux is sluggish"

https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/b770bb45adcbd7a395fd2e09f323f0e17dd32530/adafruit_dht.py#L66-L67

However, this sensor is only read once every two seconds, so it seems a bit of sluggishness is in many cases preferable to a continuous 100% CPU usage.

Is using a context something that could be reconsidered? Am I missing something that makes this unworkable?

tannewt, 2020-04-09:
See https://github.com/adafruit/Adafruit_Blinka/issues/210 for more discussion.

closed 2020-04-09 by tannewt

pepijndevos, 2020-04-10:
Thanks for pointing that out.

Regardless of the underlying issue, using a context here works for me. So I'll be running a patched version until the underlying issue is resolved.