adafruit/Adafruit_CircuitPython_DHT #43: ValueError: Object has been deinitialized and can no longer be used.

iot-crazy, 2020-04-17:
I'm relatively new to circuitpython, only started yesterday so apologies for any mistake I've made here.

Device: Teensy 4.0
DHT22 connected to pin D15
Powered from 3v3
PULL up resistor on data pin, actually the same circuit I've used successfully on other devices, just moved to the teensy 4.0
Coding on a macbook pro using Mu 1.0.3
adafruit_dht library copied to /lib from the latest library download: adafruit-circuitpython-bundle-5.x-mpy-20200417
No other modules in the lib folder

Code:

```
import board
import time
import adafruit_dht

print('Initialising...')
dht_device = adafruit_dht.DHT22(board.D15)
time.sleep(2)
print(dht_device)

while True:
    temperature = dht_device.temperature
    humidity = dht_device.humidity
    print(str(temperature))
    print(str(humidity))
    time.sleep(2)
```

Full serial output:

code.py output:
Initialising...
<DHT22 object at 20209020>
Traceback (most recent call last):
  File "code.py", line 12, in <module>
  File "adafruit_dht.py", line 242, in temperature
  File "adafruit_dht.py", line 187, in measure
  File "adafruit_dht.py", line 117, in _get_pulses_pulseio
ValueError: Object has been deinitialized and can no longer be used. Create a new object.

iot-crazy, 2020-04-17:
Text from boot_out.txt

Adafruit CircuitPython 5.2.0 on 2020-04-09; Teensy 4.0 with IMXRT1062DVJ6A

tannewt, 2020-04-17:
@hierophect Want to take a look at this? It's on iMX RT.

hierophect, 2020-04-17:
I'll check it out this week

liammcirl, 2020-06-11:
Seeing same issue on Teensy 4.0 board calling sonar = adafruit_hcsr04.HCSR04(trigger_pin=TRIG_PIN, echo_pin=ECHO_PIN, timeout=1.0)

Anybody made any progress with this? Thanks