adafruit/Adafruit_CircuitPython_DHT #44: FileNotFoundError: (...) /usr/local/lib/python3.8/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein

rsnitsch, 2020-05-05:
I am running **Ubuntu Server 20.04 LTS** on a **Raspberry Pi 3 B**. I have a **DHT11** sensor module.

I am following the instructions from https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/python-setup but the code does not work for me.

I have successfully executed these commands beforehand:

```
sudo apt-get install libgpiod2
sudo pip3 install adafruit-circuitpython-dht
```

This is my code:

```python
# ky015.py
import time
import board
import adafruit_dht

# Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT11(board.D3)

while True:
    try:
        # Print the values to the serial port
        temperature_c = dhtDevice.temperature
        temperature_f = temperature_c * (9 / 5) + 32
        humidity = dhtDevice.humidity
        print(
            "Temp: {:.1f} F / {:.1f} C    Humidity: {}% ".format(
                temperature_f, temperature_c, humidity
            )
        )

    except RuntimeError as error:
        # Errors happen fairly often, DHT's are hard to read, just keep going
        print(error.args[0])

    time.sleep(2.0)
```

This is the error I get:

```
ubuntu@ubuntu:~/ky015$ sudo python3 ky015.py
Traceback (most recent call last):
  File "ky015.py", line 6, in <module>
    dhtDevice = adafruit_dht.DHT11(board.D3)
  File "/usr/local/lib/python3.8/dist-packages/adafruit_dht.py", line 263, in __init__
    super().__init__(True, pin, 18000)
  File "/usr/local/lib/python3.8/dist-packages/adafruit_dht.py", line 69, in __init__
    self.pulse_in = PulseIn(self._pin, 81, True)
  File "/usr/local/lib/python3.8/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 65, in __init__
    self._process = subprocess.Popen(cmd)
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein'
```

I have also followed the instructions for "Setup your Linux Board for using CircuitPython Libraries". The script blinkatest.py completes successfully (Link to code: https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#blinka-test-3-15). Output:

```
ubuntu@ubuntu:~/ky015$ sudo python3 blinkatest.py
Hello blinka!
Digital IO ok!
I2C ok!
SPI ok!
done!
```

The file libgpiod_pulsein does exist. The Python code seems to try and run it with subprocess.Popen. That's strange IMHO since the "lib" prefix does imply that it is a library, not an executable, right? On the other hand, a library should have a suffix of ".so". I am dumbfounded by that file name. By the way, this is ldd output for the file:

```
ubuntu@ubuntu:/usr/local/lib/python3.8/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio$ ldd libgpiod_pulsein
        not a dynamic executable
```

This is the stat output:

```
ubuntu@ubuntu:/usr/local/lib/python3.8/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio$ stat libgpiod_pulsein
  File: libgpiod_pulsein
  Size: 19752           Blocks: 40         IO Block: 4096   regular file
Device: b302h/45826d    Inode: 391704      Links: 1
Access: (0755/-rwxr-xr-x)  Uid: (    0/    root)   Gid: (   50/   staff)
Access: 2020-05-05 10:00:24.194643069 +0000
Modify: 2020-05-05 10:00:10.230685234 +0000
Change: 2020-05-05 10:00:10.990682910 +0000
 Birth: -
```

I have tried to reinstall the packages, but it did not fix my problem.

JohnWickham, 2020-05-08:
Similar issue here; works fine on a Raspberry Pi 4, but fails on a Zero W. The blinka library is installed and passes the test script, but using adafruit_dht, I get:
`
RuntimeError: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

gsly, 2020-06-02:
Similar issue here on Raspberry Pi B+ v1.2 on an up to date Raspbian (buster 10) OS. Test script shows all ok but running the dht_simpletest.py configured for GPIO pin 4 I get:

`Traceback (most recent call last):
  File "dht_simpletest.py", line 6, in <module>
    dhtDevice = adafruit_dht.DHT22(board.D4)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_dht.py", line 273, in __init__
    super().__init__(False, pin, 1000)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_dht.py", line 69, in __init__
    self.pulse_in = PulseIn(self._pin, 81, True)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 71, in __init__
    message = self._wait_receive_msg()
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 89, in _wait_receive_msg
    "Timed out waiting for PulseIn message. Make sure libgpiod is installed."
RuntimeError: Timed out waiting for PulseIn message. Make sure libgpiod is installed.`

makermelissa, 2020-06-02:
Hi, it looks like you may not have libgpiod installed. Try running `sudo apt install libgpiod2`.

gsly, 2020-06-03:
> Hi, it looks like you may not have libgpiod installed. Try running `sudo apt install libgpiod2`.

That is not the problem for me:

```
# apt search libgpiod2
Sorting... Done
Full Text Search... Done
libgpiod2/stable,now 1.2-3+rpi1 armhf [installed]
  C library for interacting with Linux GPIO device - shared libraries
```

gsly, 2020-06-05:
Ok, my problem looks to be resolved and the cause for me must be some suspect wiring to the sensor. I just happened to move the unit and sensor and it started working again.

closed 2020-06-05 by ladyada

rsnitsch, 2020-06-06:
"ladyada closed this 15 hours ago"

Please reopen the issue. I have opened this issue originally. It was hijacked by other guys that encountered (and apparently solved) entirely DIFFERENT problems.

My problem is still unsolved, therefore this issue should be reopened.

ladyada, 2020-06-06:
@rsnitsch please try raspbian - in case that solves it. we have not tested with ubuntu :)

jfdona23, 2020-07-21:
@rsnitsch make sure your user is part of 'gpio' group.
If not, then run:
`sudo usermod -aG gpio <your_username>`

Then logout and login again, or just reboot the board.

rsnitsch, 2020-07-30:
I tried it again with Raspbian / Raspberry Pi OS and now it works.

Although I get:
> DHT sensor not found, check wiring

But this is another issue, of course.

goldyfruit, 2020-08-10:
I'm having the same issue, seems to be related to 64-bits. I'm running Ubuntu 20.04 on my RPi 4B, and the `libgpiod_pulsein` library with `adrafruit_blinka` Python package is 32-bits.

```
/home/ubuntu/.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=356bb870cf3f5a35b0862d12a708b782a9c365f6, not stripped
```

c4explosive, 2021-05-31:
> It seems to be related to 64-bits. I'm running Ubuntu 20.04 64-bits on my RPi 4B, and the `libgpiod_pulsein` library shipped with `adrafruit_blinka` Python package is 32-bits.
> 
> ```
> $ file ~/.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein
> /home/ubuntu/.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=356bb870cf3f5a35b0862d12a708b782a9c365f6, not stripped
> ```
> 
> So basically I had to recompile the library for `arm64` architecture, here are the steps:
> 
> ```
> $ sudo apt install libgpiod-dev git build-essential
> $ git clone https://github.com/adafruit/libgpiod_pulsein.git
> $ cd libgpiod_pulsein/src
> $ make
> $ cp libgpiod_pulsein ~/.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein
> ```

This works, thanks. Working on manjaro for raspberry PI 3B.