adafruit/Adafruit_CircuitPython_DHT #45: DHT (DHT22) not working on M0 boards

jerryneedell, 2020-06-19:
I am unable to read data from a DHT22 sensor on either a feather_m0_express or a fether_m0_rfm9x board -- using the dht_simpletest

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.4.0-beta.1-17-g6ae52ade7-dirty on 2020-06-19; Adafruit Feather M0 Express with samd21g18
>>> import dht_simpletest
A full buffer was not returned. Try again.
A full buffer was not returned. Try again.
DHT sensor not found, check wiring
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "dht_simpletest.py", line 24, in <module>
KeyboardInterrupt: 
>>> 
```

I am able to use it with a grandventral_m4_espress

jerryneedell, 2020-06-19:
also tried with 5.4.0 - beta.1 from S3  just to make sure my build was not an issue -- my build should not have made any changes to this board.

same problem

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.4.0-beta.1 on 2020-06-08; Adafruit Feather M0 Express with samd21g18
>>> import dht_simpletest
13 pulses: [382, 366, 365, 366, 226, 336, 183, 350, 198, 351, 366, 366, 264]
A full buffer was not returned. Try again.
9 pulses: [382, 608, 489, 733, 548, 222, 509, 205, 343]
DHT sensor not found, check wiring
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "dht_simpletest.py", line 22, in <module>
  File "dht_simpletest.py", line 11, in <module>
  File "adafruit_dht.py", line 242, in temperature
  File "adafruit_dht.py", line 187, in measure
  File "adafruit_dht.py", line 125, in _get_pulses_pulseio
KeyboardInterrupt: 
>>> 

````

jerryneedell, 2020-06-19:
hmmm -- reverted to 5.3.0 and it works normally ....

```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit Feather M0 Express with samd21g18
>>> import dht_simpletest
81 pulses: [53, 27, 53, 27, 54, 26, 53, 27, 53, 26, 54, 27, 52, 27, 53, 74, 66, 74, 54, 26, 54, 73, 54, 74, 54, 74, 53, 27, 53, 73, 54, 74, 66, 26, 54, 26, 54, 26, 54, 27, 53, 27, 53, 27, 53, 27, 53, 26, 66, 74, 53, 74, 55, 74, 53, 73, 54, 26, 54, 26, 53, 74, 54, 27, 62, 74, 54, 27, 53, 73, 54, 27, 54, 73, 54, 73, 54, 74, 53, 27, 45]
Temp: 75.6 F / 24.2 C    Humidity: 44.3% 
81 pulses: [54, 27, 53, 27, 53, 26, 54, 26, 54, 26, 53, 27, 53, 27, 55, 71, 67, 74, 54, 26, 53, 74, 55, 74, 52, 74, 54, 73, 54, 26, 54, 27, 65, 27, 54, 27, 52, 27, 53, 27, 54, 27, 52, 27, 53, 27, 53, 27, 65, 74, 54, 74, 53, 74, 53, 74, 54, 27, 53, 27, 53, 73, 54, 27, 62, 74, 54, 27, 53, 74, 53, 29, 51, 74, 55, 72, 54, 73, 54, 73, 47]
Temp: 75.6 F / 24.2 C    Humidity: 44.4% 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "dht_simpletest.py", line 24, in <module>
KeyboardInterrupt: 
```

dherrada, 2020-06-25:
I'll take a look at this in a few days

tannewt, 2020-06-26:
I assume this is the same as https://github.com/adafruit/circuitpython/issues/3049. Closing this in favor of the CircuitPython issue.

closed 2020-06-26 by tannewt