adafruit/Adafruit_CircuitPython_DHT #49: DHT sensor not found, check wiring

Talaxy009, 2020-09-11:
I'm using DHT11 on Rpi 4B and I turn to using Adafruit_CircuitPython_DHT today from [Adafruit_Python_DHT](https://github.com/adafruit/Adafruit_Python_DHT). I had changed my code to fit the Adafruit_CircuitPython_DHT, but when I run it I meet this error: 

```
Traceback (most recent call last):
  File "tmpv0.2.py", line 44, in <module>
    main()
  File "tmpv0.2.py", line 23, in main
    temperature = dht_device.temperature
  File "/usr/local/lib/python3.8/dist-packages/adafruit_dht.py", line 253, in temperature
    self.measure()
  File "/usr/local/lib/python3.8/dist-packages/adafruit_dht.py", line 205, in measure
    raise RuntimeError("DHT sensor not found, check wiring")
RuntimeError: DHT sensor not found, check wiring
```

I am sure the connection is fine, because after this error I run the former code and it work perfectly.

Here is the code that has been change to fit Adafruit_CircuitPython_DHT

```python
import datetime
import time

import adafruit_dht
from board import D4
from luma.core.interface.serial import i2c
from luma.core.render import canvas
from luma.oled.device import sh1106
from PIL import ImageFont

# OLED settings
serial = i2c(port=1, address=0x3C)
device = sh1106(serial)
font = ImageFont.truetype('simyou.ttf', 15)

# DHT11 settings
dht_device = adafruit_dht.DHT11(D4)

def main():
    while True:
        temperature = dht_device.temperature
        humidity = dht_device.humidity
        if humidity is not None and temperature is not None:
            Temperature = "温度: {0:0.1f}°C".format(temperature)
            Humidity = "湿度: {0:0.1f}%".format(humidity)
            print(Temperature, Humidity)
            for i in range(0, 10):
                now = datetime.datetime.now()
                today_time = now.strftime("%H:%M:%S")
                with canvas(device) as draw:
                    draw.text((30, 4), today_time, font=font, fill="white")
                    draw.text((10, 23), Temperature, font=font, fill="white")
                    draw.text((10, 42), Humidity, font=font, fill="white")
                time.sleep(0.2)

        else:
            print('失败')

if __name__ == "__main__":
    try:
        main()
    except KeyboardInterrupt:
        device.cleanup()
        pass

```

And here is the former code

```python
import datetime
import time

import Adafruit_DHT
from luma.core.interface.serial import i2c
from luma.core.render import canvas
from luma.oled.device import sh1106
from PIL import ImageFont

# OLED settings
serial = i2c(port=1, address=0x3C)
device = sh1106(serial)
font = ImageFont.truetype('simyou.ttf', 15)

# DHT11 settings
sensor = Adafruit_DHT.DHT11
DHT11pin = 4

def main():
    while True:
        humidity, temperature = Adafruit_DHT.read_retry(sensor, DHT11pin)
        if humidity is not None and temperature is not None:
            Temperature = "温度: {0:0.1f}°C".format(temperature)
            Humidity = "湿度: {0:0.1f}%".format(humidity)
            print(Temperature, Humidity)
            for i in range(0, 10):
                now = datetime.datetime.now()
                today_time = now.strftime("%H:%M:%S")
                with canvas(device) as draw:
                    draw.text((30, 4), today_time, font=font, fill="white")
                    draw.text((10, 23), Temperature, font=font, fill="white")
                    draw.text((10, 42), Humidity, font=font, fill="white")
                time.sleep(0.2)

        else:
            print('失败')

if __name__ == "__main__":
    try:
        main()
    except KeyboardInterrupt:
        device.cleanup()
        pass

```

I am new to python and Rpi, and thanks for any help!

Psychi1, 2020-11-30:
Hi, 

i'm facing the same problem.

This does not run ....
```
ubuntu@ubuntu:~/PiTemp$ cat dev/simpletest.py 
#!/usr/bin/python3

import time
import board
import adafruit_dht

# Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT22(board.D18)

# you can pass DHT22 use_pulseio=False if you wouldn't like to use pulseio.
# This may be necessary on a Linux single board computer like the Raspberry Pi,
# but it will not work in CircuitPython.
#dhtDevice = adafruit_dht.DHT22(board.D18, use_pulseio=False)

while True:
    try:
        # Print the values to the serial port
        temperature_c = dhtDevice.temperature
        temperature_f = temperature_c * (9 / 5) + 32
        humidity = dhtDevice.humidity
        print(
            "Temp: {:.1f} F / {:.1f} C    Humidity: {}% ".format(
            temperature_f, temperature_c, humidity
        ))

    except RuntimeError as error:
        # Errors happen fairly often, DHT's are hard to read, just keep going
        print(error.args[0])
        time.sleep(2.0)
        continue
    except Exception as error:
        dhtDevice.exit()
        raise error

    time.sleep(10.0)
```
which end up in "DHT sensor not found, check wiring"
```
$ sudo python3 simpletest.py 
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
received SIGINT

Traceback (most recent call last):
  File "simpletest.py", line 18, in <module>
^C    temperature_c = dhtDevice.temperature
  File "/usr/local/lib/python3.8/dist-packages/adafruit_dht.py", line 253, in temperature
    self.measure()
  File "/usr/local/lib/python3.8/dist-packages/adafruit_dht.py", line 205, in measure
    raise RuntimeError("DHT sensor not found, check wiring")
RuntimeError: DHT sensor not found, check wiring
```
while

```
$ cat PiTemp.py 
__version__= "0.1"
import sys
if sys.version_info < (3, 0):
    sys.stdout.write("Sorry, requires Python 3.x, not Python 2.x\n")
    sys.exit(1)

import time
import Adafruit_DHT

DELAY_SECONDS=15          # 15 Sec

DHT_SENSOR=Adafruit_DHT.DHT22
DHT_PIN=18

if __name__ == "__main__":
    print("Um das Programm anzubrechen, drücke STRG+C ...")

    while True:
        try:
            # Print the values to the serial port
            humidity, temp = Adafruit_DHT.read_retry(DHT_SENSOR, DHT_PIN)
            print("Temp: {0:0.1f} °C    Humidity: {1:0.1f} % ".format(temp, humidity))
        except RuntimeError as error:
            print(error.args[0])
            time.sleep(5.0) # In case of error retry after 5.0 seconds
            continue
        time.sleep(DELAY_SECONDS)
```
works like a charm.

```
$ sudo python3 PiTemp.py 
Um das Programm anzubrechen, drücke STRG+C ...
Temp: 20.8 °C    Humidity: 56.9 % 
Temp: 20.8 °C    Humidity: 56.9 % 
```

Any ideas?

jayceslesar, 2020-11-30:
Same issue except my GPIO pin is 4.

makermelissa, 2020-11-30:
Can you try GPIO 18? That's Pin 12 on the 40-pin header. That one has Hardware-based PWM, so it's usually good for testing. If it does work with GPIO 18, then the issue may be that it isn't working with software PWM.

jayceslesar, 2020-12-01:
> Can you try GPIO 18? That's Pin 12 on the 40-pin header. That one has Hardware-based PWM, so it's usually good for testing. If it does work with GPIO 18, then the issue may be that it isn't working with software PWM.

Doesn't work on GPIO 18 either, I'm assuming the sensor is just faulty at this point and ordered a different brand of sensor.

Talaxy009, 2020-12-03:
> Same issue except my GPIO pin is 4. At this point I don't think the DHT11 even works with an RPi4...any solutions would be nice

Try this one -> [Adafruit_Python_DHT](https://github.com/adafruit/Adafruit_Python_DHT)? DHT11 work well with my RPi4 when I use this library.

Psychi1, 2020-12-03:
> > Same issue except my GPIO pin is 4. At this point I don't think the DHT11 even works with an RPi4...any solutions would be nice
> 
> Try this one -> [Adafruit_Python_DHT](https://github.com/adafruit/Adafruit_Python_DHT)? DHT11 work well with my RPi4 when I use this library.

I think it wrong to solve a problem by using a deprecated libraray which links to this (CircuitPython) library. However it's a dirty but working hack.

etokheim, 2020-12-06:
I'm having the same issue. However, I can't get the deprecated library to work either... First I got an error due to the PI 4 not being supported, which I fixed with this [dirty hack](https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/49).

I no longer get errors, but I'm stuck with `None` values from the sensor... Any tips?

I really hope someone comes up with a solution!

alaub81, 2020-12-29:
I get the same error on a raspberry pi 3 B. On my Raspberry Pi 4 with the same wiring everything works perfect with the circuit python library. On my Pi 3 I have to use the old deprecated one, but that works!

dgriswo, 2021-01-09:
I have this issue on a FeatherS2
Running: Adafruit CircuitPython 6.0.1 on 2020-12-28; FeatherS2 with ESP32S2

Error is:
`Traceback (most recent call last):
  File "code.py", line 12, in <module>
  File "adafruit_dht.py", line 261, in temperature
  File "adafruit_dht.py", line 213, in measure
RuntimeError: DHT sensor not found, check wiring`

Running: Adafruit CircuitPython 6.1.0-rc.0 on 2021-01-06; FeatherS2 with ESP32S2

Error is:
`Traceback (most recent call last):
  File "code.py", line 12, in <module>
  File "adafruit_dht.py", line 261, in temperature
  File "adafruit_dht.py", line 245, in measure
RuntimeError: Checksum did not validate. Try again.`

bsolino, 2021-02-17:
I've been struggling with this issue too. I know the wiring is right because it works sometimes. However, I'm trying now a DHT22 and it is extremely temperamental, sometimes taking several minutes before correct readings. I've also encountered a few instances of "A full buffer was not returned. Try again.". I have modified the code to actually show the number of pulses received, and oddly enough it usually doesn't receive any pulses.

As an example, I have this code that checks the sensor every two seconds. I'm using here a DHT22 sensor on a RaspberryPi Zero W, pin 4, and using the pulseio method. I've also tried pin 18 (as recommended above), with similar results.

```
$ ./run_22.sh
2021-02-16 23:41:16.472722 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:18.746235 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:21.006975 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:23.266578 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:25.526662 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:23.266578 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:25.526662 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:27.787561 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:30.056836 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:32.317642 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:34.586726 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:36.846722      Temp: 11.5 C    Humidity: 59.9 %
2021-02-16 23:41:39.551465 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:41.826696 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:44.095804 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:46.356680 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:48.627296 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:50.887131 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:53.146693 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:55.406092 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:57.667552 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:41:59.936223 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:02.196585 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:04.456705 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:06.719312 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:08.986958 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:11.246717 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:13.506496      Temp: 11.5 C    Humidity: 60.4 %
2021-02-16 23:42:16.195411 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:18.466762 A full buffer was not returned. Try again. 79 pulses.
2021-02-16 23:42:20.764793 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:23.036675 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:25.303123 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:27.566617 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:29.827433 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:32.086793 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:34.346725 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:36.606720 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:38.937387 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:41.194744 A full buffer was not returned. Try again. 17 pulses.
2021-02-16 23:42:43.465054 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:45.737242 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:47.996850 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:50.253569 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:52.513441 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:54.773724 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:52.513441 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:54.773724 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:57.033494 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:42:59.293654 A full buffer was not returned. Try again. 31 pulses.
2021-02-16 23:43:01.568428 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:03.843388 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:06.103259 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:08.364260 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:10.623567 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:12.896756 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:15.153514 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:17.413947 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:19.673497 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:21.933526 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:24.203646 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:26.463453 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:28.732867 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:30.996279 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:43:33.253776 DHT sensor not found, check wiring. 0 pulses.
[... cut ...]
2021-02-16 23:44:00.376169 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:44:02.633598 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:44:04.943199 DHT sensor not found, check wiring. 0 pulses.
2021-02-16 23:44:07.197233  Temp: 11.5 C    Humidity: 60.6 %
```

I used to have a DHT11 sensor, and that one also had this issue often, but not as frequently. I could usually get 5 correct readings in less than 1 minute. Unfortunately, it seems I've burned it so I can't do tests with it anymore.

tannewt, 2021-02-17:
Please try the latest version from the repo. #62 just refined the bitbang version and should improve reliability.

bsolino, 2021-02-17:
> Please try the latest version from the repo. #62 just refined the bitbang version and should improve reliability.

Unfortunately, that only changes it for the worse in my case. The bug should not have affected me (I was already using a DHT22 sensor), and using bitbang seems completely unable to return a result. In ~10 minutes, not a single pulse was seen.

jposada202020, 2021-02-17:
@bsolino Hello, I did a little investigation into this, you can follow my results here https://github.com/jposada202020/DHT_Investigation
But I have never been able to read a DHT sensor in a Raspberry PI Zero. I tried both the DHT11 and the DHT22. I saw that you followed the recommendation to not to use Pin 4. In my tests I used pin D17. However, I can read in both sensor in the RP3/4 do you happen to have one in hand?

bsolino, 2021-02-17:
@jposada202020 I have also tried pins 4 and 17 with similar results (works but very infrequently), and both DHT11 and DHT22.

I have done a quick test (just 3 minutes) on a Raspberry Pi 3 and I couldn't get a reading at all, always the "Check wiring" error.

This is very odd, I will have to do a check at some point with different wires, or testing more ports on the Pi3. But it's weird that it's so inconsistent.

jposada202020, 2021-02-17:
@bsolino Agree with this kind of sensor everything could go wrong, and is very difficult to troubleshoot. These Sensor are very time dependent, and this is something that the RP is not very good at.  At least in my experience. But the RP3 is indeed strange both have worked in my case, inconsistent but worked.

john0sweet, 2021-02-20:
FWIW I'm using the below code that I set up last week (13 Feb) and it works perfectly with a DHT22 connected to pin 4. Swap out the 22 for an 11 and change the dhtDevice setting to DHT11 and it gives me DHT sensor not found (same wiring - literally just changed out the 22 for an 11). I've tried it with 4 different DHT11's and 2 DHT22's - the 22's both work fine (about 50/50 valid reading + error messages various) but the 11's don't work at all.

import time
import board
import adafruit_dht

#Initialise the dht device with the data pin connected to
# dhtDevice = adafruit_dht.dht22(board.D4)

# you can pass DHT 22 use_pulseio=False if you don't want to use pulseio
# this may be necessary on the Pi but will not work in
# circuit python

dhtDevice = adafruit_dht.DHT22(board.D4, use_pulseio=False)

while True:
    try:
        #print the values to the serial port
        temperature_c = dhtDevice.temperature
        temperature_f = temperature_c * (9/5) + 32
        humidity = dhtDevice.humidity
        print(
            "Temp: {:.1f}F  / {:.1f} C Humidity: {}%".format(
                temperature_f, temperature_c, humidity
                )
        )
    except RuntimeError as error:
        # Expect DHTs to produce regular errors
        print (error.args[0])
        time.sleep(2.0)
        continue
    except Exception as error:
        dhtDevice.exit()
        raise error

    time.sleep(2.0)

from https://tutorials-raspberrypi.com/raspberry-pi-measure-humidity-temperature-dht11-dht22/

jposada202020, 2021-02-20:
@john0sweet Are you using the new library there was a change in the library last week, could you try it and let us know, the change was to improve the trigger time int he DHT11 sensors. It would be helpful it you could try it. Thanks for your comments

john0sweet, 2021-02-20:
Pretty sure I'm on the new one - ran a pip update thisavo on it and ran the code again - same result. Still works fine with the 22 on any of the normal pins, but the 11s don't work on any pin - left one running for 5 minutes just to be sure. No valid data.

jposada202020, 2021-02-20:
@john0sweet interesting, do you know if this happens with this library only. have you test the DHT11 with other logic or board? As these DHT11 using a Pull-up resistor

alaub81, 2021-02-21:
On my side (DHT22 with latest pip3 install of the dht library), the thing with `use_pulseio=False` did the trick on my Raspberry Pi 3. But with that I got so many `Checksum did not validate. Try again.`and `A full buffer was not returned. Try again.` On my Pi Zeros no chance to get it work. So I will stay on the deprecated Library, that one still works so good on Pi zero and Pi 3.

Tob1as, 2021-02-25:
I make the same observation:
Since update from version 3.5.3/3.5.4 to 3.5.6 I also get the error `DHT sensor not found, check wiring`.  
With version 3.5.5 it works, but now I get the following messages more often: `Checksum did not validate. Try again.` and `A full buffer was not returned. Try again.` . I measure every 15sec with dht22 on pin4 and Pi 3.

jposada202020, 2021-02-25:
@Tob1as I was reviewing the changes from 3.5.3 I could see anything that badly affect the behavior. The revision introduced in 3.5.6 would improve the readings in the DHT11 sensor. It would not affect to the best of my knowledge the DHT22 readings. I just tested the new library with a DHT22 in pin D17 in a RP3. With some missing reading, but I got 15/20 readings taking the measure every 5 seconds.

jposada202020, 2021-02-25:
@Tob1as I review the PR, and I was wrong. Sorry about that. The intended change in 3.5.6 was to addresses mentioned the trigger time for DHT11. However in doing so, I used the '//' division instead of the / division, given a 0 every time. I have just submitted a new PR #64  to correct this. Thank you for pointing that out.

Tob1as, 2021-02-26:
@jposada202020 No problem, thanks for the fix 👍  
I have [build](https://github.com/Tob1as/rpi-sensors/blob/master/testing.armhf.python.Dockerfile) from git, delete one of the / division and now it works. :-)

jposada202020, 2021-02-26:
@Tob1as Thanks, Changes are merged, they will be released soon!, and thanks fr the feeback and testing on the changes (y)

Talaxy009, 2022-03-12:
Sorry, I forgot I had submitted this issue, and since I don't have the hardware to test whether the new version works and it looks like someone can already use it, I decided to close this issue, if you encounter the same problem you can open an issue yourself.

closed 2022-03-12 by Talaxy009

CesMak, 2022-03-12:
I still have this problem: (installed adafruit lib with pip3) on my **raspberry pi ZERO WH** using DHT22 sensor
```python
import adafruit_dht
import board
import time

dht_device = adafruit_dht.DHT22(board.D4)
sleep_secs = 2

while True:
	try:
		temp = dht_device.temperature
		hum  = dht_device.humidity
		print(temp, hum) 
	except Exception as e:
		print(e)
	time.sleep(sleep_secs)

```

Output:
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
22.7 52.9
DHT sensor not found, check wiring
22.6 53.2
22.6 53.2
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
22.6 53.2
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
22.5 53.3
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring

**on pin18 I only get DHT sensor not found.** 

Any hints for me?

caternuson, 2022-03-12:
https://learn.adafruit.com/modern-replacements-for-dht11-dht22-sensors/overview

CesMak, 2022-03-12:
> https://learn.adafruit.com/modern-replacements-for-dht11-dht22-sensors/overview

Well this is not an answer to my question cause with the old lib all works (except for negative values -> That is why I considered switching at all....)

CesMak, 2022-03-15:
So I have a new problem:
After 30min I get no more data: (befor all is fine) - I have to reboot than it works again. 
This lib is completely useless. I do not understand why it is not working I think I switch back to the original dht_lib - its much older but at least it works most of the time. 

```
2022-03-14 19:50:17 DateTime	Entry	Power	Temperature_outside	Humidity_outside	Temperature_inside	Humidity_inside	CorrectReadingsIn	CorrectReadingsOut
2022-03-14 19:52:07 1	9.07	65.37	15.81	59.57	27	24
2022-03-14 19:53:56 2	8.84	65.66	15.8	59.53	21	20
2022-03-14 19:55:46 3	8.58	66.29	15.73	59.59	24	20
2022-03-14 19:57:35 4	8.45	66.98	15.58	59.71	27	22
2022-03-14 19:59:25 5	8.47	67.27	15.47	59.94	25	24
2022-03-14 20:01:14 6	8.57	67.56	15.4	60.3	25	23
2022-03-14 20:03:04 7	8.86	66.88	15.34	60.5	26	21
2022-03-14 20:04:53 8	9.01	66.15	15.3	60.65	20	17
2022-03-14 20:06:43 9	9.19	65.4	15.22	60.74	22	24
2022-03-14 20:08:33 10	8.97	66.04	15.2	60.9	21	18
2022-03-14 20:10:22 11	8.76	66.28	15.11	61.08	26	22
2022-03-14 20:12:12 12	8.67	66.36	15.08	61.14	24	21
2022-03-14 20:14:01 13	8.6	66.0	15.05	61.08	20	21
2022-03-14 20:15:49 14	8.32	67.26	15.0	61.31	22	15
2022-03-14 20:17:20 15	8.2	68.0	15.0	61.3	1	1
2022-03-14 20:18:51 16	8.2	68.0	15.0	61.3	1	1
2022-03-14 20:20:22 17	8.2	68.0	15.0	61.3	1	1
2022-03-14 20:21:53 18	8.2	68.0	15.0	61.3	1	1
```

cloudlabmp, 2024-01-26:
Same problem here - except I am running the RPi 5, which as I'm finding out, is not compatible with many of the python modules written and used with older gen Pi's such as 4, and 3.

Currently trying to get the DHT11 sensor running with the Adafruit_CircuitPython_DHT module, but have yet to get any output other than: "DHT sensor not found, check wiring"

I tried two different DHT11 sensors, each a different brand, both display the same results.
I've tried wiring directly from sensor to the GPIO pins, and via a GPIO ext and breadboard - results the same.

Anyone actually managed to get a reading using a Raspberry pi 5?

vanerpr, 2025-01-19:
I understand that is this a post from last year but did anyone figure out the issue on a Pi5?

TheRaven500, 2025-01-19:
Maybe it works this way (see my post at the end):
https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/33