adafruit/Adafruit_CircuitPython_DHT #52: OverflowError in adafruit_dht.py line 137: pulses.append(self.pulse_in.popleft())

michaellass, 2020-10-28:
_Originally reported by @darton in https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/50#issuecomment-717815696_

```
Exception in thread <function get_dht_data at 0x766998a0>:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "./rpims.py", line 221, in get_dht_data
    temperature = dhtDevice.temperature
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_dht.py", line 253, in temperature
    self.measure()
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_dht.py", line 198, in measure
    pulses = self._get_pulses_pulseio()
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_dht.py", line 137, in _get_pulses_pulseio
    pulses.append(self.pulse_in.popleft())
OverflowError: unsigned short is greater than maximum
```

@darton: Was this a one-time error or does it occur regularly? Was your script afterwards  able to get valid readings?

My current suspicion:
* A transfer error between sensor and raspberry pi causes one pulse be recorded as longer than 65535 µs by libgpiod_pulsein
* In https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py#L142 this value is received as UTF-8 string and converted into an int
* In https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/master/adafruit_dht.py#L137 we attempt to put that value into an array that can only hold uint16, i.e., numbers between 0 and 65535.

Proposed fix: Check the value received from `self.pulse_in.popleft()` and return an empty array in case any of the pulse lengths is out of bounds.

darton, 2020-10-29:
When the error occurs, the script is unable to obtain a read.

michaellass, 2020-10-29:
That's expected. The question is: what happens on following readings? Do they work properly again, do they raise the same exception again or does something else happen?

darton, 2020-10-29:
The script stops working.

 strace -f -p $(pidof libgpiod_pulsein)
strace: Process 10594 attached with 2 threads
[pid 10595] futex(0x24084, FUTEX_WAIT_PRIVATE, 2, NULL <unfinished ...>
[pid 10594] msgrcv(90,

michaellass, 2020-10-29:
Could you please post your python script here?

darton, 2020-10-30:
def get_dht_data():
    import adafruit_dht
    from time import sleep

    read_interval = 5
    pin = 17
    dhtDevice = adafruit_dht.DHT22(pin)
    while True:
        try:
            temperature = dhtDevice.temperature
            humidity = dhtDevice.humidity
            print(" Temperature: {:.1f}°C ".format(temperature))
            print(" Humidity: {}% ".format(humidity))
            print("")
        except RuntimeError as error:
            print(" DHT - " + str(error))
        sleep(read_interval)

michaellass, 2020-10-30:
Unless the call of `get_dht_data()` is in a `try` block as well, I would expect this script to just quit on the shown Exception and with that also `libgpiod_pulsein` should terminate.

Anyway, instead of throwing the `OverflowError`, adafruit_dht should probably throw a `RuntimeError` like on every other sensor read error.

For now you should be able to deal with this by just adding another except block like the follwoing:
```
        except OverflowError as error:
            print(" DHT - " + str(error))
```

darton, 2020-10-30:
When handling all exceptions, reading stops completely when an exception occurs " DHT sensor not found, check wiring"

 strace -f -p $(pidof libgpiod_pulsein)
strace: Process 1835 attached with 2 threads
[pid  1836] futex(0x24084, FUTEX_WAIT_PRIVATE, 2, NULL <unfinished ...>
[pid  1835] msgrcv(8, {1, "c\000000\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"...}, 4095, 1, 0) = 1
[pid  1835] msgrcv(8, {1, "t1000\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"...}, 4095, 1, 0) = 5
[pid  1835] futex(0x24084, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
[pid  1836] <... futex resumed> )       = 0
[pid  1835] <... futex resumed> )       = 1
[pid  1836] gettimeofday({tv_sec=1604083673, tv_usec=206717}, NULL) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] futex(0x24084, FUTEX_WAKE_PRIVATE, 1) = 0
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=207143}, NULL) = 0
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL, 0xb6dd6c9c) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] gettimeofday({tv_sec=1604083673, tv_usec=207801}, NULL) = 0
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=207848}, NULL) = 0
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL, 0xb6dd6c9c) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] gettimeofday({tv_sec=1604083673, tv_usec=208488}, NULL) = 0
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=208668}, NULL) = 0
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL, 0xb6dd6c9c) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] gettimeofday({tv_sec=1604083673, tv_usec=209396}, NULL) = 0
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=209443}, NULL) = 0
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL, 0xb6dd6c9c) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] gettimeofday( <unfinished ...>
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=210074}, NULL) = 0
[pid  1836] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=210271}, NULL) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL <unfinished ...>
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=210781}, NULL) = 0
[pid  1836] <... ioctl resumed> , 0xb6dd6c9c) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] gettimeofday( <unfinished ...>
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=211605}, NULL) = 0
[pid  1836] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=211733}, NULL) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL <unfinished ...>
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=212018}, NULL) = 0
[pid  1835] gettimeofday({tv_sec=1604083673, tv_usec=212340}, NULL) = 0
[pid  1835] gettimeofday({tv_sec=1604083673, tv_usec=212659}, NULL) = 0
[pid  1835] gettimeofday({tv_sec=1604083673, tv_usec=212866}, NULL) = 0
[pid  1835] gettimeofday({tv_sec=1604083673, tv_usec=213103}, NULL) = 0
[pid  1836] <... ioctl resumed> , 0xb6dd6c9c) = 0
[pid  1835] gettimeofday({tv_sec=1604083673, tv_usec=213425}, NULL) = 0
[pid  1836] gettimeofday( <unfinished ...>
[pid  1835] gettimeofday({tv_sec=1604083673, tv_usec=213837}, NULL) = 0
[pid  1836] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=213911}, NULL) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL <unfinished ...>
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=214274}, NULL) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] <... ioctl resumed> , 0xb6dd6c9c) = 0
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=214833}, NULL) = 0
[pid  1836] gettimeofday( <unfinished ...>
[pid  1835] gettimeofday({tv_sec=1604083673, tv_usec=216434}, NULL) = 0
[pid  1836] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=216511}, NULL) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL <unfinished ...>
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=216812}, NULL) = 0
[pid  1836] <... ioctl resumed> , 0xb6dd6c9c) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] gettimeofday( <unfinished ...>
[pid  1835] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=217371}, NULL) = 0
[pid  1836] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=217688}, NULL) = 0
[pid  1835] gettimeofday( <unfinished ...>
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL <unfinished ...>
[... 1260 lines cut ...]
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL, 0xb6dd6c9c) = 0
[pid  1836] gettimeofday({tv_sec=1604083673, tv_usec=451533}, NULL) = 0
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL, 0xb6dd6c9c) = 0
[pid  1836] gettimeofday({tv_sec=1604083673, tv_usec=451791}, NULL) = 0
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL, 0xb6dd6c9c) = 0
[pid  1836] gettimeofday({tv_sec=1604083673, tv_usec=452048}, NULL) = 0
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL, 0xb6dd6c9c) = 0
[pid  1836] gettimeofday({tv_sec=1604083673, tv_usec=452305}, NULL) = 0
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL, 0xb6dd6c9c) = 0
[pid  1836] gettimeofday({tv_sec=1604083673, tv_usec=452587}, NULL) = 0
[pid  1836] ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL, 0xb6dd6c9c) = 0
[pid  1836] gettimeofday({tv_sec=1604083673, tv_usec=452833}, NULL) = 0
[... cut ...]
[pid  1836] gettimeofday( <unfinished ...>
[pid  1835] <... msgrcv resumed> {1, "l\000000\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"...}, 4095, 1, 0) = 1
[pid  1836] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=454272}, NULL) = 0
[pid  1835] msgsnd(8, {2, "0"}, 1, 0 <unfinished ...>
[pid  1836] futex(0x24084, FUTEX_WAIT_PRIVATE, 2, NULL <unfinished ...>
[pid  1835] <... msgsnd resumed> )      = 0
[pid  1835] msgrcv(8, strace: Process 1835 detached
 <detached ...>
^Cstrace: Process 1836 detached

michaellass, 2020-10-30:
>  Re-reading is possible only after disconnecting the power supply from DHT22 for a while.

> ```
> [pid  1835] <... msgrcv resumed> {1, "l\000000\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"...}, 4095, 1, 0) = 1
> [pid  1836] <... gettimeofday resumed> {tv_sec=1604083673, tv_usec=454272}, NULL) = 0
> [pid  1835] msgsnd(8, {2, "0"}, 1, 0 <unfinished ...>
> ```

Here `libgpiod_pulsein` reports that the sensor has not communicated at all (0 pulses received).

Two questions:
1. Is your raspberry pi under significant load or is the CPU more or less idle?
2. Does this issue also occur if you do not run the `strace`? I ask because the `strace` has significant impact on the performance of the sensor readout and basically makes it impossible to get valid readings.

If this occurs with an idle raspberry pi and no strace running, I wonder if your sensor actually is defect or if there is any radiation near the cabling, causing errors in the communication.

darton, 2020-11-01:
CPU RPi nie ma nic do roboty, pracuje tylko skrypt odczytu czujnika DHT22.

I started strace only after 80 unsuccessful attempt to read data from DHT22. I never run strace while the script is running normally. 

After disconnecting the power supply from the DHT22 sensor, the sensor worked without any problems for some time.

michaellass, 2020-11-05:
I guess that stopping and restarting your script also does not help? Maybe (just to make sure) try this again and after stopping the script make sure that `libgpiod_pulsein` also has been stopped (e.g., running `pidof libgpiod_pulsein` which should create no output then).

If really only disconnecting the sensor helps, I am afraid that I cannot provide much help on this particular issue. Looks like an issue in the sensor itself or some communication pattern causes it to misbehave. But I have never seen that before.

michaellass, 2021-10-07:
I don't think that I will invest any additional time here due to a lack of communication by the Adafruit folks (see: https://github.com/adafruit/libgpiod_pulsein/pull/2). Maybe someone else can pick this up?

Jemie-Wang, 2021-12-03:
I met the exact same error with the same error report, but my problem turned out to be a loose wire to 3.3v on RPi.