adafruit/Adafruit_CircuitPython_DHT #53: DHT (DHT11) not working on ESP32S2 board

jposada202020, 2020-11-15:
I am unable to read DHT11 sensor on ESP32S2. Following similar procedure from issue 45 I commented out the print statement on the lib.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-rc.0 on 2020-10-16; Metro ESP32S2 with ESP32S2
>>> import dht_simpletest
0 pulses: []
DHT sensor not found, check wiring
0 pulses: []
DHT sensor not found, check wiring
0 pulses: []
DHT sensor not found, check wiring
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/dht_simpletest.py", line 29, in <module>
KeyboardInterrupt: 

Sensor is working and tested in M4 Express.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-rc.2 on 2020-11-12; Adafruit Metro M4 Express with samd51j19
>>> import dht_simpletest
81 pulses: [54, 24, 54, 24, 54, 24, 53, 71, 53, 70, 54, 24, 54, 24, 54, 71, 54, 24, 53, 24, 54, 24, 54, 24, 54, 24, 53, 24, 54, 24, 54, 25, 54, 24, 54, 24, 53, 24, 54, 70, 54, 70, 54, 24, 54, 70, 53, 25, 54, 24, 54, 24, 54, 24, 54, 24, 53, 70, 54, 24, 54, 24, 54, 25, 53, 25, 53, 24, 54, 70, 54, 70, 54, 70, 54, 24, 53, 70, 54, 71, 54]
Temp: 78.8 F / 26.0 C    Humidity: 25% 
81 pulses: [54, 24, 54, 24, 54, 24, 53, 70, 54, 70, 54, 24, 54, 24, 54, 71, 53, 25, 53, 24, 54, 24, 54, 24, 54, 24, 53, 24, 54, 24, 54, 25, 54, 24, 53, 24, 54, 24, 54, 70, 54, 70, 54, 24, 53, 71, 53, 25, 54, 24, 54, 24, 54, 24, 53, 24, 54, 24, 54, 70, 54, 24, 54, 71, 53, 24, 54, 24, 54, 70, 54, 70, 54, 70, 53, 24, 54, 24, 54, 25, 53]
Temp: 78.8 F / 26.0 C    Humidity: 25% 
81 pulses: [54, 24, 54, 24, 53, 24, 54, 70, 54, 70, 54, 24, 54, 70, 53, 25, 54, 24, 54, 24, 54, 24, 53, 25, 53, 24, 54, 24, 54, 24, 54, 25, 53, 24, 54, 24, 54, 24, 54, 70, 54, 70, 53, 24, 54, 70, 54, 25, 54, 24, 54, 24, 53, 24, 54, 24, 54, 24, 54, 70, 54, 24, 53, 72, 53, 24, 54, 24, 54, 70, 54, 70, 54, 70, 53, 24, 54, 24, 54, 71, 54]
Temp: 78.8 F / 26.0 C    Humidity: 26% 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/dht_simpletest.py", line 34, in <module>
KeyboardInterrupt:

dgriswo, 2021-01-09:
I have this issue on a FeatherS2
Running: Adafruit CircuitPython 6.0.1 on 2020-12-28; FeatherS2 with ESP32S2

Error is:
Traceback (most recent call last):
  File "code.py", line 12, in <module>
  File "adafruit_dht.py", line 261, in temperature
  File "adafruit_dht.py", line 213, in measure
RuntimeError: DHT sensor not found, check wiring

Running: Adafruit CircuitPython 6.1.0-rc.0 on 2021-01-06; FeatherS2 with ESP32S2

Error is:
Traceback (most recent call last):
  File "code.py", line 12, in <module>
  File "adafruit_dht.py", line 261, in temperature
  File "adafruit_dht.py", line 245, in measure
RuntimeError: Checksum did not validate. Try again.

The error changed with a CircuitPython update.

tannewt, 2021-01-12:
@microdev1 any ideas?

jerryneedell, 2021-01-12:
I ran the dht_simplest for DHT11 on a UMFEATHERS2 and teh first reading had the error reported:
subsequent readings were OK
```
Adafruit CircuitPython 6.1.0-rc.0-16-gde5b138dc on 2021-01-08; FeatherS2 with ESP32S2
>>> 
>>> import dht_simpletest
Checksum did not validate. Try again.
Temp: 77.0 F / 25.0 C    Humidity: 34% 
Temp: 77.0 F / 25.0 C    Humidity: 34% 
Temp: 77.0 F / 25.0 C    Humidity: 34% 
Temp: 77.0 F / 25.0 C    Humidity: 34% 
```

jerryneedell, 2021-01-12:
However -- I am using pin IO5 and it only worked if I specified board.IO5
board.D5 did not work  -- that gave the DHT sensor not fond error

dgriswo, 2021-01-12:
@jerryneedell The FeatherS2 documentation shows A5=**D19**=IO5 for the pin silkscreened as 5.  If you specified D5, that is A10=**D5**=IO1 silkscreened as 1. 

I updated to Adafruit CircuitPython 6.1.0-rc.0-23-g1fb3d2448 on 2021-01-12; FeatherS2 with ESP32S2 and am still seeing the checksum failure on the first and subsequent requests using your sample code.

I have 2 DHT11 modules that both behave the similarly.  I'm powering with LDO1 (3v3) with a 10k ohm pull up resistor.  The sensors work as expected on a Feather M4 Express with CircuitPython 6.0.1

jerryneedell, 2021-01-12:
I concur with your description of the pin naming. I am using IO5.
I have no idea why it is not working for you but it works for me. I have my DHT11 connected the same way you describe.

dgriswo, 2021-01-12:
Not sure if this helps, I removed the comment from adafruit_dht.py line 209 and got the below output.  I'm not sure how to interpret these.

80 pulses: [24, 54, 24, 57, 24, 54, 72, 54, 72, 54, 24, 57, 24, 54, 72, 57, 24, 54, 24, 54, 27, 54, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 27, 54, 24, 54, 24, 57, 72, 54, 24, 54, 72, 54, 72, 54, 27, 54, 24, 57, 24, 54, 24, 57, 24, 54, 24, 54, 72, 57, 69, 57, 72, 54, 24, 57, 24, 54, 72, 54, 72, 54, 24, 57, 69, 57, 72, 54, 24, 87]
Checksum did not validate. Try again.
80 pulses: [24, 54, 24, 54, 27, 54, 72, 54, 72, 54, 24, 57, 24, 54, 72, 54, 27, 54, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 24, 57, 69, 57, 24, 54, 72, 54, 72, 54, 27, 54, 24, 57, 24, 54, 24, 57, 24, 54, 24, 54, 72, 57, 69, 57, 72, 54, 24, 57, 24, 54, 72, 54, 72, 54, 24, 57, 69, 57, 72, 54, 24, 87]
Checksum did not validate. Try again.
80 pulses: [27, 54, 24, 54, 24, 57, 72, 54, 72, 54, 24, 54, 24, 57, 72, 54, 24, 57, 24, 54, 24, 54, 27, 54, 24, 54, 27, 54, 24, 54, 27, 54, 24, 54, 27, 54, 24, 54, 72, 54, 27, 54, 72, 54, 72, 54, 27, 54, 24, 54, 24, 57, 24, 54, 24, 57, 69, 57, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 72, 54, 72, 54, 27, 54, 72, 54, 72, 54, 72, 87]
Checksum did not validate. Try again.

dgriswo, 2021-01-12:
Alright... success sort of.  I examined the pulses and determined that the first pulse in the array is the correct bit.  By changing adafruit_dht.py:100 from hi_sig = False to hi_sig = True, the bits are generated correctly from the pulses.  I know this isn't a solution, as it would break my sensor on the M4 Express.  Could this be a timing issue on the ESP32-S2?

80 pulses: [24, 57, 24, 54, 24, 54, 72, 54, 27, 54, 72, 54, 72, 54, 72, 57, 24, 54, 24, 54, 27, 54, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 24, 57, 69, 57, 24, 54, 72, 54, 72, 54, 72, 57, 24, 54, 24, 57, 24, 54, 24, 54, 27, 54, 72, 54, 72, 54, 27, 54, 24, 54, 24, 57, 69, 57, 72, 54, 24, 54, 72, 54, 24, 57, 24, 90]
Temp: 73.4 F / 23.0 C    Humidity: 23% 
80 pulses: [24, 57, 24, 54, 24, 54, 72, 57, 24, 54, 72, 54, 72, 54, 72, 57, 24, 54, 24, 54, 27, 54, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 27, 54, 24, 54, 24, 57, 69, 57, 24, 54, 72, 54, 72, 54, 72, 57, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 72, 54, 72, 54, 27, 54, 24, 54, 24, 57, 72, 54, 72, 54, 24, 54, 72, 54, 27, 54, 24, 90]
Temp: 73.4 F / 23.0 C    Humidity: 23%

dgriswo, 2021-01-12:
Sorry to spam the thread, but I repeated capturing the pulses on a Trinket M0 and the FeatherS2.  The Trinket recorded 81 pulses, with the data on odd number pulses (hi_sig=False to convert) while the FeatherS2 recorded 80 pulses with data on the even number pulses (hi_sig=True).  The missing initial pulse on the FeatherS2 is shifting the bits.

Trinket M0
81 pulses: [54, 24, 54, 24, 54, 24, 54, 71, 54, 24, 54, 71, 54, 24, 54, 72, 54, 24, 54, 24, 54, 24, 54, 24, 54, 24, 54, 24, 54, 24, 54, 25, 54, 24, 54, 24, 54, 24, 54, 71, 54, 26, 52, 71, 54, 71, 54, 25, 54, 24, 54, 24, 54, 24, 54, 24, 54, 24, 54, 71, 54, 71, 54, 72, 54, 24, 54, 24, 54, 71, 54, 71, 54, 24, 54, 24, 54, 71, 54, 26, 53]
Temp: 71.6 F / 22.0 C    Humidity: 21% 

FeatherS2
80 pulses: [24, 57, 24, 54, 24, 54, 72, 54, 27, 54, 72, 54, 24, 54, 75, 54, 24, 54, 27, 54, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 24, 57, 24, 54, 72, 54, 24, 57, 69, 57, 72, 54, 24, 57, 24, 54, 24, 54, 27, 54, 24, 54, 72, 54, 27, 54, 24, 54, 27, 54, 24, 54, 27, 54, 72, 54, 72, 54, 24, 57, 24, 54, 72, 54, 72, 90]
Temp: 71.6 F / 22.0 C    Humidity: 21%

jposada202020, 2021-01-13:
Hello I test the code from @jerryneedell and it worked
`Adafruit CircuitPython 6.1.0-rc.0 on 2021-01-06; Adafruit Metro ESP32S2 with ESP32S2
>>> import code
Checksum did not validate. Try again.
Temp: 75.2 F / 24.0 C    Humidity: 35% 
Temp: 75.2 F / 24.0 C    Humidity: 37% 
Temp: 75.2 F / 24.0 C    Humidity: 38% 
Temp: 75.2 F / 24.0 C    Humidity: 38% 
Temp: 75.2 F / 24.0 C    Humidity: 38% 
Temp: 75.2 F / 24.0 C    Humidity: 37% 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "code.py", line 34, in <module>
KeyboardInterrupt: 

Thanks`

jposada202020, 2021-02-08:
@dgriswo Have you been able to replicate the failure, with the new pin definitions?

dgriswo, 2021-02-08:
I updated my FeatherS2 to "Adafruit CircuitPython 6.2.0-beta.1-130-g1043d619f on 2021-02-08; FeatherS2 with ESP32S2" and adafruit_dht.mpy from "adafruit-circuitpython-bundle-6.x-mpy-20210206" and re-tested.

Still seeing the checksum validation with the new binaries. Enabling debug in adafruit_dht.py still shows a missing pulse (reads 80, should be 81).  I have two DHT11 with the same behavior, but I don't have a second FeatherS2 yet.

dgriswo, 2021-02-15:
I grabbed another UM FeatherS2 and loaded the code up.  Still seeing the issue of only 80 pulses with the second board.  Is this a DHT11 library issue or potentially a pulseio issue?

The DHT11 sensors that I'm testing with do work correctly with the Trinket M0.

jposada202020, 2021-02-23:
@dgriswo I got my hands in in of the FeatherS2. And I have similar results. 
I have tested both the DHT11 and DHT22 sensors, and in both I receive 80 pulses. I getting the same results as you.
I proceed to test with my Adafruit Metro Express S2, and I can read the temperature no problem.  I know the sensors work as I have been doing some testing with these.

I follow your recommendation to change hi_sig=True.
1. For the Metro ESP32S2 if I use hi-sig=True: Only the first measure will be good, as the pulses will start with low and then hig, however, after the first measure the pulses will start at high, and our flag hi_sig=True will not be longer be valid
80 pulses: [24, 54, 24, .....
Temp: 71.6 F / 22.0 C    Humidity: 37% 
81 pulses: [54, 24, 54, .....

2. For the featherS2, it is the opposite situation as you showed previously.

I need to dig more into the C library, and/or understand if the reason for this is that we are not using a context manager to close pulsein, and gets shifted + 1.

I will do some more testing during the weekend, I will need to build cp to see if can find anything in the c code. Let me know if you make more test in your end.

jposada202020, 2021-02-24:
Just to wrap here, I did more testing on this, I modified the library to create and destroy the Pulse_in object each measure, this only helps a little, I got 50% of the readings. 
The Pulse_in object is sending information like this
The first byte will have a high and then a low, after the first reading this behavior change and you will get a low first.
This cold be repeated consistently
In the METRO ESP32S2 happens the other way around, making the library work for the as the pulse buffer will contain after the first reading all "high" values
I tried pausing the pulse, clearing the buffer, in between to see if the pulses chain will be the same for the first and the rest without success. I would like to test more, but i don have the means to read the pulse accurately

FoamyGuy, 2024-11-13:
There was a fix in the core for this issue: https://github.com/adafruit/circuitpython/issues/6318 which I think might have been the solution to to the DHT11 not working. 

I'm going to close this for now, if anyone who finds this is still having trouble under the latest version of circuitpython we can re-open.

closed 2024-11-13 by FoamyGuy