adafruit/Adafruit_CircuitPython_DHT #57: How set GPIO Pin for DHT22 over ENV Variable?

Tob1as, 2021-01-05:
Hello,

i use the example file [dht_simpletest.py](https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/5ea758917ba854383f4b08e856efbef4d6727f02/examples/dht_simpletest.py) and replace D18 with D4 for use on Raspberry Pi 3.

But now I would like to set the pin over an enviroment variable. I add this to the script:

```python
import os

DHT_DATA_PIN = int(os.environ.get('GPIO_PIN_DHT22', 4))
```

But now I have problem to replace the number 4 in this line:
```python
dhtDevice = adafruit_dht.DHT22(board.D4)
```

This:
```python
dhtDevice = adafruit_dht.DHT22(DHT_DATA_PIN)
# or
dhtDevice = adafruit_dht.DHT22(4)
```
works local on Raspberry, but not in Docker Container... error: `self._pin = Pin(pin.id) AttributeError: 'int' object has no attribute 'id'`. So i think i must use it with "board".

Can anyone help or have an idea? :-)

(My complete [Code](https://github.com/Tob1as/rpi-sensors) if it helps.)

jposada202020, 2021-02-24:
@Tob1as Hello are you still need to solve this, I think this is maybe this is more a question than an issue. You could ask in the discord channel for Adafruit. However:
1. Yes you need to pass the object in this case, meaning board.D4. I do not use docker but normally when I need to pass arguments, objects or other things between libraries I used a dictionary, something like this 
            import board
            boardspins = {'D0': board.D0, 'D1': board.D1, 'D2': board.D2, 'D3': board.D3, 'D4': board.D4, 'D5': board.D5,
                        'D6': board.D6, 'D7': board.D7, 'D8': board.D8, 'D9': board.D9, 'D10': board.D10,
                        'D13': board.D13}
           pin4 = boardspins['D4']
and then you will call them using dhtDevice = adafruit_dht.DHT22(pin4)

Tob1as, 2021-02-24:
@jposada202020 
Hello, thank you, that works. :-) 

Now I hope for a solution for "[A full buffer was not returned. Try again.](https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/33)" also and then I am happy.

closed 2021-02-24 by Tob1as