adafruit/Adafruit_CircuitPython_DHT #60: DHT22 reports wrong negative temperatures

brlnr23, 2021-02-01:
First off: it sounds like the bug which has been reported here: https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/9 but it happend to me with the latest version as well. 

I have recently migrated from the old Adafruit lib to this one, because I hoped that the bug which occurred in the old lib as well has been fixed in the new one. Unfortunately it seems as if there is an issue with the new lib in my setup as well. 

* I am running a DHT22 sensor with a Raspberry Pi
* positive temperature and humidity are reported correctly
* negative values are currently around -3276,1°C (-5865,0°F) - the decimal varies probably because the outside temp is as well
* I followed this tutorial (in German) - https://tutorials-raspberrypi.de/raspberry-pi-luftfeuchtigkeit-temperatur-messen-dht11-dht22/ basically it's about installing it via
```sudo pip3 install adafruit-circuitpython-dht
sudo apt-get install libgpiod2```

Unfortunately I am not really familiar which informtation I might need to provide for this bug report nor I have no clue what I could try next. 

Therefore: any help is appreciated :)

Side note: the sensor I have in use does not require a pull up resistor - and I assume that this is correct - as said: all values except negative temperatures are pretty much accurate.

jposada202020, 2021-02-08:
@brlnr23 Hello,
I adapted a little the code in the webpage that you mentioned.  Could you try this code, changing DHT11 for DHT22, unfortunately I do not have that Sensor for testing.  It worked for mi Using a Raspberry PI 3. I can tested also in a ZERO W if needed.
```
import Adafruit_DHT as dht
from time import sleep

DHT = 4  # Set DATA pin
while True:
    #  Read Temp and Hum from DHT11
    h, t = dht.read_retry(dht.DHT11, DHT)  # If using DHT22, change the Value here
    print('Temp={0:0.1f}*C  Humidity={1:0.1f}%'.format(t, h))
    sleep(5) #  Wait 5 seconds and read again

Could you test with this code again?
```

jerryneedell, 2021-02-08:
Just a warning. Adafruit_DHT is not the same as Adafruit_CircuitPython_DHT  -- they are different libraries.
Adafruit_CircuitPython_DHt should handle negative temperatures properly for a DHT22. A DHT11 cannot measure negative temperatures.

jerryneedell, 2021-02-08:
I'm a bit confused by this discussion -- Adafruit_DHT is a deprecated library https://github.com/adafruit/Adafruit_Python_DHT  and is not the same as Adafruit_CircuitPython_DHT.

Adafruit_CircuitPython_DHT "should" handle negative temperatures properly for a DHT22. The DHT11 cannot measure negative temperatures.

brlnr23, 2021-02-08:
I am confused now as well .oO
The docs for CircuitPython mention `import Adafruit` (https://github.com/adafruit/Adafruit_CircuitPython_DHT#basics) 

I would be happy to test and help, but I would need some more details how I could make use of the non-deprecated library. 

@jposada202020 , thanks for this snippet. It runs but unfortunately the current temp is - believe it or not - above 0 ;) Nevertheless I will closely monitor this and post updates if any.

jposada202020, 2021-02-08:
Hello, just to clarify:
1. Regarding the library: my mistake, I was indeed using the old library as @jerryneedell  mentioned. Thanks for the clarification
2. Code: if the snippet works, it could mean that you have as me installed the old library.

Sorry for the confusion.  And you are lucky @brlnr23 regarding the temp here today is -14 ;)

brlnr23, 2021-02-09:
@jposada202020, now I can confirm that the snippet does not work. As soon as it's below 0°C the values are way off :/ If the new library fixes this, I would be happy if someone could highlight how I can use the new library (ideally besides the old one which is used in separate other scripts)

jposada202020, 2021-02-09:
@brlnr23. You should try the simple test in the examples. https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/master/examples/dht_simpletest.py. This example use the new library. And as jerryneedell mentioned before it should be working for below 0 temperatures..

jerryneedell, 2021-02-09:
If you are using a Raspberry Pi, then you will have to also install the "Blinka" compatibility system as descried here https://learn.adafruit.com/circuitpython-on-raspberrypi-linux

brlnr23, 2021-02-09:
not sure what I am doing wrong, but I can't manage to get negative values :/ 
Is there any possibility for further checks?

This is the script I used after installing blinka. 
```
#!/usr/bin/env python3
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time
import board
import adafruit_dht

# Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT22(board.D4)

# you can pass DHT22 use_pulseio=False if you wouldn't like to use pulseio.
# This may be necessary on a Linux single board computer like the Raspberry Pi,
# but it will not work in CircuitPython.
dhtDevice = adafruit_dht.DHT22(board.D4, use_pulseio=False)

while True:
    try:
        # Print the values to the serial port
        temperature_c = dhtDevice.temperature
        temperature_f = temperature_c * (9 / 5) + 32
        humidity = dhtDevice.humidity
        print(
            "Temp: {:.1f} F / {:.1f} C    Humidity: {}% ".format(
                temperature_f, temperature_c, humidity
            )
        )

    except RuntimeError as error:
        # Errors happen fairly often, DHT's are hard to read, just keep going
        print(error.args[0])
        time.sleep(2.0)
        continue
    except Exception as error:
        dhtDevice.exit()
        raise error

    time.sleep(2.0)
```

called from CLI via 
`./dht_simpletest.py `

and this is the output I got
`Temp: -5865.9 F / -3276.6 C    Humidity: 75.2% `

jerryneedell, 2021-02-09:
Do you still have the old adafruit_DHT library installed? Possibly some confusion...

brlnr23, 2021-02-09:
do you see at glance which are old and which are new modules? Possibly capital/lower `adafruit`?

```
 $ pip list
Package                    Version
-------------------------- ---------
Adafruit-Blinka            6.2.0
adafruit-circuitpython-dht 3.5.4
Adafruit-DHT               1.4.0
Adafruit-PlatformDetect    2.28.0
Adafruit-PureIO            1.1.8
adafruit-python-shell      1.2.3
args                       0.1.0
asn1crypto                 0.24.0
certifi                    2018.8.24
chardet                    3.0.4
clint                      0.5.1
cryptography               2.6.1
entrypoints                0.3
idna                       2.6
keyring                    17.1.1
keyrings.alt               3.1.1
pip                        18.1
pycrypto                   2.6.1
pyftdi                     0.52.9
PyGObject                  3.30.4
pyserial                   3.5
python-apt                 1.8.4.3
pyusb                      1.1.1
pyxdg                      0.25
requests                   2.21.0
rpi-ws281x                 4.2.5
RPi.GPIO                   0.7.0
SecretStorage              2.3.1
setuptools                 53.0.0
six                        1.12.0
ssh-import-id              5.7
sysv-ipc                   1.1.0
urllib3                    1.24.1
wheel                      0.32.3

```

jerryneedell, 2021-02-09:
uninstall adafruit-DHT -- it is the deprecated library

brlnr23, 2021-02-09:
I just uninstalled it - now waiting for negative values ;) Thanks anyways so far for your support! I really appreciate it. 
I will keep you posted

jposada202020, 2021-02-10:
Just FYI, I am doing some test on my side also with interesting results. I just do not want to spam this issue. if you want to further we can discuss the in the Adafruit Discord channel.  Anyway, all this and reading through the current and past issues made me order some DHT22 :) to test.

jposada202020, 2021-02-13:
@brlnr23 I did my tests with the DHT22:  This was with a RP4 running Buster with the Adafruit_Circuit_Python library. It is not perfect but this sensor will never be :)
```
Python 3.7.3 (/usr/bin/python3)
>>> %Run Testing_DHT22.py
Temp: 9.3 F / -12.6 C    Humidity: 43.4% 
Temp: 9.3 F / -12.6 C    Humidity: 45.0% 
Temp: 9.1 F / -12.7 C    Humidity: 45.0% 
Checksum did not validate. Try again.
Temp: 9.0 F / -12.8 C    Humidity: 46.6% 
Temp: 8.8 F / -12.9 C    Humidity: 46.7% 
A full buffer was not returned. Try again.
Temp: 8.8 F / -12.9 C    Humidity: 47.4% 
Temp: 8.8 F / -12.9 C    Humidity: 47.5% 
```

brlnr23, 2021-02-13:
@jposada202020 thanks for testing this on behalf! Two questions I have atm: 
* Have you wired your sensor with a resistor or without? 
* Is `Testing_DHT22.py` exact the same what I use (see above: `dht_simpletest.py`)

Two reasons I could imagine which might cause the fail is probably the missing resistor, or something got messed up with the old/new lib installation (which would require a clean Buster setup)

jerryneedell, 2021-02-13:
> @brlnr23 I did my tests with the DHT22: This was with a RP4 running Buster with the Adafruit_Circuit_Python library. It is not perfect but this sensor will never be :)
>
> ```
> Python 3.7.3 [...]

FYI - This is typical of what I have seen with DHT22 sensors on RPi 4 as well. Errors are common.

jposada202020, 2021-02-13:
> @jposada202020 thanks for testing this on behalf! Two questions I have atm:
>
>     * Have you wired your sensor with a resistor or without?
>
>     * Is `Testing_DHT22.py` exact the same what I use (see [...]

@brlnr23 
1. DHT22 include the resistor already. 
2. The same code as per dht_simpletest.py

I did indeed use a new SD card. I just did not want to make my same mistake again :). Then I installed Buster, Installed Blinka, Installed the Adafruit-Circuitpython-dht, download the code.  So fresh from the box installation.

newb87392, 2021-11-19:
Good morning, long time listener, first time caller:

I am also experiencing this issue with a DHT22. As soon as the temperature crosses 0C it becomes -3276.7C (-5866.1 F)

> Fri Nov 19 06:14:34 2021 -- Temp: 32.0 F / -0.0 C    Humidity: 37.5%
> Fri Nov 19 06:14:45 2021 -- Temp: 32.0 F / -0.0 C    Humidity: 37.5%
> Fri Nov 19 06:14:55 2021 -- Temp: 32.0 F / -0.0 C    Humidity: 37.5%
> Checksum did not validate. Try again.
> Fri Nov 19 06:15:16 2021 -- Temp: -5866.1 F / -3276.7 C    Humidity: 37.5%
> Fri Nov 19 06:15:26 2021 -- Temp: -5866.1 F / -3276.7 C    Humidity: 37.6%

----------------------------------

```
Package                    Version
-------------------------- ---------
Adafruit-Blinka            6.15.0
adafruit-circuitpython-dht 3.6.4
Adafruit-PlatformDetect    3.17.2
Adafruit-PureIO            1.1.9
adafruit-python-shell      1.3.3
arandr                     0.1.10
args                       0.1.0
astroid                    2.5.1
asttokens                  2.0.4
```

I am using import adafruit_dht

module path: 
/usr/local/lib/python3.9/dist-packages/adafruit_dht.py

I really can't explain why this is happening. I have tried with and without 10k pullup resistor.

Can anyone with a freezer check and see if it's just me?

apra00, 2022-12-15:
Hi everyone!

I had this issue too. I just did some try&error testing and found out that with my python script I can add the wrong value at the actual -0.1˚C to the wrong number (ex: temperature_c+3276.6) and then multiply that with -1 to get the actual negative temperature.

For this you have to get your reported value at -0.1˚C and write it down. Then go into your code and add it there as the correction value.

I don't have an other DHT22 for testing but I'm not sure if every one has the same reported value at -0.1˚C. So be aware of that.

Tested it for a week and compared it the an actual thermometer and it worked for me.

There is my code:
```
import time
import board
import adafruit_dht

dhtDevice = adafruit_dht.DHT22(board.D6)
dhtDevice._trig_wait = 1500

temperature_c = dhtDevice.temperature
humidity = dhtDevice.humidity

try:
    if temperature_c < -1:
        temperature_c = round((temperature_c+3276.6)*(-1), 2)
    if (humidity != None) and (temperature_c != None) and (100 > temperature_c > -100):
        # do something with the correct temperature
        print(temperature_c, humidity)
    else:
        print("None value")
        
except RuntimeError as error:
            # Errors happen fairly often, DHT's are hard to read, just keep going
            print(error.args[0])
except Exception as error:
            dhtDevice.exit()
            raise error
```

This gives the correct output: `-0.1 99.9`
Yes, there is fog outside :)

dg8ygz, 2022-12-15:
It seems some DHT22 sensors act different at negative temperatures; my ones returns a twos compliment value.
I adapt the adafruit_dht.py to my needs (maybe a solution for further versions?!)

here's my patch for the actual 3.7.8
```
246,249c246,247
<                 new_temperature = (((buf[2] & 0x7F) << 8) | buf[3]) / 10
<                 # set sign
<                 if buf[2] & 0x80:
<                     new_temperature = -new_temperature
---
>                 new_temperature = int.from_bytes(buf[2:4], byteorder="big", signed=True) / 10
>
```