adafruit/Adafruit_CircuitPython_DHT #61: DHT22 not working on Raspberry Pi Pico using CircuitPython

gregwa1953, 2021-02-02:
Trying to use the DHT22 program on my Raspberry Pi Pico.
I changed the pin to GP1 and let the program autoload into Mu.

I got the following error message:

Traceback (most recent call last):
  File "code.py", line 10, in <module>
  File "adafruit_dht.py", line 274, in __init__
  File "adafruit_dht.py", line 52, in __init__
Exception: Bitbanging is not supported when using CircuitPython.

Any help would be appreciated.  Thank you in advance!
Greg

jerryneedell, 2021-02-02:
The DHT library requires the pulseio module. Unfortunately, the pulseio module has not yet been ported to the Pico.
The Bitbang reference is generated because on some platforms (Raspberry Pi using the "Blinka" compatibility library) the library will use "bit-banging" when pulseio is not available - this option is not available to microcontrollers runing CircuitPython.

Until pulseio is made available, the DHT sensors will not be useable on the Pico.

jerryneedell, 2021-02-02:
Closing since this is not an issue with the DHT library.

gregwa1953, 2021-02-02:
Thank you very much!

jerryneedell, 2021-02-02:
re-open until this is resolved

jerryneedell, 2021-03-13:
Now that PulseIO had ben added to CP I tried the dht_simpletest with a DHT22 on pin GP22
```
Adafruit CircuitPython 6.2.0-beta.4-121-g550eab2dd on 2021-03-13; Raspberry Pi Pico with rp2040
>>> 
>>> import dht_simpletest
Checksum did not validate. Try again.
Temp: 71.1 F / 21.7 C    Humidity: 23.9% 
Checksum did not validate. Try again.
Checksum did not validate. Try again.
Checksum did not validate. Try again.
Checksum did not validate. Try again.
Temp: 71.1 F / 21.7 C    Humidity: 23.8% 
Checksum did not validate. Try again.
Checksum did not validate. Try again.
Checksum did not validate. Try again.
Temp: 71.1 F / 21.7 C    Humidity: 24.1% 
Temp: 71.1 F / 21.7 C    Humidity: 24.1% 
Temp: 71.1 F / 21.7 C    Humidity: 24.3% 
Checksum did not validate. Try again.
Temp: 71.1 F / 21.7 C    Humidity: 24.4% 
```
It works --- not great, but the DHT22 is finicky...

gregwa1953, 2021-03-13:
I didn't know that CircuitPython had been updated to include PulseIO.
DHT22 is now working under MicroPython for the Pico.  You can see the
thread at https://www.raspberrypi.org/forums/viewtopic.php?f=146&t=303606
.  The code provided (at github) also supports DHT11.

I found no major issues like you are seeing with the checksum errors.  I
ran it continuously for about 36 hours without a hitch.  That having been
said, I remember from earlier work with the DHT22 and DHT11, timing is
fairly critical.  In my test code, I think I was only pulling a temp about
every 5 seconds, which for me is pretty much as fast as I can imagine
anything would need, but your mileage may vary ;o)

I'll try to carve out some time this weekend to play with the new version
of CircuitPython and see if I can get some reasonable readings.
Greg

jerryneedell, 2021-03-13:
The PulseIO PR was merged yesterday and it is not in the "released" version ...yet. It is available from the "S3" latest builds.

gregwa1953, 2021-03-13:
Thanks for the information.

Trying to deal with MicroPython, CircuitPython, other Microcontrollers in a
learning mode with everything else that life throws at me (mostly my fault)
keeps me hopping from one project to another without time to keep up to
date on everything I should.

Again, thank you for the information.

Greg

mperino, 2021-03-30:
This works well enough.  Wired up to pin 10 (GP7) on Circutpython 6 from adafruit:

```
import time
import board
import adafruit_dht

print("library loaded")
sleep_timer = 2.0
deep_sleep = 90
retries = 4
samples = 3
dhtDevice = adafruit_dht.DHT22(board.GP7)

def sample_and_avg_DHT(sleep_timer, retries, samples):
    valid = 0
    invalid = 0
    temperature_c =0
    humidity = 0
    while valid < samples:
        try:
            temperature_c = temperature_c + dhtDevice.temperature
            humidity = humidity + dhtDevice.humidity
            valid += 1

        except:
            invalid += 1
            if invalid > retries:
                print("Retries exceeded Aborting")
                print("Last Error{}".format(dhtDevice.exit()))
        time.sleep(sleep_timer)
    return (round(temperature_c/valid, 1), round(humidity/valid,1))

print("Temperature:{}C, Humidity:{}%".format(*sample_and_avg_DHT(sleep_timer, retries, samples)))

gregwa1953, 2021-03-30:
Thank you for the update!

closed 2021-03-30 by gregwa1953