adafruit/Adafruit_CircuitPython_DHT #63: Unable to set line 18 to input

Mogabow, 2021-02-25:
Hello, 
I am using a Raspberry Pi 4 with a DHT11 module and used the code given in the example "simple dht test" while changing it to the pin I am using. The first time running the programs works fine. However, after stopping the program and running again I end up with the same issue where it says "Unable to set line 18 to input". Does anyone know how I can prevent this. apologies if anything was vague I am new to this. Thanks

jposada202020, 2021-02-25:
@Mogabow  No worries, in order to understand this better couple of questions:
Are you using the Blinka library in your pi? 
According to your description I am guessing that you are talking about the actual error raised and not the line in the `simple_dht_test.py` script. am I right assuming this?

Mogabow, 2021-02-25:
@jposada202020 Thank you for the fast reply! 
I am unsure if I am using the Blinka library however I did install the Adafruit_CircuitPython_DHT
there seems to be no error in the lines of the script the error only rises when I try to actually run the program for the second time after turning on my Raspberry Pi.
I have used the following coding on python 3 to run the program.
[New Text Document.txt](https://github.com/adafruit/Adafruit_CircuitPython_DHT/files/6046015/New.Text.Document.txt)
The first time I run the program I get the following values which prove it is working fine, with errors from time to time that automatically fix themselves.
[image]
But if I stop it and try to run it again I get this message in the shell.
[image]

hope that clears the questions you had.

jposada202020, 2021-02-25:
@Mogabow No problem, my favorite library in the whole world :). I need to connect my raspberry 4. But if you need more hep you can join the Adafruit discord channel and the question in help-with-circuitpython.

Mogabow, 2021-02-26:
@jposada202020 thanks! would you mind sending me the link to the discord channel? I tried to have another go at it with no success.

jposada202020, 2021-02-26:
@Mogabow https://discord.gg/adafruit leave the question in the sub-channel and we can discuss, I tried yesterday starting and stoping to see if I can reproduce the failure without succes, but maybe I am missing something. Thanks

caternuson, 2021-03-05:
> However, after stopping the program and running again 

Is your program an infinite loop that you stop via `<CTRL><C>`?

Mogabow, 2021-03-06:
thank you for your reply. Yes, it is an infinite loop that will keep on getting readings. using ```<CTRL><C>``` interrupts the code and rerunning it again gives the error specified above. I am reading the forum threads and attempting some changes to the code now.

jposada202020, 2021-03-06:
@caternuson Thanks

Mogabow, 2021-03-06:
Hello, once again I would like to thank you all for the help. I have followed the forum's and tested if all my libraries were correctly installed and up to date, which they were. I have then tried to use the pulse in command, which end up with the error ```DHT sensor not found, check wiring``` so i checked the wiring but I am quite certain they are connected properly otherwise i would not be getting values without the pulsein command the first time i run the program.

caternuson, 2021-03-06:
@Mogabow In your first post it sounded like it was working.

>  The first time running the programs works fine.

You are unable to repeat that previous success? (getting it to work on first run)

Mogabow, 2021-03-06:
@caternuson @jposada202020 alright after I updated the libraries now it works, I was using the wrong code after updating the libraries. Silly mistake on my part. Thank you so much for all your help!!

closed 2021-03-18 by Mogabow

jposada202020, 2021-03-18:
@Mogabow No problem Thanks to @caternuson he make the link here