adafruit/Adafruit_CircuitPython_DHT #65: DHT22 not working on Raspberry Pi wh

QS3445, 2021-03-05:
Hope this is the right place to post this.. I'm a newby trying to setup a DHT 22 with rpwh, i am trying the dht_simpletest.py to see if my DHT 22 is reading ok, but getting this error...any thoughts on what going on.. Thanks

pi@raspberrypi:~ $ sudo python3 dht_simpletest.py
Traceback (most recent call last):
  File "dht_simpletest.py", line 9, in <module>
    dhtDevice = adafruit_dht.DHT22(board.D4)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_dht.py", line 275, in __init__
    super().__init__(False, pin, 1000, use_pulseio)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_dht.py", line 56, in __init__
    self.pulse_in = PulseIn(self._pin, 81, True)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 73, in __init__
    message = self._wait_receive_msg(timeout=0.25)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 92, in _wait_receive_msg
    "Timed out waiting for PulseIn message. Make sure libgpiod is installed."
RuntimeError: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

I checked python3, adafruit_dht.py, and Libgpiod are all installed !!

Would be thankful for any fix suggestions..

Thanks
QS3445

jposada202020, 2021-03-05:
@QS3445 not sure but it seems that you are using PulseIn with the RP. you may need to setup your script to not to use it. This is mentioned in the note:
https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/ceb883d09587612d2f96037f54c93805e4175db3/examples/dht_simpletest.py#L11
https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/ceb883d09587612d2f96037f54c93805e4175db3/examples/dht_simpletest.py#L12
I am assuming that you install Adafuit Blinka library to use the board pin definition, is that right?
Is this board a ZERO W? is so I have been very unlucky using the zero with the DHT.

QS3445, 2021-03-05:
Ok ..thanks i'll try without using the PulseIn 

Yes the Adafruit Blinka library is installed, i am using the Pi Zero WH

I let you know how it goes later... Thank you

QS3445

QS3445, 2021-03-06:
I got a response now...that DHT senor not found 

pi@raspberrypi:~ $ sudo python3 dht_simpletest.py
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
DHT sensor not found, check wiring
^Z
[1]+  Stopped                 sudo python3 dht_simpletest.py
pi@raspberrypi:~ $

wiring looks good to me, i have VCC to P2, GND to P6, and DAT to P7

So i don't know what can be wrong with the wiring... any ideas anyone ?

razerraz, 2021-03-24:
I'm facing the exact same issue with an antique (but working perfectly) pi one rev B, and DHT22.
Dht reading from old adafruit library works most of the time, even if it goes bad now and then.
I've tried enabling or disabling pulseio, libgpiod is installed and sysv_ipc as well.
If the plan is to lean on pulseio, it may be another bug to report directly on blinka repo since it goes very bad heating all cpu process when a PulseIn is enabled on any pin.
Can provide more info on demand

SunSDSE, 2021-05-20:
I have the exact same issue but with a Raspberry Pi 3.  
All my code worked fine with the old Adafruit_DHT library which is now "DEPRECATED" 
I have 6 sensors wired and were working fine, before the attempted use with this new one.

Just for grins, I decided to test all my pins using the attached code.  I can't find any with this new library.

Is the Raspberry Pi 3 supported?

`# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time
import board
import adafruit_dht
active_gpio = {}
print(board.D12)
test = board.D12
active_gpio[1] = board.D1
active_gpio[2] = board.D2
active_gpio[3] = board.D3
active_gpio[4] = board.D4
active_gpio[5] = board.D5
active_gpio[6] = board.D6
active_gpio[7] = board.D7
active_gpio[8] = board.D8
active_gpio[9] = board.D9
active_gpio[10] = board.D10
active_gpio[11] = board.D11
active_gpio[12] = board.D12
active_gpio[13] = board.D13
active_gpio[14] = board.D14
active_gpio[15] = board.D15
active_gpio[16] = board.D16
active_gpio[17] = board.D17
active_gpio[18] = board.D18
active_gpio[19] = board.D19
active_gpio[20] = board.D20
active_gpio[21] = board.D21
active_gpio[22] = board.D22
active_gpio[23] = board.D23
active_gpio[24] = board.D24
active_gpio[25] = board.D25
active_gpio[26] = board.D26
active_gpio[0] = board.D27
# Initial the dht device, with data pin connected to:
# you can pass DHT22 use_pulseio=False if you wouldn't like to use pulseio.
# This may be necessary on a Linux single board computer like the Raspberry Pi,
# but it will not work in CircuitPython.
# dhtDevice = adafruit_dht.DHT22(board.D18, use_pulseio=False)

for i_pin in range(len(active_gpio)):
    print("Trying Pin {}: {}".format(i_pin, active_gpio[i_pin]))
    dhtDevice = adafruit_dht.DHT22(active_gpio[i_pin],  use_pulseio=False)
    for j in range(2):
        try:
            # Print the values to the serial port
            temperature_c = dhtDevice.temperature
            temperature_f = temperature_c * (9 / 5) + 32
            humidity = dhtDevice.humidity
            print(
                "Temp: {:.1f} F / {:.1f} C    Humidity: {}% ".format(
                    temperature_f, temperature_c, humidity
                )
            )

        except RuntimeError as error:
            # Errors happen fairly often, DHT's are hard to read, just keep going
            print(error.args[0])
            time.sleep(2.0)
            continue
        except Exception as error:
            print("Exception : {} ".format(error.args[0]))
            continue
            # dhtDevice.exit()
            # raise error

        time.sleep(2.0)`

JavanXD, 2021-08-14:
For me the DHT22 is working on my Raspberry Pi Zero WH but only if I run the python3 script without sudo. I also need to use `use_pulseio=True`.
I commented with a code snippet here: https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/28#issuecomment-898899721

ccvaillant1992, 2021-10-04:
I have the same error @QS3445 has,  **DHT sensor not found, check wiring**. I am using a Raspberry Pi 4 Model B.

Does anyone know how to resolve this error?

vrudikov, 2024-01-23:
> I have the same error @QS3445 has, **DHT sensor not found, check wiring**. I am using a Raspberry Pi 4 Model B.
> 
> Does anyone know how to resolve this error?

Please show your code. I had the same. I think I have a solution for you