adafruit/Adafruit_CircuitPython_DHT #66: DHT11 not working properly

yeyeto2788, 2021-03-16:
Using version `3.5.8` in a loop it almost never gets a reading with the following code:

```python

import time
import board
import adafruit_dht

dhtDevice = adafruit_dht.DHT11(board.PG6)

for try_number in range(1,200):
  print(f"Try number {try_number}")
  try:
    print(f"Got temp of: {dhtDevice.temperature}")
    print(f"And hum of: {dhtDevice.humidity}")
    break
  except RuntimeError as e:
    print(e)
    time.sleep(2.1)
  else:
    time.sleep(2.1)
  print("\n")
```

The weird part is that using the version `3.5.5` it actually get the readings correctly.

More data:
**Python**: 3.8.5
**SBC**: Orange Pi Lite
**Installed libraries**:
```console
Adafruit-Blinka==6.4.1
adafruit-circuitpython-busdevice==5.0.6
adafruit-circuitpython-dht==3.5.5
adafruit-circuitpython-mcp3xxx==1.4.5
Adafruit-PlatformDetect==3.4.0
Adafruit-PureIO==1.1.8
appdirs==1.4.4
attrs==20.3.0
bcrypt==3.2.0
cached-property==1.5.2
certifi==2019.11.28
cffi==1.14.3
chardet==3.0.4
cryptography==3.2.1
dbus-python==1.2.16
distlib==0.3.1
distro==1.5.0
distro-info===0.23ubuntu1
docker==4.3.1
docker-compose==1.27.4
dockerpty==0.4.1
docopt==0.6.2
filelock==3.0.12
idna==2.8
iotop==0.6
jsonschema==3.2.0
mudpi==0.10.0
paho-mqtt==1.5.1
paramiko==2.7.2
pycparser==2.20
pyftdi==0.52.9
PyGObject==3.36.0
PyNaCl==1.4.0
pyrsistent==0.17.3
pyserial==3.5
python-apt==2.0.0+ubuntu0.20.4.4
python-dotenv==0.15.0
pyusb==1.1.1
PyYAML==5.3.1
redis==3.5.3
requests==2.22.0
requests-unixsocket==0.2.0
six==1.14.0
texttable==1.6.3
unattended-upgrades==0.1
urllib3==1.25.8
virtualenv==20.1.0
websocket-client==0.57.0
```

jposada202020, 2021-03-16:
Thanks, for the report, @yeyeto2788 
Triggering time was changed in version 3.5.6, and corrected in 3.5.7 to improve trigger reading time in the DHT11. So in you case, this change is not working correctly? DHT11 needs more time to react after the initial signal, so this was taken into consideration.  Could you try to change the value as show in the code below, and put the old value. to see if it works. 

V3.5.6
https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/85b6a367f9389f0b03b85e1399950c6b8872e40e/adafruit_dht.py#L142

V3.5.7
https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/ceb883d09587612d2f96037f54c93805e4175db3/adafruit_dht.py#L142

yeyeto2788, 2021-03-17:
Hey @jposada202020,

Thanks for the quick reply. I try both versions you mentioned and none of those worked. The code/wiring still remain the same but changing the library version give a completely different output

```python
import time
import board
import adafruit_dht

dhtDevice = adafruit_dht.DHT11(board.PG6)

for try_number in range(1,200):
  print(f"Try number {try_number}")
  try:
    print(f"Got temp of: {dhtDevice.temperature}")
    print(f"And hum of: {dhtDevice.humidity}")
    break
  except RuntimeError as e:
    print(e)
    time.sleep(2.1)
  else:
    time.sleep(2.1)
  print("\n")
```

**`V3.5.7`**
```console
(venv) yeyeto2788@orangepilite:~/workspace/mudpi-core$ pip install adafruit-circuitpython-dht==3.5.7
Collecting adafruit-circuitpython-dht==3.5.7
  Downloading adafruit-circuitpython-dht-3.5.7.tar.gz (28 kB)
Requirement already satisfied: Adafruit-Blinka in ./venv/lib/python3.8/site-packages (from adafruit-circuitpython-dht==3.5.7) (6.4.1)
Requirement already satisfied: Adafruit-PureIO>=1.1.7 in ./venv/lib/python3.8/site-packages (from Adafruit-Blinka->adafruit-circuitpython-dht==3.5.7) (1.1.8)
Requirement already satisfied: Adafruit-PlatformDetect>=3.1.0 in ./venv/lib/python3.8/site-packages (from Adafruit-Blinka->adafruit-circuitpython-dht==3.5.7) (3.4.0)
Requirement already satisfied: pyftdi>=0.40.0 in ./venv/lib/python3.8/site-packages (from Adafruit-Blinka->adafruit-circuitpython-dht==3.5.7) (0.52.9)
Requirement already satisfied: pyserial>=3.0 in ./venv/lib/python3.8/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-dht==3.5.7) (3.5)
Requirement already satisfied: pyusb>=1.0.0 in ./venv/lib/python3.8/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-dht==3.5.7) (1.1.1)
Building wheels for collected packages: adafruit-circuitpython-dht
  Building wheel for adafruit-circuitpython-dht (setup.py) ... done
  Created wheel for adafruit-circuitpython-dht: filename=adafruit_circuitpython_dht-3.5.7-py3-none-any.whl size=6837 sha256=64936289ef4ee9d95648779129c46efd18e8aaab255f6156d5a1735d1977b4e8
  Stored in directory: /home/yeyeto2788/.cache/pip/wheels/30/3e/2b/c43aa6a3c0afd819cf5cadd2089a674b9f2a580e1b3ef25b0b
Successfully built adafruit-circuitpython-dht
Installing collected packages: adafruit-circuitpython-dht
  Attempting uninstall: adafruit-circuitpython-dht
    Found existing installation: adafruit-circuitpython-dht 3.5.6
    Uninstalling adafruit-circuitpython-dht-3.5.6:
      Successfully uninstalled adafruit-circuitpython-dht-3.5.6

(venv) yeyeto2788@orangepilite:~/workspace/mudpi-core$ sudo /home/yeyeto2788/workspace/mudpi-core/venv/bin/python test.py
Try number 1
A full buffer was not returned. Try again.

Try number 2
A full buffer was not returned. Try again.

Try number 3
A full buffer was not returned. Try again.

Try number 4
A full buffer was not returned. Try again.

Try number 5
A full buffer was not returned. Try again.
^CTraceback (most recent call last):
  File "test.py", line 11, in <module>
    print(f"Got temp of: {dhtDevice.temperature}")
  File "/home/yeyeto2788/workspace/mudpi-core/venv/lib/python3.8/site-packages/adafruit_dht.py", line 244, in temperature
    self.measure()
  File "/home/yeyeto2788/workspace/mudpi-core/venv/lib/python3.8/site-packages/adafruit_dht.py", line 200, in measure
    raise RuntimeError("A full buffer was not returned. Try again.")
RuntimeError: A full buffer was not returned. Try again.
[... 83 lines cut ...]
```console
(venv) yeyeto2788@orangepilite:~/workspace/mudpi-core$ find ./ -name adafruit*dht*
./venv/lib/python3.8/site-packages/__pycache__/adafruit_dht.cpython-38.pyc
./venv/lib/python3.8/site-packages/adafruit_dht.py
(venv) yeyeto2788@orangepilite:~/workspace/mudpi-core$ sed -n -e 138,152p ./venv/lib/python3.8/site-packages/adafruit_dht.py
            dhtpin.value = True
            time.sleep(0.1)
            dhtpin.value = False
            time.sleep(0.001)
            timestamp = time.monotonic()  # take timestamp
            dhtval = True  # start with dht pin true because its pulled up
            dhtpin.direction = Direction.INPUT

            try:
                dhtpin.pull = Pull.UP
            # Catch the NotImplementedError raised because
            # blinka.microcontroller.generic_linux.libgpiod_pin does not support
            # internal pull resistors.
            except NotImplementedError:
                dhtpin.pull = None
```

And this might correspond to [Line 141 on commit e22d289ab480468fa305a451e3b9c1961c7f0fb2](https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/e22d289ab480468fa305a451e3b9c1961c7f0fb2/adafruit_dht.py#L141)

To summarize I don't really see a difference on using `self._trig_wait/1000000` which will give you `0.0018` and it is on version `3.5.7` than using just `0.001`. I mean, is that sensitive to time the sensor? 🤔

jposada202020, 2021-03-17:
@yeyeto2788 No problem,
Interesting, actually for the DHT11 will be `0.018` = it comes from here https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/ceb883d09587612d2f96037f54c93805e4175db3/adafruit_dht.py#L265. The interesting thing here is that in your case, the wait time of 0.001 works better. 
In a note, yes 3.5.6 would not work as the waiting time was 0 as you get an `python` int division. That was my mistake.
Could you make some test changing the value, maybe hardcoded it. it is very interesting. maybe is something that I am not seeing. :octocat: 
 I did as you can see in here some testing, so for me the library works in both sensors. with the trigger time https://github.com/jposada202020/DHT_Investigation
But I am very curious

yeyeto2788, 2021-03-17:
Sure, so want me to try out your script (modified with the pin used on my SBC) or you just want me to manually change the values on the library itself? It is almost the same as mine except for the checksum.

jposada202020, 2021-03-17:
Both :) but for this case it would be better to verify the library behaviour, understand why when we use 0.001 works and when we use the division in the trigger do not. So if you could manually change the library values that would really helpful.

yeyeto2788, 2021-03-17:
Based on your [script](https://raw.githubusercontent.com/jposada202020/DHT_Investigation/main/DHT_compraison_test.py) 

I get this output:
```console
(venv) yeyeto2788@orangepilite:~/workspace/mudpi-core$ sudo /home/yeyeto2788/workspace/mudpi-core/venv/bin/python test_a.py
DHT sensor not found, check wiring
DHT sensor not found, check wiring
Temp: 69.8 F / 21.0 C    Humidity: 20%
Traceback (most recent call last):
  File "test_a.py", line 34, in <module>
    print(f"Cycle: {i}--Checksum Errors: {dhtDevice.checksum_test}--Data Buffer Error: {dhtDevice.data_buffer_test}"
AttributeError: 'DHT11' object has no attribute 'checksum_test'
```

**I also tried with manual changes and my code and the results are below.**

**0.001** Gives 3 out of 10 reads
```console
141             # Using the time to pull-down the line according to DHT Model
142             time.sleep(0.001)
143             timestamp = time.monotonic()  # take timestamp

(venv) yeyeto2788@orangepilite:~/workspace/mudpi-core$ sudo /home/yeyeto2788/workspace/mudpi-core/venv/bin/python test.py
Try number 1
Got temp of: 21
And hum of: 20

Try number 2
DHT sensor not found, check wiring

Try number 3
DHT sensor not found, check wiring

Try number 4
DHT sensor not found, check wiring

Try number 5
DHT sensor not found, check wiring

Try number 6
DHT sensor not found, check wiring

Try number 7
Got temp of: 21
And hum of: 20

Try number 8
Got temp of: 21
And hum of: 20

Try number 9
DHT sensor not found, check wiring

Try number 10
DHT sensor not found, check wiring
```

**0.0001** Gives 2 out of 10
```console
141             # Using the time to pull-down the line according to DHT Model
142             time.sleep(0.0001)
143             timestamp = time.monotonic()  # take timestamp
144             dhtval = True  # start with dht pin true because its pulled up
145             dhtpin.direction = Direction.INPUT

(venv) yeyeto2788@orangepilite:~/workspace/mudpi-core$ sudo /home/yeyeto2788/workspace/mudpi-core/venv/bin/python test.py
Try number 1
DHT sensor not found, check wiring

Try number 2
[... 158 lines cut ...]
Got temp of: 21
And hum of: 20

Try number 6
Got temp of: 21
And hum of: 20

Try number 7
Got temp of: 21
And hum of: 20

Try number 8
Got temp of: 21
And hum of: 20

Try number 9
Got temp of: 21
And hum of: 20

Try number 10
Got temp of: 21
And hum of: 20
```

I hope it gives you an idea more or less on timing. Of course I haven't test it on other boards so would be great to test same timings on other boards.

jposada202020, 2021-03-17:
@yeyeto2788 did you try with the division :) why on earth it does not work hen dividing, do we need to use an if statemt. 😄 Thanks for testing. so timing is good, at least. why is not working with the class attribute ❓  😞

jposada202020, 2021-03-17:
@yeyeto2788 by the way thanks for testing :)

yeyeto2788, 2021-03-17:
@jposada202020 I forgot about it 😛 

Here you have the test:

```console
141             # Using the time to pull-down the line according to DHT Model
142             time.sleep(self._trig_wait / 1000000)
143             timestamp = time.monotonic()  # take timestamp
144             dhtval = True  # start with dht pin true because its pulled up
145             dhtpin.direction = Direction.INPUT
146
147             try:
148                 dhtpin.pull = Pull.UP
149             # Catch the NotImplementedError raised because
150             # blinka.microcontroller.generic_linux.libgpiod_pin does not su>
151             # internal pull resistors.

(venv) yeyeto2788@orangepilite:~/workspace/mudpi-core$ sudo /home/yeyeto2788/workspace/mudpi-core/venv/bin/python test.py
Try number 1
DHT sensor not found, check wiring

Try number 2
A full buffer was not returned. Try again.

Try number 3
A full buffer was not returned. Try again.

Try number 4
Got temp of: 21
And hum of: 20

Try number 5
A full buffer was not returned. Try again.

Try number 6
A full buffer was not returned. Try again.

Try number 7
A full buffer was not returned. Try again.

Try number 8
A full buffer was not returned. Try again.

Try number 9
A full buffer was not returned. Try again.

Try number 10
A full buffer was not returned. Try again.
```

So the thing is that it is actually working but giving more false positives or false reads using that conversion `time.sleep(self._trig_wait / 1000000)` which will be `0.018`. 

**NOTE:** This is a total different number to let's say `0.00185` which was the most promising number to return more reads.

jposada202020, 2021-03-17:
@yeyeto2788 :) are you sure that you have a DHT11, if so, it is pretty good.... 💨 , look at page 5 of this datasheet https://osoyoo.com/driver/DHT11-datasheet.pdf

yeyeto2788, 2021-03-17:
I hope it is a DHT11 because it actually looks like one, my sensor is [this one](https://www.electronicaembajadores.com/es/Productos/Detalle/LCA2008/modulos-electronicos/arduino-d1-mini/wemos-dht-d1-mini-sensor-de-humedad-y-temperatura-dht11?gclid=Cj0KCQjw0caCBhCIARIsAGAfuMyEblRY45h0-RudnI0HMjoeTxE_OvYncQg0NsG256-pGpxRc__82qwaAgVdEALw_wcB) which was bought on other site. 

Regarding that datasheet it states the following:
>the programme of MCU will set Data Single-bus voltage level from high to low
and this processmust take at least 18ms to ensure DHT’s detection of MCU's signal

So it has to be AT LEAST 18ms and I'm actually getting better results with less than that (1.85ms)🤮

If you need me to test something else just let me know.

P.D: I can also contact me through Discord and probably it would be more faster.

jposada202020, 2021-03-17:
@yeyeto2788 Thanks. See.. 👀  that is why I am so confused. I will contact you in the Adafruit discord :)

jposada202020, 2021-03-19:
@yeyeto2788 Hello I tried to contact you in the Adafruit discord channel. but there are two people with the same nickname, if you could leave me a message in the help-with-circuitpython channel, I can trace it.  Thanks I have an idea of what to do but I would need your help, and need to discuss also what your think and feedback. thanks

olixr, 2021-03-30:
I wanted to add some testing to help give more data. I have a RPi4 that works fine with both the DHT11 and DHT22. I also have a RPi zero w but it has more trouble getting reads from both the DHT11 and DHT22. I thought I would run tests on the zero and see if it helps reveal more on the timings. 

Here are some details:

Installed packaged
```
Package                          Version
-------------------------------- -------------------
Adafruit-Blinka                  6.4.1
adafruit-circuitpython-74hc595   1.2.4
adafruit-circuitpython-bme680    3.3.2
adafruit-circuitpython-busdevice 5.0.6
adafruit-circuitpython-charlcd   3.1.2.dev7+g07007b4
adafruit-circuitpython-debouncer 1.3.10
adafruit-circuitpython-dht       3.5.8
adafruit-circuitpython-mcp230xx  2.4.5
Adafruit-PlatformDetect          3.4.1
Adafruit-PureIO                  1.1.8
asn1crypto                       0.24.0
certifi                          2018.8.24
chardet                          3.0.4
cryptography                     2.6.1
entrypoints                      0.3
idna                             2.6
keyring                          17.1.1
keyrings.alt                     3.1.1
mudpi                            0.10.0
numpy                            1.16.2
paho-mqtt                        1.5.1
pip                              18.1
pycron                           3.0.0
pycrypto                         2.6.1
pyftdi                           0.52.9
PyGObject                        3.30.4
pyserial                         3.5
python-apt                       1.8.4.3
pyusb                            1.1.1
pyxdg                            0.25
PyYAML                           5.4.1
redis                            3.5.3
requests                         2.21.0
rpi-ws281x                       4.2.6
RPi.GPIO                         0.7.0
SecretStorage                    2.3.1
setuptools                       40.8.0
six                              1.12.0
ssh-import-id                    5.7
sysv-ipc                         1.1.0
urllib3                          1.24.1
wheel                            0.32.3
```

**Board**: RPi Zero W
**Python**: 3.7.3

Test Script Results
```
RPi Zero W DHT22 (Range: 500 - 2000) (Increments: 100)
Maximum reads: 10  out of 10 with the following times: 1000, 1900

RPi Zero W DHT11 (Range: 15000 - 24800) (Increments: 100)
Maximum reads: 10  out of 10 with the following times: 15100, 15300, 15400, 15500, 16800, 17000, 17200, 17300, 18900, 19900, 20000, 20200, 20600, 21000, 22200, 22700, 24100
```