adafruit/Adafruit_CircuitPython_DHT #67: Overflow Error after Querying Sensor Every 2 Seconds for ~2 Hours

DrewMayne, 2021-03-18:
The sensor is queried via a long running process (run as a systemd service under the user account pi) which crashes every two hours or so. The process gracefully exit and the service can be immediately restarted but it would be nice if the crash could be avoided. 

The following is an except from the log of the most recent crash (all previous instances were reported in an identical fashion):

    Mar 18 14:02:51 raspberrypi python3[30114]: De-initializing self.pulse_in
    Mar 18 14:02:51 raspberrypi python3[30114]: Traceback (most recent call last):
    Mar 18 14:02:51 raspberrypi python3[30114]:   File "/home/pi/climate_regulation/vpd_controller/simple_vpd_controller.py", line 166, in <module>
    Mar 18 14:02:51 raspberrypi python3[30114]:     raise error
    Mar 18 14:02:51 raspberrypi python3[30114]:   File "/home/pi/climate_regulation/vpd_controller/simple_vpd_controller.py", line 137, in <module>
    Mar 18 14:02:51 raspberrypi python3[30114]:     raw_temperature = dhtDevice.temperature
    Mar 18 14:02:51 raspberrypi python3[30114]:   File "/home/pi/.local/lib/python3.7/site-packages/adafruit_dht.py", line 244, in temperature
    Mar 18 14:02:51 raspberrypi python3[30114]:     self.measure()
    Mar 18 14:02:51 raspberrypi python3[30114]:   File "/home/pi/.local/lib/python3.7/site-packages/adafruit_dht.py", line 189, in measure
    Mar 18 14:02:51 raspberrypi python3[30114]:     pulses = self._get_pulses_pulseio()
    Mar 18 14:02:51 raspberrypi python3[30114]:   File "/home/pi/.local/lib/python3.7/site-packages/adafruit_dht.py", line 119, in _get_pulses_pulseio
    Mar 18 14:02:51 raspberrypi python3[30114]:     pulses.append(self.pulse_in.popleft())
    Mar 18 14:02:51 raspberrypi python3[30114]: OverflowError: unsigned short is greater than maximum

jposada202020, 2021-03-25:
Hello, thanks for the report could you provide some of your code. What kind of hardware are you using.  Thanks

DrewMayne, 2021-03-26:
I am using an Adafruit DHT22 sensor with a Rabserry Pi 3B+ running Raspbian GNU/Linux 10 (buster).

Here's the python code for interacting with the device: 

    # Initialize the dht device, with data pin connected to:
    dhtDevice = adafruit_dht.DHT22(board.D24)

    # Called on process interruption or unknown exception. 
    def endProcess(signalnum=None, handler=None, sysexit=True):

        global dhtDevice

        # de-initialize sensor
        dhtDevice.exit()

        # release resources
        vpd_controller.exit()

        # debug signum
        print("process ended with signal num:")
        print(signalnum)

        if (sysexit) :
            sys.exit()

    # ...

    # main processing, loop until program is shutdown or unknown exception is thrown
    while(True):
        try:

            # Note that sometimes you won't get a reading and
            # the results will be null (because Linux can't
            # guarantee the timing of calls to read the sensor).
            # If this happens try again!

            try:
                raw_temperature = dhtDevice.temperature
                new_temperature = vpd_controller.temperature(raw_temperature)
                
                raw_relative_humidity = dhtDevice.humidity
                new_relative_humidity = vpd_controller.relative_humidity(raw_relative_humidity)
                
            except RuntimeError as error:

                # Errors happen fairly often, DHT's are hard to read, just keep going

                if (debug):
                    print("Device 1: " + error.args[0])
                time.sleep(1)

        except Exception as error:

            print("simple_vpd_controler: " + error.args[0])

            # gracefully end program
            endProcess(sysexit=False)

            raise error

        time.sleep(2)