adafruit/Adafruit_CircuitPython_DHT #69: Initialization error?

yeyeto2788, 2021-03-30:
Please take a look at the following snippet (explanation after it):

```console
(venv) pi@raspberrypi:~/mudpi-core $ python
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> import adafruit_dht
>>> dht = None
>>> dht
>>> type(dht)
<class 'NoneType'>
>>> dht = adafruit_dht.DHT11(board.D26)
>>> Unable to set line 26 to input

>>> type(dht)
<class 'adafruit_dht.DHT11'>
>>> dht._use_pulseio
True
>>> dht = adafruit_dht.DHT11(board.D26, use_pulseio=False)
>>> dht._use_pulseio
False
>>> dht.temperate
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'DHT11' object has no attribute 'temperate'
>>> dht.temperature
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/mudpi-core/venv/lib/python3.7/site-packages/adafruit_dht.py", line 243, in temperature
    self.measure()
  File "/home/pi/mudpi-core/venv/lib/python3.7/site-packages/adafruit_dht.py", line 199, in measure
    raise RuntimeError("A full buffer was not returned. Try again.")
RuntimeError: A full buffer was not returned. Try again.
>>>
```

**Expected behavior**:
Could not instantiate the DHT object as when setting the `use_pluseio` to `False` so behavior is the same.

**Issue/Doubt**:
Shouldn't the library raise an `Exception` when (by default) `use_pulseio` is set to `True` instead of actually instantiate the DHT11 object?

SBC: **Raspberry Pi A+**
Libraries:
```
Adafruit-Blinka==6.4.1
adafruit-circuitpython-dht==3.5.5
Adafruit-PlatformDetect==3.4.1
Adafruit-PureIO==1.1.8
```
Python: **3.7.3**

jposada202020, 2021-04-02:
@yeyeto2788 I read this a couple of days ago, interesting, I will look into this during the weekend, hopefully :) I am doing some other stuff, Also I wanted to test the other issue, so maybe I can do both at the same time but no promises, :)

jposada202020, 2021-04-06:
@yeyeto2788 Do you have a solution in mind, maybe some brainstorming in Discord? let me know. Thanks :)

yeyeto2788, 2021-04-08:
Sure, sorry for the late reply.

My suggestion would be to actually either raise the error so anyone can handle it as one wants (My preferred option) or by just now raising the error but not instantiating any object.

If you want to discuss it further do not hesitate on sending me a DM on Discord.

jposada202020, 2021-04-08:
No problem, wlil do. Thanks