adafruit/Adafruit_CircuitPython_DHT #72: [RaspberryPi 4B & DHT11] Checksum did not validate. /A full buffer was not returned.

Luupw3d, 2021-06-09:
- Device: **RaspberryPi 4B**
- Sensor: **DHT11**

When running for a long time, errors are reported frequently.
[image]

FSXLiveriesdotcom, 2021-06-27:
Same issue here. 
RPI 4B
DHT11

[image]

dsaad68, 2021-08-22:
I have the same problem, I have also tried 1 kΩ and 10 kΩ resistors. 
I have also tried this:

`dhtDevice = adafruit_dht.DHT22(board.D4 , use_pulseio=False)`

ilhamsyarief21, 2024-05-11:
> I have the same problem, I have also tried 1 kΩ and 10 kΩ resistors. I have also tried this:
> 
> `dhtDevice = adafruit_dht.DHT22(board.D4 , use_pulseio=False)`

has it been solve?

idotj, 2024-07-10:
Same problem here.  
Running Bookworm and adafruit-circuitpython-dht 4.0.4