adafruit/Adafruit_CircuitPython_DHT #73: DHT does not work with root user using Raspberry Zero WH

JavanXD, 2021-08-16:
I am using a clean Raspberry Pi OS install (version: 2021-03-04-raspbian-buster) on a Raspberry Zero WH. You can [download the image](https://github.com/Honey-Pi/HoneyPi-Build-Raspbian/releases/tag/HoneyPi) or you can check the used install instructions [here](https://github.com/Honey-Pi/HoneyPi-Build-Raspbian).

I installed the following two packages: `sudo pip3 install adafruit-circuitpython-dht && sudo apt-get install libgpiod2`

PoC:
```python
import logging
import adafruit_dht
import time
import board
import digitalio

logger = logging.getLogger('HoneyPi.read_dht')
if __name__ == '__main__':
    try:

        SENSOR_PIN = digitalio.Pin(4) # change GPIO pin
        dht = adafruit_dht.DHT22(SENSOR_PIN, use_pulseio=True)

        timer = 0
        while timer <= 10:
            try:
                temperature = dht.temperature
                temperature_f = temperature * (9 / 5) + 32
                humidity = dht.humidity

                print("Temp: {:.1f} F / {:.1f} °C    Humidity: {}% ".format(temperature_f, temperature, humidity))

                break # break while if it worked
            except RuntimeError as error:
                # Errors happen fairly often, DHT's are hard to read, just keep going
                print(error.args[0])
                time.sleep(2.0)
                timer = timer + 2

            if timer > 10: # end reached
                print("Loop finished. Error.")

    except (KeyboardInterrupt, SystemExit):
       pass

    except RuntimeError as error:
        logger.error("RuntimeError in DHT measurement: " + error.args[0])

    except Exception as e:
       logger.exception("Unhandled Exception in DHT measurement")
```

[(full script)](https://github.com/Honey-Pi/rpi-scripts/blob/master/read_dht.py)

Console:

```
pi@HoneyPi:~/HoneyPi/rpi-scripts $ sudo -u root python3 read_dht.py
RuntimeError in DHT measurement: Timed out waiting for PulseIn message. Make sure libgpiod is installed.
pi@HoneyPi:~/HoneyPi/rpi-scripts $ sudo -u pi python3 read_dht.py
Temp: 78.3 F / 25.7 °C    Humidity: 66.2% 
```

Notes:
yes, root user is part of gpio group (`sudo usermod -aG gpio root`)

I noticed that I am not the only one affected by this issue:
- https://stackoverflow.com/questions/63292226/dht22-works-as-user-but-not-as-sudo-on-pi-zero-w August-2020
- https://forums.adafruit.com/viewtopic.php?f=19&t=160892&start=15#p804598 March-2020

I have been using the Adafruit_DHT library for a long time but because you stopped fixing issues such as [this](https://stackoverflow.com/questions/63232072/cannot-import-name-beaglebone-black-driver-from-adafruit-dht) I've migrated to your new library. Unfortunately it turns out that you do not fix issues for Raspi Zero. What I am supposed to use now? Please fix this issue.

alaub81, 2021-11-02:
I have a similar issue with `adafruit-circuitpython-hcsr04` on a raspberry pi zero w. It works as the normal pi user, but not with sudo or as root user. 

I installed:

```
pip3 install adafruit-circuitpython-hcsr04
```

and my simple script looks like this:
```
#!/usr/bin/python3

import time
import board
import adafruit_hcsr04

sonar = adafruit_hcsr04.HCSR04(trigger_pin=board.D5, echo_pin=board.D6)

while True:
    try:
        print("Distance: %.2f cm" % sonar.distance)
    except RuntimeError:
        print("Retrying!")
    time.sleep(1)
```

I got the same error when I start the script as root user:

`RuntimeError: Timed out waiting for PulseIn message. Make sure libgpiod is installed.`

JavanXD, 2021-11-02:
@alaub81 I created this support request in the Adafruit forum: https://forums.adafruit.com/viewtopic.php?f=24&t=183236&p=890751#p890221  I would recommend to you to save yourself some time and use another library.