adafruit/Adafruit_CircuitPython_DHT #80: DHT sensor not found check wiring

andreiarg, 2022-01-12:
[image]
I have this DHT22 sensor on a board with a 5k resistor between VCC and data line and a capacitor connected to ground. It's connected to my Raspberry Pi 4B. It worked previously with the deprecated  adafruit DHT library but I started getting a Unknown platform issue so I tried moving to the circuitpython library and now I'm getting "DHT sensor not found check wiring" when trying to run the simpletest example code. I've also tried running the following lines of code

```
import board
import adafruit_dht
dhtDevice = adafruit_dht.DHT22(board.D4, use_pulseio=False)
dhtDevice.temperature
```
and get the same error. I tried switching VCC from 3.3V to 5V and also checked continuity using a multimeter. I've tried both GPIO 4 and GPIO 18 and changing the python file accordingly and still get nothing. I don't know what to do any more.

andreiarg, 2022-01-12:
I have also measured the voltage between VCC and data line and it seems to idle at 1.16V and if I blow on the sensor to heat it up and increase humidity voltage drops to around 1.08V if that's of any use. I've also tried using my raspberry pi zero 2w. I will be trying to connect it to my Pico which has a different library on it to see if it works.

jerryneedell, 2022-01-12:
Just a suggestion -- lookin gat the pictured board, it appears to have a pull-up resist already on board. Have you tried removing your external pull-up.

andreiarg, 2022-01-12:
> Just a suggestion -- lookin gat the pictured board, it appears to have a pull-up resistor already on board. Have you tried removing your external pull-up.

Thanks for replying. I haven't installed any external pull up resistor, just connected it straight to the board, and now having it plugged into my Pico using a different library it reads out data fine from the sensor so I know at this point that the sensor is functional.

jerryneedell, 2022-01-12:
hmmm --- FWIW I just hooked up  a DHT22 (with a 10 k resistor) to a Pi400 using 3.3V and GPIO4 and it works normally
```
import time
import board
import adafruit_dht

# Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT22(board.D4)

while True:
    try:
        # Print the values to the serial port
        temperature_c = dhtDevice.temperature
        temperature_f = temperature_c * (9 / 5) + 32
        humidity = dhtDevice.humidity
        print("Temp: {:.1f} F / {:.1f} C    Humidity: {}% "
              .format(temperature_f, temperature_c, humidity))

    except RuntimeError as error:
        # Errors happen fairly often, DHT's are hard to read, just keep going
        print(error.args[0])

    time.sleep(2.0)
```
```
pi@gjnpi400:~/projects/blinka $ python3 dht_simpletest.py 
Checksum did not validate. Try again.
Temp: 70.5 F / 21.4 C    Humidity: 20.1% 
Temp: 70.5 F / 21.4 C    Humidity: 20.1% 
Temp: 70.5 F / 21.4 C    Humidity: 20.1% 
Temp: 70.5 F / 21.4 C    Humidity: 20.1% 
Temp: 70.5 F / 21.4 C    Humidity: 20.1% 
Temp: 70.5 F / 21.4 C    Humidity: 20.1% 
```

jerryneedell, 2022-01-12:
also with use_pulseio=False as you  have it.
```
>>> import board
>>> import adafruit_dht
>>> dht=adafruit_dht.DHT22(board.D4,use_pulseio=False)
>>> dht.temperature
21.4
>>> 
```

andreiarg, 2022-01-12:
I think pulseio was the problem in the end. I installed gpiod and now with pulseio enabled it works. It still doesn't work if it's disabled but at this point I'm satisfied with the result so I think at this point it's fair to close this.

closed 2022-01-12 by andreiarg