adafruit/Adafruit_CircuitPython_DHT #82: De-initializing self.pulse_in

WontedAcorn44, 2022-02-16:
I am trying to use a raspberry pi 4 to capture and store values taken from a DHT22 sensor and I get these errors whenever I try to connect it in properly:
DHT sensor not found, check wiring
#this is when I connect my DH22 sensor
De-initializing self.pulse_in
De-initializing self.pulse_in
Traceback (most recent call last):
  File "/home/pi/humidity.py", line 26, in <module>
    temperature_c = dhtDevice.temperature
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_dht.py", line 274, in temperature
    self.measure()
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_dht.py", line 219, in measure
    pulses = self._get_pulses_pulseio()
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_dht.py", line 141, in _get_pulses_pulseio
    self.pulse_in.clear()
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 150, in clear
    self._mq.send("c", True, type=1)
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/humidity.py", line 44, in <module>
    dhtDevice.exit()
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_dht.py", line 93, in exit
    self.pulse_in.deinit()
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 111, in deinit
    procs.remove(self._process)
ValueError: list.remove(x): x not in list

Can anyone make any sense of this?

My code is as follows:
`# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time
import datetime
import board
import adafruit_dht

# Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT22(board.D4)

e = datetime.datetime.now()
date = "%s-%s-%s" % (e.day, e.month, e.year)
t = time.localtime()
current_time = time.strftime("%H%M%S", t)
file = open('humidityValues'+ date + '_' + current_time +'.txt', 'w')

# you can pass DHT22 use_pulseio=False if you wouldn't like to use pulseio.
# This may be necessary on a Linux single board computer like the Raspberry Pi,
# but it will not work in CircuitPython.
# dhtDevice = adafruit_dht.DHT22(board.D18)

while True:
    try:
        # Print the values to the serial port
        temperature_c = dhtDevice.temperature
        temperature_f = temperature_c * (9 / 5) + 32
        humidity = dhtDevice.humidity
        t = time.local
        current_time = time.strftime("%H:%M:%S", t)
        temperature_c = dhtDevice.temperature
        temperature_f = temperature_c * (9 / 5) + 32
        humidity = dhtDevice.humidity
        str = "time=" + current_time + "   temp={0:0.1f}ºC   humidity={1:0.1f}%".format(temperature_c, humidity)
        print(str)
        file.write(str + "\n")

    except RuntimeError as error:
        # Errors happen fairly often, DHT's are hard to read, just keep going
        print(error.args[0])
        time.sleep(2.0)
        continue
    except Exception as error:
        dhtDevice.exit()
        #raise error

    time.sleep(2.0)`

I tried to remedy the problem with this:
`# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time
import datetime
import board
import adafruit_dht

# Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT22(board.D4)

e = datetime.datetime.now()
date = "%s-%s-%s" % (e.day, e.month, e.year)
t = time.localtime()
current_time = time.strftime("%H%M%S", t)
file = open('humidityValues'+ date + '_' + current_time +'.txt', 'w')

# you can pass DHT22 use_pulseio=False if you wouldn't like to use pulseio.
# This may be necessary on a Linux single board computer like the Raspberry Pi,
# but it will not work in CircuitPython.
# dhtDevice = adafruit_dht.DHT22(board.D18, use_pulseio=False)

while True:
    try:
        # Print the values to the serial port
        temperature_c = dhtDevice.temperature
        temperature_f = temperature_c * (9 / 5) + 32
        humidity = dhtDevice.humidity
        t = time.local
        current_time = time.strftime("%H:%M:%S", t)
        temperature_c = dhtDevice.temperature
        temperature_f = temperature_c * (9 / 5) + 32
        humidity = dhtDevice.humidity
        str = "time=" + current_time + "   temp={0:0.1f}ºC   humidity={1:0.1f}%".format(temperature_c, humidity)
        print(str)
        file.write(str + "\n")

    except RuntimeError as error:
        # Errors happen fairly often, DHT's are hard to read, just keep going
        print(error.args[0])
        time.sleep(2.0)
        continue
    except Exception as error:
        dhtDevice.exit()
        #raise error

    time.sleep(2.0)`

But, all I get are things like:
Checksum did not validate. Try again.
A full buffer was not returned. Try again.

NOTICE: I am in no way passing the code above off as my own purely.

aurorapar, 2023-01-18:
I am having a similar issue. Only getting checksum/not full buffer errors.

```
import time

import board
from adafruit_dht import DHT22

dht = DHT22(board.D15, use_pulseio=False)

while True:
        try:
                dht.measure()
                print(f"Temp: {dht.temperature} Humidity: {dht.humidity}")
        except Exception as e:
                print("Failed")
                print(e)
        finally:
                time.sleep(2.5)
                print('-----')
```

xgqfrms, 2023-05-16:
# As the error message said `DHT sensor not found, check wiring`

So, I think it's your code that sets an error GPIO output PIN.

What's the `board.D4` PIn? Raspberry Pi GPIO is not that PIN as far as I know.

> Try changing your code as below shows

```diff
- dhtDevice = adafruit_dht.DHT22(board.D4)

+ # BCM GPIO 18 PIN
+ dhtDevice = adafruit_dht.DHT22(board.18)
```

[image]

https://pinout.xyz/

FoamyGuy, 2024-11-13:
Closing this for now. Please post in the forums or discord if anyone is still having trouble and needs help with getting it wired up correctly. Please be ready to provide pictures of your wiring.

closed 2024-11-13 by FoamyGuy