adafruit/Adafruit_CircuitPython_DHT #98: Raspberry Pi 4 + DHT11 Sensor no floating point temps

dvelaren, 2024-04-14:
Hello Adafruit Devs,

I would like to report a problem dealing with DHT11 sensor readings using your library. It seems that the decimals for the temperature are missing. I'm only getting temperatures as like an integer.

Here is the minimal code to reproduce the issue:

```python
import time
import board
import adafruit_dht

sensor = adafruit_dht.DHT11(board.D17)

while True:
    try:
        temperature = sensor.temperature
        humidity = sensor.humidity
        print(f"Temp: {temperature:.2f} °C, Hum: {humidity:.2f} %")
    except RuntimeError as e:
        logging.error(f"Reading from DHT failure: {e.args[0]}")
        time.sleep(2)
        continue
    except Exception as e:
        sensor.exit()
        raise e
    time.sleep(2)
```

I'm getting this output alwas:
```sh
Temp: 28.00 °C, Hum: 64.00 %
Temp: 27.00 °C, Hum: 64.00 %
Temp: 28.00 °C, Hum: 64.00 %
```

TheChilliPL, 2024-05-24:
I have the same issue

manyhuis, 2024-10-24:
I did a little digging and found a fix for myself. If you are able to edit the source adafruit_dht.py line...

Change Line 239 from
`new_temperature = buf[2]`

to
`new_temperature = buf[2] + float(buf[3])/10`

PR #102 (2024-11-20): Read decimal value also for dht11
@ladyada 

Resolves: #98 

Looking at the datasheet does seem to confirm the 4th byte holding data for the decimal value by my understanding. I also referenced the Arduino driver and found comparable step in that code here: https://github.com/adafruit/DHT-sensor-library/blob/master/DHT.cpp#L95 and borrowed from it's syntax some.

I didn''t think the float cast mentioned in the issue code is necessary because the division will make it into a float.
closed 2024-11-20 by FoamyGuy