adafruit/Adafruit_CircuitPython_DHT #100: What is the difference between pulsein=Ture and pulsein=False?

Y-k-Y, 2024-05-20:
I'm using it on pi 5 and i disabled pusein since the 2rd executing issue still persist. As example shows the usage of `pulsein=False` i can confirm that this fixes the 2rd executing issue.

I just want to know the difference between `pulsein=Ture` and `pulsein=False` How does it work when `pulsein=False`? I've read that `pulseio` is the main component that does most of work of reading IR receiver pulses then how does it read without `pulseio`?

dhalbert, 2024-05-20:
It reads the pulses by polling the data pin:
https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/cec6795c87c32624946994664cdce4332ca0b44e/adafruit_dht.py#L151.

Could you explain what you mean by the "2rd executing issue"? Is there an open issue that is what you mean?

FoamyGuy, 2024-11-13:
Closing this for now. Main question was answered and no further details about the "2nd executing issue" were provided, I didn't see anything in the currently opened issues that seemed related. Feel free to re-open if there is still some issue.

closed 2024-11-13 by FoamyGuy