adafruit/Adafruit_CircuitPython_DHT #105: dht11 module issue

TAHMINAISLAM673, 2025-02-13:
hello i am trying to get temperature and humidity using DHT11 sensor using resbi 4 B .But it says unable to set pin 4.I tried to another pin it still says same thing.

TAHMINAISLAM673, 2025-02-13:
import board
import adafruit_dht

dht = adafruit_dht.DHT11(board.D4)  # Change to DHT22 if using one

try:
    temp = dht.temperature
    humidity = dht.humidity
    print(f'Temperature: {temp}°C, Humidity: {humidity}%')
except RuntimeError as e:
    print(f"Error: {e}")
 

this is my code

TAHMINAISLAM673, 2025-02-13:
import sys
import time
import Adafruit_DHT

# Set pin to GPIO 4
pin = 4  

# Set sensor type to DHT11 (use 22 for DHT22)
sensor_type = 11  

try:
    while True:
        # Introduce a delay
        time.sleep(2)  

        # Read data from the sensor
        humidity, temperature = Adafruit_DHT.read_retry(sensor_type, pin)

        # Convert temperature to Fahrenheit
        fahrenheit = (temperature * 1.8) + 32  
        celsius = temperature  

        # Check if data is valid, then print; otherwise, show an error
        if humidity is not None and temperature is not None:
            print("Temperature: {:.1f}°C ({:.1f}°F)  Humidity: {:.1f}%".format(celsius, fahrenheit, humidity))
        else:
            print("Cannot read from device")

except KeyboardInterrupt:
    sys.exit()
when I use this code it says get platform

caternuson, 2025-02-13:
I edited your post to add code markup to make them readable.

caternuson, 2025-02-13:
Python is case sensitive, so your second code example is not importing correctly. Should result in some error on that line.

Your first code example looks generally OK.

> But it says unable to set pin 4

What is the full text of the error message you get here?

TAHMINAISLAM673, 2025-02-13:
> [`[/](url)`](url)

[image]
[image]

TAHMINAISLAM673, 2025-02-13:
> Python is case sensitive, so your second code example is not importing correctly. Should result in some error on that line.
> 
> Your first code example looks generally OK.
> 
> > But it says unable to set pin 4
> 
> What is the full text of the error message you get here?

My first code say unable to set pin 4 to inpuy

caternuson, 2025-02-13:
The majority of the error message is not shown. Can you copy all the lines of error message text here instead of a photograph?